bigtreetech/BIGTREETECH-TouchScreenFirmware #1280: TFT 70 Trouble Installing

RavaszEmber, 2020-11-22:
I'm having trouble installing the latest firmware on my TFT 70.

I downloaded "BIGTREETECH-TouchScreenFirmware-master" and navigated to Copy to "SD Card root directory to update - Unified Menu Material theme".
I then moved "config.ini" "BIGTREE_TFT70_V3.0.26.x" and the "TFT70" folder to my SD card. These files are in the attached zip file.
[Files.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5579110/Files.zip)

I then loaded this to my TFT70 and powered on my printer. When updating I had errors that icons were missing, namely "S_SAVE.bmp" and "S_LOAD.bmp".
When updating configuration, I had errors that keywords were invalid, namely "mesh_min_color:7" and "mesh_max_color:2"

After the display finished updating I was not prompted with a calibration screen and I was unable to use the touchscreen.

oldman4U, 2020-11-22:
@bigtreetech 

Next user struggling with outdated binaries.

When will you release the PRs!!??

oldman4U, 2020-11-22:
Hi.

You did everything right, but unfortunately the owner of this repository is - for an unknown reason - not able to merge even the most important PR's in an acceptable timeframe. Sorry

Please use the repository below, instead of the official one.

Thank you

https://github.com/radek8/BIGTREETECH-TouchScreenFirmware/tree/updateBIN

RavaszEmber, 2020-11-22:
Thanks old man, that update worked great. However, I am still unable to use the touchscreen. I assume I have to calibrate it but I can't get to the calibration menu without the touchscreen in the first place. Is there a way to run the calibration command from the SD card after updating?

Thanks!

kisslorand, 2020-11-22:
Make an empty "reset.txt" file on your card and reboot TFT with the card inserted.

oldman4U, 2020-11-22:
https://github.com/oldman4U/BIGTREETECH-TouchScreenFirmware#troubleshooting

Please help the community and close the ticket once you do not need it anymore.

Thank you

RavaszEmber, 2020-11-22:
Thanks for the help, I just have one last issue. My screen is stuck reporting that no printer is connected. I checked that the 5 pin TFT connector is connected to my SKR 1.3 and checked the that the baudrate for the display is the same as the baudrate I have in Marlin (115200).
I'm guessing the problem is my serial pin isn't defined for the TFT connector, but I'm not sure how to check this. Here's my relevant Marlin code:
[image]

RavaszEmber, 2020-11-22:
Should SERIAL_PORT be set to 0 and SERIAL_PORT_2 be set to -1?

oldman4U, 2020-11-22:
Yes, I would change them, so the connection to the TFT is the one with more priority, even the two SHOULD be the same......

Send a picture which shows the connection from the TFT to the mainboard.

RavaszEmber, 2020-11-22:
Bit of a mess but hope this helps.
[image]
[image]
The pin on the far left of the tft header on the mainboard is the single reset pin. I tried this the other way around just to make sure it wasn't a connection issue.

oldman4U, 2020-11-22:
Thank you for the picture. And what you wrote is exactly what I wanted to find out;-)

For the Touchscreen connection 3 things are important.

1. cable working and connected right
2. serial port defined for TFT connection
3. baud rate the same on TFT and mainboard

The No Printer Connected simply means that the serial port of the TFT can not "find" the mainboard. Check the cable with a multimeter.... last hope I must say.;-(

RavaszEmber, 2020-11-23:
I checked the cable for continuity and it all checks out (I'd be baffled if they were somehow defective).
I've been putting off updating my mainboard because of the pain that is to do. It sounds like I should change SERIAL_PORT to 0 and SERIAL_PORT_2 to -1 and hope that works. I always doubt that the hardware would be an issue. Maybe I have the wrong display type set in marlin or something? I've attached my Marlin configuration.h in case you're able to spot an issue there. If not I'll update the mainboard only changing SERIAL_PORT and SERIAL_PORT_2.

Thanks for the help!

[Marlin_Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5580676/Marlin_Configuration.zip)

oldman4U, 2020-11-23:
No Ptinter Attached error only the previous mentioned parameters are
relevant. All other parameters are Marlin mode related.

Doyle4, 2020-11-23:
Try changing the Baudrate in the settings, even if they match in the firmware, its been known to still not work and need changing, it maybe set to 115000 at the moment, try 25000 or vice versa.

oldman4U, 2020-11-23:
@Doyle4 

You mean to change the baud rate back and forth between two values, where the one which finally works has been selected the time the TFT has been started, but it even did not work having the correct setting enabled???

Do you experience this?

Doyle4, 2020-11-23:
I have had a screen that would only work with 115000 when the printer was set to 250000, how it worked I have no idea.
Wrong naming for types maybe?

oldman4U, 2020-11-23:
And you know exactly which screen and which firmware version in combination with which mainboard and firmware? Was it BTT stuff?

radek8, 2020-11-23:
Note: Ensure that following options are enabled in Marlin firmware

  M115_GEOMETRY_REPORT (in Configuration_adv.h)
  M114_DETAIL  (in Configuration_adv.h)
  REPORT_FAN_CHANGE (in Configuration_adv.h)
  EMERGENCY_PARSER (in Configuration_adv.h)
  SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)
  HOST_ACTION_COMMANDS  (in Configuration_adv.h)

 For printing from onboard SD

  SDSUPPORT (in Configuration.h)
  AUTO_REPORT_TEMPERATURESS (in Configuration_adv.h)
  AUTO_REPORT_SD_STATUS (in Configuration_adv.h)
  LONG_FILENAME_HOST_SUPPORT (in Configuration_adv.h)
  SDCARD_CONNECTION ONBOARD (in Configuration_adv.h)

oldman4U, 2020-11-23:
> Note: Ensure that following options are enabled in Marlin firmware
>
> M115_GEOMETRY_REPORT (in Configuration_adv.h)
> M114_DETAIL (in Configuration_adv.h)
> REPORT_FAN_CHANGE (in Configuration_adv.h) [...]

@traffic-light Could you please add this information to your ReadMe first document. Thank you

radek8, 2020-11-23:
According to the attached photo, the connector is connected upside down.

RavaszEmber, 2020-11-24:
I haven't gotten a chance to update the Marlin code on my mainboard yet. I'll try changing the mainboard's baudrate to 250000, SERIAL_PORT to 0, SERIAL_PORT_2 to -1 and I'll check that the settings mentioned by radek8 are enabled.
I double checked my wiring based on radek8's comment and the way I have it wired is correct. This can be seen with the attached picture that shows the pin labels on the bottom of my physical board. I'm not sure why my pinout has reset in a different spot than in the skr 1.3 pinout. When the connection is flipped the screen does not turn on. I can conclude the physical connection is correct.
[image]

I have tried changing the buadrate to every preset available and the "printer not attached" issue was never solved with any baudrate. 
If my understanding is correct, the only things needed for the display to attach to the board is wired TFT connection and the same baudrate. I'm not sure why this issue persists, perhaps I'm looking over something simple? It seems to me that I set everything required to connect the screen to the printer. I'm hoping a Marlin update will fix this.

RavaszEmber, 2020-11-24:
I just looked at it again and my connector is in backwards, but the right way leaves me with a blank screen (but a light on in the back) and the wrong way leaves me with an operating screen. I'm very confused. Pulling out jumpers and the multimeter.

RavaszEmber, 2020-11-24:
Alright at this point I'm just baffled. Why does my screen light up when I connect RS232 + and - to RST and R0 (respectively) but the screen does not light up when I connect RS232 + and - to 5V and G respectively?

radek8, 2020-11-24:
You got everything wrong

RavaszEmber, 2020-11-24:
@radek8 care to elaborate? When the connector is placed backwards my screen turns on, when it's in the right way the screen remains off.

radek8, 2020-11-24:
Connect the serial cable via RS232 correctly rotated.
Connect cables EXP1 and EXP2 as shown.
You used EXP3, so the print gets into the reset loop
You must not use the EXP3 connector

[image]

radek8, 2020-11-24:
[image]

RavaszEmber, 2020-11-24:
[image]
That worked, thank you!
One last thing, when I switch to Marlin Mode now the screen remains blank. Am I not able to switch between the two? (pic below)
[image]

radek8, 2020-11-24:
Marlin mod communicates after EXP1 and EXP2
The cables must be connected correctly.
In Marlin you have to enable (delete //)
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

all other display drivers including
// # define CR10_STOCKDISPLAY
you have to comment

RavaszEmber, 2020-11-24:
Thanks a bunch!

TLDR: Wrong connections 😛

closed 2020-11-24 by RavaszEmber

oldman4U, 2020-11-24:
Benjamin, which instructions did you follow initially, to connect the TFT?
What made you believe the cables are connected right?

radek8, 2020-11-24:
Unnecessary mistakes due to inattention.
You're lucky you didn't destroy anything

RavaszEmber, 2020-11-24:
I had originally connected EXP 3 so I could use the original marlin display when I first got it. I guess I flipped the TFT connection to the reverse because the correct way led to a blank screen. I overlooked the EXP connections and thought it was an issue with the serial connection from our conversation.

radek8, 2020-11-24:
Did you fix the Marlin configuration?
Is the Marlin simulation already working?

RavaszEmber, 2020-11-24:
I did not @radek8. I think I'll be able to make the changes based off your comments. I'll do it when I have time and let you know if I run into any problems.

radek8, 2020-11-24:
OK,
don't forget the settings I sent earlier.

oldman4U, 2020-11-24:
Usually a manual or YouTube instructions are used to get the information
needed to install something like a display. Or did you just try it and once
it did not work the next was to open a ticket here? Have you seen and read
the ReadMe at the front page of this repository?

RavaszEmber, 2020-11-24:
@oldman4U I did reference a couple YouTube videos before installing. The original motivation for this thread was due to the errors I encountered when updating the display's firmware from the BTT Github. Having errors with the official github code caused me to open this thread (I'm not sure how I would have resolved that if I didn't make a thread). Since I still had issues after that I continued the thread until it was pointed out that connections were overlooked.

oldman4U, 2020-11-24:
Ok. Thank you

Happy printing

billyd60, 2021-04-13:
Hi

I am trying to connect the tft 70 to an archim2.  My serial port pins (on the main board) are not in the same order as the connector, e.g. the rst pin is in the center and the ground and + are reversed. Anyway I re-ordered the pins in the connector (that will attach to the main board) to match but I have a question. Should tx from the display go to rx on the board and vice versa? That's normally what you do in serial connections but I wanted to make sure.

-Bill

oldman4U, 2021-04-14:
Check the documentation of SKR board and your TFT where the connection works without twisting wires.

billyd60, 2021-04-14:
> Check the documentation of SKR board and your TFT where the connection works without twisting wires.

I'm using the archim2. I've isolated the issue to the serial port. Can't seem to get the archim2 serial port functioning in marlin2

radek8, 2021-04-14:
You must select the correct port number in Marin

billyd60, 2021-04-14:
> You must select the correct port number in Marin

Thanks I heard back from ultimachine and I needed to define serial_port -1 and serial_port_2 0

Of course BAUD had to match on both as well I used 250000 and that resolved the serial connection on J28 (on the Archim2 board).

Now I have the BTT TFT70 v3 working via serial, but I can't get the 12864 marlin emulator working through the two ribbon cable connectors. When I plug those in to SC-1 and SC-2 on the Archim2 board the BTT TFT70 gets stuck at the choosing screen (Touch screen or Marlin emulator) and will not respond. I've verified the cables are all correct, pin1 to pin1 but nothing works until I disconnect those cables. I suppose I could just use the touchscreen and forget about Marlin emulation but it would be nice to have everything working as it should.

radek8, 2021-04-14:
Depending on the connection, it seems that the Archim2 board has a different connection of the SC1 and SC2 connectors, opposite the EXP1 and EXP2 connectors on the TFT display. Your old display works because it has the same atypical connection as Archim2

[image]

[image]

radek8, 2021-04-14:
The pins defined are, if you select the right display driver, it could work ...

#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

[image]

billyd60, 2021-04-14:
So are you saying the pins are wrong or right? How do I fix it?

radek8, 2021-04-14:
And the Marlin mod works with
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER ?

radek8, 2021-04-14:
I don't have your board and I can't try it.
But when I look at the pin definition and wiring diagram, it looks like 2 pins have been swapped in the pin definition file
#define BEEPER_PIN 23 // D24 PA15_CTS1
#define LCD_PINS_ENABLE 24 // D23 PA14_RTS1

[image]

billyd60, 2021-04-14:
I just discovered something strange on the archim. My sc1 connector has 3.3v across 9 and 10 which is supposed to be on sc2. And my sc2 has 5v across pins 1 and 2 which should be 9 and 10 on sc1!

So it appears they are not only marked wrong but one is upside down.

radek8, 2021-04-14:
Yes SC1 has a 5V supply
and comes to connect with EXP1 on the TFT
CS2 has a 3.3V power supply but the pin is not connected to EXP1.
Check that the connection is correct

radek8, 2021-04-14:
[image]

billyd60, 2021-04-14:
Yes and in that picture my sc2 is apparently where sc1 is supposed to be and pin 1 is the upper left. And sc2 is where sc1 is supposed to be and pin 1 is the upper right.

billyd60, 2021-04-14:
How in the world did that happen? Is it something I. The pins file? Seems no because you can't change ground with firmware can you?

radek8, 2021-04-14:
No, the power pins are set and cannot be rotated. But in the pin file you can give individual pins of the processor different functions ...

radek8, 2021-04-14:
[image]

radek8, 2021-04-14:
Disconnect the serial cable. Only connect EXP1 - SC1.
The TFT display should boot if the cable is oriented correctly, the touch screen will report the printer is not connected, Marlin mode should display the initial screen but will not respond to knob rotation.
Can you try it?

billyd60, 2021-04-14:
In your picture I get 3.3 volts on the right hand most pins of the top connector labelled on the board as sc1 with the ground on the top pin.

I get 5v on the left hand set of pins of the bottom connector in your photo labelled sc2 with ground on the bottom pin.

Crazy

radek8, 2021-04-14:
You measure 3.3V if the processor pin is at a high level, that's fine.
The important thing is where you measure the 5V power supply

radek8, 2021-04-14:
Either the manufacturer has poor documentation, or there are several HW variants of this board.

radek8, 2021-04-14:
Try the connection I suggested above

[3 more replies]