bigtreetech/BIGTREETECH-TouchScreenFirmware #1281: THT35v3E3 can`t activate mesh leveling menu on touch mode

FoxtrottPL, 2020-11-22:
SKRmini e3v2 with TFT 35 v3e3 
1.board firmware prepare to use mesh bed leveling with 6x6 grid
2 after factory reset of motherboard then  prepare to upload TFT firmware
3 I used 3 firmwares Unified , "normal" ,Red  
 a.on unified - missing few bmp
 b.on normal wrong keywords mesh_min_color and mesh_max_color 
 c.Red - bltouch menu present (without blt connected )
4. When I switch to Merlin mode - mesh leveling present and used to calibrate bed . Stored to eeprom, activate eeprom.
5.Switching back to touch mode - mesh leveling menu not present , mesh grid not present.

oldman4U, 2020-11-22:
In case your firmware under Menu - Settings - Info is from September, please read this and see if it solves your problem:

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1282

Thank you

oldman4U, 2020-11-25:
Please help the community and close the ticket once you do not need it anymore. Thank you

Codel1417, 2020-12-29:
Same issue here. MArlin 2.0.7.2, TFT December 26 master

Codel1417, 2020-12-30:
`switch (key_num)
    {
      case KEY_ICON_0:
      case KEY_ICON_1:
      case KEY_ICON_2:
      case KEY_ICON_3:
        ublSlot = key_num;

        setDialogText(UBLSaveLoadItems.title.index, LABEL_CONFIRMATION, LABEL_CONFIRM, LABEL_CANCEL);
        showDialog(DIALOG_TYPE_QUESTION, ublSaveloadConfirm, NULL, NULL);
        break;`
This looks valid but it could be the issue. Every other piece of gcode checks out so i wonder if the correct slot is being used. I need to debug over serial to confirm

Codel1417, 2020-12-30:
Strange.  UBL suddenly working perfectly through tft on same firmware from my original response. I wonder if this is a ram overwrite issue of some kind?

Codel1417, 2020-12-30:
1. G29 P1  // initial Level
2. G29 P3 // fill in some gaps
3. M118 A1 UBL Complete
4. G29 S0  // save to slot 0
5. M500 // save to eeprom
6. M420 S1 //enable ubl

> Send: M420 S1
Recv: echo:Invalid mesh.
Recv: Error:Failed to enable Bed Leveling
Recv: echo:Bed Leveling OFF

Only 57/100 points were filled using this command setup, which is what is currently used. all 100 must be filled in.

Codel1417, 2020-12-30:
@FoxtrottPL have you tried any of the newer binaries?

Codel1417, 2020-12-30:
Sorry for solving another issue within this one. kinda got carried away. Still would have fit under the same title though.

oldman4U, 2020-12-30:
Hi. I add @digant73. He developed this and is very likely not watching this 1y old ticket.

digant73, 2020-12-31:
I have also Marlin 2.0.7.2 and with a 10x10 grid a didn't have that problem (with 57 of 100 points probed). Also in the Marlin home page, I didn't read any need to use multiple G29 P3 to fill in the grid.
However, it seems that the solution works for you

Codel1417, 2020-12-31:
@digant73 its specific to Unified Bed Leveling. Marlin requires all points to be filled. in their demo gcode they indicate to rerun g29 p3 as many times as needed

digant73, 2021-01-02:
@foxtrottpl if you don‘t see bl and mesh editor menu, verify all the marlin options reported at the beginning of config.ini are enabled in marlin fw

Nascire, 2021-01-06:
I had the same problem (Invalid Mesh / Cannot activate bed leveling) - through this issue I tried G29 P3 (as some spots where empty) and now it´s working.
Manually activating in LCD-Mode was no problem however.

PR #1518 (2021-01-21): Added Home Offset menu + bugfixes and cleanup
**NEW FEATURES:**
1) Added Home Offset menu. Previous ProbeOffset menu is now a generic Z Offset menu (just to not duplicate code) that will invoke Home Offset or Probe Offset API. Both the Home and Probe Offset menus are available on the Bed Leveling menu. Furthermore, Probe Offset menu is available only if probe is enabled (supported) in Marlin fw.
2) Added support to MBL Z offset (G29 S4 Zxx) in Babystep menu. Babystep menu will allow to change/save Z offset based on the following rules (matching sequence):
 *) Probe Offset: If probe is supporte [... cut]
closed 2021-01-21 by bigtreetech