bigtreetech/BIGTREETECH-TouchScreenFirmware #1282: BTT unpredictable behaviour to maintain its own repositories

oldman4U, 2020-11-22:
Dear user.

BTT is for an unknown reason not able to merge the valuable work of the open source community into its own repository at a regular timebase. Because of this, fixed bugs, new features, precompiled firmware and up to date ReadMe files are still waiting to be merged.

Unfortunately only @bigtreetech and @Msq001 are able to do so, so please use this PR  for the latest pre compiled firmware, config.ini TFT folder and language: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1265
and read the ReadMe from here, before doing so:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1251

Please read the pinned "Read me before posting" plus the pinned "Feature Request List" or https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1218, in case you would like to ask for a certain feature or report an issue.

Thank you

Stellarspace, 2020-11-26:
The solution would be to stop helping maintain the repository. You don't work for them therefore it isn't your problem. GitHub allows anyone to create a fork of a repository. If the developers that submit pull requests for this repository agree that BigTreeTech isn't properly maintaining the repository, then a fork can be created that merges pull requests faster by giving write access to more people.

oldman4U, 2020-11-26:
Maybe this would be the best solution. I believe BTTs behavior frustrates
most of us and it is getting worse every week. No new version for months. 9
PRs waiting and not a single issue solved by the manufacturer itself. I
also do not understand how the sponsoring works. At least the two persons
currently sponsored I have never seen here and other persons like Gurmeet
which are contributing frequently get nothing. Not sure I understand the
concept right🧐

oldman4U, 2020-11-27:
Hope we will see BTT to show up more often in the future.

closed 2020-11-27 by oldman4U

Codel1417, 2020-12-29:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1450
Binaries are now up to date with the master branch

oldman4U, 2020-12-29:
Hi. 

I saw your PR but was afraid to ask if it really does what I expected that it would do. 

Thank you