bigtreetech/BIGTREETECH-TouchScreenFirmware #1285: Extruder moves to front left but does NOT park when filament runout is detected

Chad541121, 2020-11-24:
I am using a SKR 1.4 turbo and a BTT tft35 e3 V3.0 on an ender 5 plus.
I have the filament runout sensor plugged into the display cause I want to use the touchscreen.
Filament runout works properly, however the nozzle does NOT lock.  So when I change the filament, if I move it the slightest
bit, then the print is ruined.

Please help.

oldman4U, 2020-11-24:
Please close https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/89

Thank you

radek8, 2020-11-24:
Permission required for M600
ADVANCED_PAUSE_FEATURE
in Marlin. Are you allowed

radek8, 2020-11-24:
> Please close [bigtreetech/BIGTREETECH-TFT35-V3.0#89](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/89)
> 
> Thank you

Why do you still want to close all the tickets without looking at the problem if it can't be solved?

oldman4U, 2020-11-24:
Because I told him to start the ticket here.

The TFT35 v3 repository is useless when it comes to support from users.

radek8, 2020-11-24:
yeah close has 89 :-)

oldman4U, 2020-11-24:
I knew you would agree;-) There are already enough orphaned tickets. Let's see what happens with this one.

oldman4U, 2020-11-24:
> Because I told him to start the ticket here.
> 
> The TFT35 v3 repository is useless when it comes to support from users.

......

there is no radek8 there...

;-)))

radek8, 2020-11-24:
isn't it the same problem?
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/859

radek8, 2020-11-24:
When paused, the head parks on the edge, the motors are active and it is possible to eject the thread.
What do you have a problem with?
How old are you FW in TFT?

Chad541121, 2020-11-24:
The head moves to the edge but doesn’t park, or lock. So if I barely move the head then press resume, my print is misaligned. I am using the newest firmware for the tft and marlin 2.0.5 I believe. Whichever was newest 2 months ago

radek8, 2020-11-25:
This means that you use old FW in both Marlin and TFT.
Update both FWs before you want to solve anything.
Do not use precompiled bin files for TFT, they are 2 months old.
Compile your own bin file.

Chad541121, 2020-11-25:
I guess I am confused.  I thought Oldman4u asked me to close this ticket and post in the general tft section.  I am new.  I thought that's where this was posted.  So I closed this and the other one I had asked awhile ago, and created a new one.  Now I'm being asked why I created a new one.  Ugh. What should I do now?  
radek8 says I use old FW, however I just downloaded latest marlin FW for board 2 months ago.  I also downloaded bin file for the tft.  How would I compile my own tft file to fix this issue?

radek8, 2020-11-25:
2 months old is too old.
download the current FW for TFT here

https://github.com/radek8/BIGTREETECH-TouchScreenFirmware/tree/updateBIN/Copy%20to%20SD%20Card%20root%20directory%20to%20update%20-%20Unified%20Menu%20Material%20theme

radek8, 2020-11-25:
You configure the TFT with an INI file.
You have to compile Marlin yourself according to your own configuration.

Chad541121, 2020-11-25:
If I understand correctly, compiling my own firmware with vscode/platformio will always give me the latest updates?

radek8, 2020-11-25:
The FW for TFT that is exposed on official storage is obsolete. in 2 months a lot has been fixed and new functions have been added.
Compile your own or use the link I sent.
Maybe that will solve your problem.

Compile the FW for the Marlin board yourself. Use the bugfix branch is more current. 
Be sure to set the parameters that the comment in the config.ini file with the TFT configuration points out.

oldman4U, 2020-11-25:
Hello Chad541121.

It is all good with the ticket. radek8 did not know that you already started a ticket with the same topic in another repository - and this is where the confusion began. Now you have one ticket for this topic in the correct repository and all is good.

The remaining problem now is, that your hardware uses old firmware and that the reported issue is very likely related to this old firmware.

Compiling your own software allows you to use the latest firmware, but also to change parameters inside the firmware. radek8 was so kind to compile new versions using almost up to date firmware version, which you can find here (link from radek8 above): https://github.com/radek8/BIGTREETECH-TouchScreenFirmware/tree/updateBIN/Copy%20to%20SD%20Card%20root%20directory%20to%20update%20-%20Unified%20Menu%20Material%20theme

To know how to update the hardware to new firmware, or in case you would like to learn how to compile your own firmware (which is easy for the TFT and not so easy for the Marlin firmware of your SKR), I would recommend to read this: https://github.com/oldman4U/BIGTREETECH-TouchScreenFirmware

Please let us know, once you updated the firmware using one or the other method.

Thank you

Chad541121, 2020-11-25:
Thank you for that.  I understand the TFT firmware part.  If I update the board firmware, are all 3d printer settings lost?  Like the PID tuning, speed, E-steps, etc?  Will it be like starting over with all the calibration?

radek8, 2020-11-25:
Yes, but ...
Connect a PC, with the command M503 you will get the current configuration, which you will save.
Most configuration parameters (PID, step, offset, current, acceleration, jerk, TMC threshold ...) can then be saved directly to the Marlin configuration.
When restoring the default configuration, you will then have all parameters current.

Chad541121, 2020-11-25:
Which branch/repository should I use? marlin 2.0.x or the marlin bugfix-2.0.x ?

radek8, 2020-11-25:
Bugfix - 2.0.x

oldman4U, 2020-11-29:
Hi.

What is the status of the reported issue?

Please let us know

Thank you

Chad541121, 2020-11-29:
I compiled the firmware Friday just before I left work. I will upload the firmware and report back Monday.

Chad541121, 2020-11-30:
Well I tried to compile marlin and I am having issues with the touchscreen saving my settings.  I need to save the offset of the bltouch probe and it won't save.  It reverts back to zero when I go into the menu to check if it saved.  I am working on this and will report back if the newest firmware fixed the park nozzle issue.

radek8, 2020-11-30:
The probe offset is stored here

[image]

Chad541121, 2020-12-01:
I don't know what went wrong with compiling my own firmware.  I ended up finding precompiled firmware thanks to Kersey Fabrications.  All the settings save fine now.  ADVANCED_PAUSE_FEATURE is working!  I want to note that I have the filament runout sensor hooked to the tft35.  I only need to figure out how long the nozzle will stay locked for?  I print at work and may run out of filament when I'm gone for the weekend.

Chad541121, 2020-12-01:
Update. I have been away from my desk for about an hour and the nozzle now moves freely. What should I change to make this an indefinite pause until I press the resume?

Chad541121, 2020-12-02:
Update.  I only use the tft35 in touchscreen mode.  When I compile the lasted Marlin2.0Bugfix, I can't save my settings, such as probe offset.  When I change the value and exit the menu, the setting is lost.  However, I do have older firmware without this issue.  Is this a bug?  I have used Winmerg to compare my config.h and config_adv.h files with the bugfix ones.  I then changed the bugfix config files to match my known working config files.  Compiled the firmware and now have this issue.  I can use the older firmware without this issue.  Please advise.  I want to do this right and have been instructed to make sure I use the latest firmware, but its causing this issue.  Once I have this issue fixed then I'm hoping to finish fixing the nozzle park until filament change issue.

radek8, 2020-12-02:
In Marlin (configuration_adv.h) you can set the time after which the motors are released.
#define DEFAULT_STEPPER_DEACTIVE_TIME 120
value120 is the time in seconds
I assume that if you set -1, the motors will not be switched off.
You can try it.

radek8, 2020-12-02:
If you don't save offset values, it looks like you don't have eeprom enabled
Can you check that you have the M503 command enabled?
It should look like this:

[image]

Chad541121, 2020-12-02:
Thanks radek8.  Reading the notes in marlin. I am going to try setting to 0.  If that doesn't work then I will try setting x,y,z to false.  
` * Idle Stepper Shutdown
 * Set DISABLE_INACTIVE_? 'true' to shut down axis steppers after an idle period.
 * The Deactive Time can be overridden with M18 and M84. Set to 0 for No Timeout.
 */
#define DEFAULT_STEPPER_DEACTIVE_TIME 0
#define DISABLE_INACTIVE_X true
#define DISABLE_INACTIVE_Y true
#define DISABLE_INACTIVE_Z true  // Set 'false' if the nozzle could fall onto your printed part!
#define DISABLE_INACTIVE_E true`
As for the lost settings, I have determined that the all probe offset changes made in touchscreen mode are being saved to eeprom.  However, the values for the x and y offsets ONLY are reverting to 0 in touchscreen mode.  In marlin mode they are the values I have saved.  The z value always displays correctly. This must be a bug?  When starting a print in marlin mode,  its using the eeprom settings for x,y,z offset, but if i start a print in touchscreen mode it is ONLY using ) for the x and Y offsets and my save value for the z offset.

radek8, 2020-12-02:
You are right the probe offset does not display well.

Chad541121, 2020-12-02:
I edited my comment to also state that it is ignoring the offsets completely.  Can you confirm this as well?

radek8, 2020-12-02:
I will try to make a correction on the probe offset

radek8, 2020-12-02:
So the problem is in Marlin, it does not send the correct values to the M851 or M503 command

Chad541121, 2020-12-02:
Do I need to go to the Marlin Github page and post this issue?

radek8, 2020-12-02:
https://github.com/MarlinFirmware/Marlin/issues/20358

oldman4U, 2020-12-02:
> MarlinFirmware/Marlin#20358

This is a full service repository 😎

radek8, 2020-12-02:
Oldman how old is your version of Marlin?

oldman4U, 2020-12-02:
I use the one which has been released the same day 2.0.7.2 has been
release.

radek8, 2020-12-02:
If you use a probe, can you send the M851 command via the terminal if you have the X and Y parameters in the answer?

Chad541121, 2020-12-02:
I just downloaded the latest marlin bugfix-2.0.x repository and recompiled it today.

Chad541121, 2020-12-02:
When I run M851 from the touchscreen terminal.  My probe offsets are X0.0000 Y0.0000 Z-2.9000
Marlin mode they are X-48 Y-0.3 Z-2.9 but is not using these values when homing.

radek8, 2020-12-02:
Sorry it was a question for Oldman

radek8, 2020-12-02:
I have a 6 day old version and it behaves just like yours.
I wanted a comparison with Oldman

radek8, 2020-12-02:
Chad541121, do you have this mistake with today's Marlin compilation?

oldman4U, 2020-12-02:
Sorry, but I have no printer with a probe at the moment. The one with a probe is currently under construction;-(

radek8, 2020-12-02:
does not matter

radek8, 2020-12-02:
Chad541121compile the latest Marlin.
There is no longer a problem with the new build.

Chad541121, 2020-12-03:
I just downloaded the Marlin bug fix 2.0.X today. What marlin version is newer that I should use?￼

Chad541121, 2020-12-03:
I downloaded the bugfix 2.0.x repository again this morning and recompiled the firmware. All issues are resolved, including the nozzle park. Thanks for all your help.

oldman4U, 2020-12-03:
radek8 is THE BEST!!;-)

Please close this ticket once you do not need it anymore.

Happy printing

closed 2020-12-03 by Chad541121