bigtreetech/BIGTREETECH-TouchScreenFirmware #1286: Nozzle doesn’t park on tft35-e3 v3.0

Chad541121, 2020-11-24:
I am using a SKR 1.4 turbo and a BTT tft35 e3 V3.0 on an ender 5 plus.
I have the filament runout sensor plugged into the display cause I want to use the touchscreen.
Filament runout works properly, however the nozzle does NOT lock. So when I change the filament, if I move it the slightest
bit, then the print is ruined.

Please help.

radek8, 2020-11-25:
Why are you registering new tickets for the same problem?

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1285

oldman4U, 2020-11-25:
He is new. Likely he closed the ticket by mistake...

Chad541121, 2020-11-25:
I had asked the question a couple times. Somebody told me to close those tickets and post the question in the general touchscreen section. I hope I did this right.

closed 2020-11-25 by Chad541121