bigtreetech/BIGTREETECH-TouchScreenFirmware #1287: several bugs / feature requests

danorder, 2020-11-25:
encountered bugs / flaws
ram usage is insaine in compiler suggest remove powerloss recovery its redundant.
jerk remains missing /acceleration menu remains incomplete 
M503 doesn't always report via terminal otherwise missing info via esp module 
printing around 200mms and high acceration may result in a nozzle crash 2 carriage plates bent as a result -possibly buffer related? 
icon refresh appears to be patched z on vase mode rapid refreshes result in stuttering presumably other issues may be linked to nozzle crashes / garbled gcode to the mainboard   
print from onboad sdcard allegedly no longer works on older firmware suggest revert or fix not everyone has a 32 bit board... marlin 2.0 is known to use larger libs which is not optimal for melzi board... 
suggest- make stable github fork to get around big trees lack of developer support

oldman4U, 2020-11-25:
Hi.

Could you please check #1170 and #1218 to see which of the topics are already covered and adjust your valuable ticket. Seems also that you have some experience as a developer, so it would be great if you could help us to improve the firmware.

kisslorand, 2020-11-25:
> M503 doesn't always report via terminal otherwise missing info via esp module

Check out [this](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1276) PR.

> icon refresh appears to be patched z on vase mode rapid refreshes result in stuttering presumably other issues may be linked to nozzle crashes / garbled gcode to the mainboard

Check out [this](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1284) PR.

danorder, 2020-11-25:
ill adjust in reference to mentioned tickets.

Bastlwastl84, 2020-11-25:
> encountered bugs / flaws with the tft e3 3.5 drop in display.
> ram usage is insane in compiler suggest remove power loss recovery its redundant.
> jerk remains missing /acceleration menu remains incomplete
> printing around 200mms and high acceleration may result in a nozzle crash 2 carriage plates bent as a result -possibly buffer related?
> print from onboad sdcard allegedly no longer works on display with older marlin firmware suggest revert or fix not everyone has a 32 bit board... marlin 2.0 is known to use larger libs which is not optimal for melzi board...
> suggest- make stable github fork to get around big trees lack of developer support for drop in models.

I needed to format the SD Card (not fast formating fat32 and standard-blocksize and set to 250000 Baud, than i coulded Connect to the sd card onboard from tft

danorder, 2020-11-26:
does that make the icon appear 250000 on latest firmware. prior firmware it
simply disappeared. due to other mentioned issues i have yet to flash the
config.ini with onboard specified vs the autodetect which may be all it is
on my older version. note

oldman4U, 2020-11-29:
@danorder 

I do not really understand your last reply. Could you please explain again?

Thank you

danorder, 2020-11-30:
the onboard sdcard icon disappears in older versions of marlin. 2.05 it is
no longer present. i am not certain if the config.ini needs to be
specified from auto detect or not to make it reappear. on another note i
updated to later 2.0.7 which returned the onboard sdcard option. Prior tft
firmware detected onboard sdcard with marlin 2.05 then stopped detecting i
think on the first version of tft firmware with the 3d preview addon for
cura. allegedly after those changes the tft no longer can read
onboard sdcards if the main board is using older marlin. far from ideal if
that is in fact the case. note i haven't invested much interest in the tft
i updated to 2.07 for other reasons but can confirm the icon is now present
and can print off skr1.3 again without swapping to marlin mode. as for the
mentioned ui issues i think the primary focus should be to have complete
menus / fill in blanks like jerk / junction , fw retract cant use - on the
keypad which requires swap to marlin to use a - retract. , im probs
forgetting other ui qwerks currently i find swapping back fourth makes the
tft borderline useless, the other mention if it fully replaced marlin ui /
others it would only require 1 wire for touch only vs 3 in some cases
depending on the board.

oldman4U, 2020-11-30:
Hi.

There is and there will always be a kind of dependency in case 2 or more systems have to work together. In many situations the TFT firmware has to adjust because of something changed in Marlin.

Missing features are listed as well as known issues. So what is the outcome of this particular ticket?

closed 2021-02-05 by stale[bot]