bigtreetech/BIGTREETECH-TouchScreenFirmware #1289: @CBDesignS I did check, somehow I dont have the "Sequential Mode" appear in the settings. I download the BIGTREETECH-TouchScreenFirmware_peter-master firmware and compiled using VScode then i uploaded the bin together with TFT image folder together with my config file. am I do it wrongly?

CBDesignS, 2020-11-27:
option 1:-
First off have you power cycled (turned the whole printer off for at least 30 seconds) after flashing the update and did it complete with no errors . as in config ini keyword missing errors ?

option 2:
did you rebuild the config.ini from the new build with the settings from your old working build or did you just copy your old config.ini and use it ? . there have been a few additions in newer versions of the tft firmware (that are not in the official btt version of the config.ini file because they have not updated it for months and are behind with the current fixes " yup the owners do not care") if your config.ini does not look like this near the top you will need to use the new config.ini file

#  Note: Ensure that following options are enabled in Marlin firmware
#
#  M115_GEOMETRY_REPORT (in Configuration_adv.h)
#  REPORT_FAN_CHANGE (in Configuration_adv.h)
#  EMERGENCY_PARSER (in Configuration_adv.h)
#  SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)

option 3:
relates to option 2  because you may need to build and flash the latest marlin bugfix or find a build less than a month old to get the above settings and poss to enable sequential mode to show up in the menus. 

Give them a read thru and see if anything may help.

CBDesignS, 2020-11-27:
too late at night to reply and get it wrong....

closed 2020-11-27 by CBDesignS

traffic-light, 2020-11-27:
We are only human after all ;-)