bigtreetech/BIGTREETECH-TouchScreenFirmware #1290: Arab translation[Q] (Your Question?)

yannick-alch, 2020-11-27:
Hi everyone,

I work on a project which will be available in different language, event with arabic language.

I have a SKR V1.3 motherboard with TFT35 V2.0 touchscreen, and I don't find any arabic language.
I have added a language_ae.h file with my arabic translation, but everything goes wrong when I upload it on my screen.

Character are inverted (for example hello become olleh) because arabic is read from right to left.
Other problem, arab language have concat character like "السابق" or "وقفة"
For my screen, to display menu (القائمة), I have this "ة م ئ ا ق ل ا" instead of ةمئاقلا. It is like there was a space between each character, so I decided to ignore every spaces directly from the server, but nothing change.

Why arabe language is not supported on the TFT35 screen ? What can I do to display those characters ?

Thanks in advance !

[image]

oldman4U, 2020-11-27:
Hi.

Could you please check if this helps you further:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/readme/English

yannick-alch, 2020-11-27:
> Hi.
> 
> Could you please check if this helps you further:
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/readme/English

Thanks for the reply.

Yes I already did that. My problem now is the display text. I want to print "لعربية" but I have this "ا ل ع ر ب ي ة", each character are not link between the other.

oldman4U, 2020-11-27:
Ok. I am afraid I can not help. Sorry

HadiJadallah, 2020-11-27:
Hello,

I had to deal with a similar problem back in the day when Adobe Flash did not support RTL and connected glyphs.
Two ways are possible:
1- Implement full RTL/Connected glyph support (medium difficult)
2- Use the unicode of connected glyphs directly (easy by difficult to maintain) -> this way, your "عربية" will be included in the language header file as "0xFECB0xFEAE0xFE910xFEF40xFE94" (https://en.wikipedia.org/wiki/Arabic_script_in_Unicode)

You easily develop a small script that converts arabic text to their unicode form.
Good luck

oldman4U, 2020-11-29:
Hello @yannick-alch 

What is the status of the reported issue?

@HadiJadallah tries to help you, so it would be friendly from you to answer his message.

yannick-alch, 2020-11-30:
> Hello,
> 
> I had to deal with a similar problem back in the day when Adobe Flash did not support RTL and connected glyphs.
> Two ways are possible:
> 1- Implement full RTL/Connected glyph support (medium difficult)
> 2- Use the unicode of connected glyphs directly (easy by difficult to maintain) -> this way, your "عربية" will be included in the language header file as "0xFECB0xFEAE0xFE910xFEF40xFE94" (https://en.wikipedia.org/wiki/Arabic_script_in_Unicode)
> 
> You easily develop a small script that converts arabic text to their unicode form.
> Good luck
> 
> Here is a resource that might be useful to you:
> https://stackoverflow.com/questions/6175053/how-to-convert-arabic-characters-to-unicode-using-php

Hi, thanks, I will try it today and tell you after what I have

yannick-alch, 2020-12-01:
I have test something, but when I want to encode arabic char to unicode, it's not the write format of the char. For the yā, I have 
ي and ﻱ and ﻲ and ﻴ and ﻳ from the wikipedia page [https://en.wikipedia.org/wiki/Arabic_script_in_Unicode](https://en.wikipedia.org/wiki/Arabic_script_in_Unicode)
The encoded char is always the general unicode form. Is their a way to get the right format ?

Thanks

HadiJadallah, 2020-12-01:
Hi,

I think this could help
https://sourceforge.net/projects/ar-php/
the zip file should have a Glyph.php. If my memory serve right, it is what I used and modified to get my encoding correct.

This should achieve what you are looking for and can be a base for full implementation in C/C++ for firmware.

Hope it helps

guruathwal, 2020-12-13:
currently, diacritics are not supported by this firmware.

yannick-alch, 2020-12-14:
I am actually working on this stuff. I will use an array with char, width and height. Then, use this char to write correctly my words.

HadiJadallah, 2020-12-17:
@yannick-alch
I think you are on the write track. Thats what I remember doing back then and it worked flawlessly.
Another option is to get the complete code from the Unicode implementation routines from Linux.

Looking forward to seeing Arabic on the printer screen.

yannick-alch, 2021-02-17:
I have almost finish translation. I juste need tu add the scroll function and normally, everything will work fine  :)

oldman4U, 2021-02-17:
Thats great to hear.

Could you please open a PR once your work is done.

In the meantime it would be great if you could close the ticket.

Thank you

abeksis, 2021-03-19:
@yannick-alch  Were you able to find a solution for the spaces between the letters? I also have the Hebrew language written from right to left has this problem.

closed 2021-05-27 by stale[bot]