bigtreetech/BIGTREETECH-TouchScreenFirmware #1294: TFT35 - Stop Print from Board SD Card - Stuck in "Loading..." Screen

kageryu42, 2020-11-27:
### Description

Hi, I just installed a TFT35 V3 display and an SKR E3 Turbo board for my Ender 5 Plus.  I used Cheetah to compile the software and enabled all option mentioned in the config.ini file.  So, when I start a print from the board's SD card slot and stop it, I get stuck in a "Loading..." screen.  My only option at this point is to either push the reset button or turn the machine off/on.  I've noticed the behavior is a little erratic where sometimes it will not get stuck on this screen.  When it doesn't get stuck it sends me to the screen where you chose the menu type (touch vs analog).  I've noticed there is a yellow dot in the upper right corner of the screen.  I'm not sure if this is a clue or not to what is going on.

This does not happen if I print from the display SD card slot.

### Steps to reproduce

1. Start a print from the board's SD card slot
2. Stop the print

**Expected behavior**
From the way printing behaves when printing from the display SD card slot, I should be sent back to the SD card's menu of items to be printed.

**Actual behavior**
The screen gets stuck in the "Loading..." screen

### Hardware Variant
TFT35 V3
### TFT Firmware Version & Main Board Firmware details

BIGTREE_TFT35_V3.0.26.x 02-Nov-2020 12:01AM - Compiled by radek8
Cheetah Marlin 2.0.7.2

### Additional Information
[config.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5608456/config.zip)
[image]

oldman4U, 2020-11-27:
Hi.

Another user reported similar issues and has been able to fix it by:

I needed to format the SD Card (not fast formating fat32 and standard-blocksize and set to 250000 Baud, than i coulded Connect to the sd card onboard from tft.

Please try this and let us know

kageryu42, 2020-11-27:
I just tried reformatting the SD card.  I removed the tick for a Quick format, confirmed fat32 and set allocation unit size to "Default allocation size".  Unfortunately, the issue is still happening.

It looks like the second suggestion is to change the baud rate to 250000.  Do I do this for the display's firmware or for the board's firmware or both?  Please not that this issue only happens for my board's SD card slot.

Thanks

radek8, 2020-11-27:
This is standard behavior.
If I print from SD tart to TFT and stop printing, printing will be interrupted and I will return to the file selection menu.

However, if I print from the SD card on the board, and the printer is already printing, and I stop printing, the printing will end after the print queue is emptied. Therefore, there is a slight delay.

However, if I stop printing when the bed and nozzle are not heated, the printer will first finish warming up and only then will printing stop and display the file selection menu. Therefore, there may be a delay of several minutes

It is understandable?

radek8, 2020-11-27:
Download the new FW for TFT, several fixes have been merged today.

kageryu42, 2020-11-27:
Ok.  Let me try that and see.

kageryu42, 2020-11-27:
Well, the new firmware has a different behavior.  I start a print, stop it and wait at the "Loading..." screen.  The screen changes to a dialog box stating "Action Command - Nozzle Park" with a Continue button, however, the Continue button is not responsive.  I waited 10 minutes and still could not click the Continue button.

I enabled all of the option listed in the config.ini file for the display's firmware, but I had a question about enabling HOST_ACTION_COMMANDS.  I also enabled HOST_PROMPT_SUPPORT and HOST_START_MENU_ITEM, but maybe that was a mistake.  Is there anyway these are triggering the unresponsive message box?

[image]

kageryu42, 2020-11-27:
It's actually gotten worse with the latest firmware.  I now get the "Loading..." screen whenever I try to access the SD card on the board through the touch screen menus.  I've tried this with multiple SD cards and formatting them multiple times.

kageryu42, 2020-11-27:
Ok.  Here is some more weirdness.  I revered back to the version radek8 compiled for ralfiii back on 02-Nov-2020 and I can get to the print list again through the touch screen.  However, if I cancel a print it goes to the "Loading..." screen, but then the following message pops and the Continue button does not work.  

So, here are the changes I've made so far:
Reverted back to old version of TFT35 firmware
changed baud rate of both the display and board to 250000
commented out HOST_ACTION_COMMANDS in Marlin

I bet I would have the same behavior if I changed the baud rate back to 115200, which I may do.
Any other thoughts?  Everything works great if I use the display SD card slot.  I may have to settle for that right now.  Could this have anything to do with me using the new board SKR E3 Turbo?

[image]

radek8, 2020-11-27:
With the new PR, new problems have emerged that will need to be overcome.

Another problem: after switching to Marlin mode and back to touch, the printer loses its printer configuration. It does not detect Autoleveling, SD card onboard and the like.
  You must send the M115 command manually

kageryu42, 2020-11-27:
Hmmm...How do I manually send that command?  Through an interface like Pronterface?

radek8, 2020-11-27:
yes or via terminal icon

oldman4U, 2020-11-27:
@radek8 Can you see which PR caused this?

kageryu42, 2020-11-27:
Ok.  I've been troubleshooting the issue all day with the latest version of the TFT35 firmware.  After changing baud rates, display firmware and board firmware I think I've finely zeroed in on the culprit and there are two parts to it.  First, I had to disable "#define LONG_FILENAME_HOST_SUPPORT" in the Marlin firmware.  I can consistantly get the "Loading..." screen when I access the list of items to print from the touch menu if this is enabled.  Second, it seems you can't switch SD cards without a reboot of the machine or you get the "Loading..." screen again.

The second part I can sort of understand, but the first I would really like to find a solution for, so I can see the full name of the file.  Any ideas?  Is there anything that needs to be enabled in the config.ini file when flashing the display firmware to get this to work?

Thanks

oldman4U, 2020-11-29:
@kageryu42 

What is the status of the reported issues?

Printing from on board SD card, you can see long filenames when you watch your files before the print starts, but as far as I know there is currently no way to see the long filenames while printing. Please see this ticket which is related to it and which is also a verified bug: #1245

radek8, 2020-11-29:
If I first display the files on the TFT or USB SD card, the sudden opening of the SD onboard works normally.

oldman4U, 2020-11-29:
radek8. This is about the screen - while you print. During the print the file name is shown in the upper bar, but is 8.3 in case you print from on board SD. Using TFT based storage, the correct full name is shown.

Or I understand the ticket wrong.

kageryu42, 2020-11-30:
@oldman4U 
Radek8 has the issue correct.  I was trying to access the contents of the SD card on the board and got stuck in the "Loading..." screen.  I was able to fix this by commenting out "#define LONG_FILENAME_HOST_SUPPORT" in the Marlin board firmware.  I have not been able to try Radek8's observation of accessing the SD card in the display first and then trying to access the SD card on the board as my board had other issues (I saw big temperature swings when I tried to heat up my hotend and bed) and I had to send it back, so right now, I'm board-less.

radek8, 2020-11-30:
@kageryu42 
#define LONG_FILENAME_HOST_SUPPORT
no need to comment. Long names work, but you must first view the contents of the SD TFT.
This is a new bug

oldman4U, 2020-11-30:
I am trying to reproduce and find the reason for this new bug right now.

Big swing could be related to wrong/missing PID tuning of the nozzle. The bed should not swing, with or without PID tuning.

kageryu42, 2020-11-30:
I know, it's strange right.  I tried doing the PID autotune a few times.  I think I may just have a bad board.  It was new (SKR E3 Turbo), and there is a chance a capacitor went bad on it, so I sent it back.  I'll try with the stock silent board (once I get it) and see if I have the same issues.

oldman4U, 2020-11-30:
OK.

Btw, I was not able to see any problems accessing the on board SD card from my TFT35 E3 v3,...;-(

oldman4U, 2020-11-30:
When exactly does it stuck, after selecting it or when accessing the sd card?

kageryu42, 2020-11-30:
When trying to access the sd card on the board from the touch screen menu.  This is where you expect a list of gcode files to be presented for printing.  Instead, I get the "Loading...' screen.  Just to be clear, this is the smaller SD card slot that is on the motherboard, not the larger SD card slot on the display.

oldman4U, 2020-11-30:
Yes, I understand - but can not verify it. Have you tried it with a single code file on the card and maybe also another card? And have you tried what radek explained above?

kageryu42, 2020-11-30:
I tried multiple cards with reformatting them multiple times.  Each time, I only had one code file on the card.  I wish I could try radek's observation, as I think he is on to something, but I'm without my mother board and display right now as I want to go back to my stock configuration for troubleshooting into the temp issue I had.  It will be a few days before I can install the stock board and confirm everything is fine.  From there I might try a new SKR e3 Turbo board or maybe look at the SKR 1.4 Turbo since it has been out longer and is more proven at this point.

oldman4U, 2020-11-30:
So you had this problem with a broken board and you can not test it atm
because you have no board left to test?

kageryu42, 2020-11-30:
Yes, at least I think the board may have been bad.  I will not know until I can test with my stock configuration.  I used Cheetah to compile my firmware for the board, but since it is a brand new board just released, even though Cheetah had support for it, who knows.  It's not really tested at this point.  I think I fell into the trap of having too many new things to troubleshoot an issue (New display, new motherboard, new firmware for motherboard, new firmware for display...)  I need to regroup and take things a little slower.

I don't think the issue with the SD card was because of the board, though, as it worked fine with an older version of the the TFT firmware.  In fact, the older version that worked fine was the version radek compiled for another user in the beginning of November, 02-Nov-2020, I believe.

oldman4U, 2020-12-01:
I tested the reported issue using several versions and they all work the same way. Cancel a print from on board sd card takes a while and printing from TFT works immediately. There is no issue so Please help the community and close the ticket once you do not need it anymore. Thank you

kageryu42, 2020-12-01:
Thanks oldman4U.  Will do.

closed 2020-12-01 by kageryu42