bigtreetech/BIGTREETECH-TouchScreenFirmware #1296: BTT-TFT35-E3-V3.0 filament runout don't beep any alarm and X & Y mottors are lock disabled after a while...

mpcolman, 2020-11-27:
Hi, have two question with the same procedure.....
When filament out, TFT activate "filament runout" and appears the alrt message but without any beep sound like alarm, can I set up a beep alarm with a specified time beeps?? in Marlin mode it's work fine but not with the TFT, have connected the sensor directly to the TFT screen.

And when a certain time passes (don't have certainly time lapse) the motors of the X&Y get unlocked, free moves and lost positions, bed and filament heaters still get on, but the motors don't in lock mode......, in Marlin mode its works fine......
thank for the help

oldman4U, 2020-11-29:
Hi.

I can not really help, because I am not using a filament runout sensor...

@radek8 Do you know if there is a way to define a beep in case of certain events like filament run out and do you know why the motors are turned off in such a case after a while and if it is possible to prevent this?

Would be great if we could help this lost soul after waiting fro an answer for 2 days;-)

Thank you

radek8, 2020-11-29:
I don't know. I don't use a filament sensor

oldman4U, 2020-11-29:
OK, thank you.

I would recommend to search all tickets, open and closed for more information on filament sensor.

mpcolman, 2020-11-30:
@oldman4U  thanks for your concern, I searched but not found about this throuble, I still waiting a few days more, maybe before I create a bug for this...... thank you again!

oldman4U, 2020-11-30:
OK. Please do so.

Chad541121, 2020-11-30:
Im watching this thread also.  I currently have the same question posted.  I was advised to update firmware on the board and the tft.  I am currently trying to figure it out.
[https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1285](url)

mpcolman, 2020-12-01:
@Chad541121 your issue appears at the moment when the filament runout starts or when spend a time waiting for the filament change? it's a t the moment (not in my case) you need to activate the ADVANCED_PAUSE_FEATURE in marlin......

Chad541121, 2020-12-01:
ADVANCED_PAUSE_FEATURE is working!  I want to note that I have the filament runout sensor hooked to the tft35.  I only need to figure out how long the nozzle will stay locked for?  I print at work and may run out of filament when I'm gone for the weekend.

oldman4U, 2020-12-04:
Maybe you can close this ticket, now that you opened the same in the firmware repository.

Thank you

Chad541121, 2020-12-04:
The newest firmware resolved the motors not locking issue for me.

closed 2020-12-10 by mpcolman

mpcolman, 2020-12-10:
tested again with the last published firmware, after about 15 mins paused in filament out mode and tried to move the head and it moves freely.... not resolved.....

oldman4U, 2020-12-10:
@Chad541121 

Latest Marlin or TFT firmware solved the problem for you?

Chad541121, 2020-12-10:
It was the latest marlin firmware that solved my issue

oldman4U, 2020-12-10:
Thank you

Mpc. What is the Marlin version you are using?

mpcolman, 2020-12-10:
@Chad541121  you save my day! 
I change to "0" the  DEFAULT_STEPPER_DEACTIVE_TIME, and "false" for  DISABLE_INACTIVE ,and all work fine, locked motors are fine right now....
Pending the beep alarm for  when activate the filament runout.....
@oldman4U  you think that I close this issue and open another one for the filament runou beep exclusively??

oldman4U, 2020-12-11:
Yes. Close this and start a new one.

👍🏻

oldman4U, 2020-12-15:
@mpcolman Please close also the ticket in TFT Firmware: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1338