bigtreetech/BIGTREETECH-TouchScreenFirmware #1297: [BUG] Today's merger of PR has created new problems.

radek8, 2020-11-27:
Today's merger of PR has created new problems.

If the map has a connected display, after turning on or restarting the printer, I sometimes run the command M1 - pause, which is waiting for confirmation.
This is due to the fact that after switching on, the command M501 is sent, which is damaged and changed to M1, see the picture.
Further communication is without errors ....

[image]

[image]

[image]

Another problem is that when switching to marlin mode and back, the display loses configuration information and therefore SD on board, autoleveling and more are not available. the configuration must be restored with command M115

digant73, 2020-11-27:
The lost machine settings switching from marlin to touch mode is due to PR 1244 that moved the initMachineSettings funtion from main.c to mode.c

guruathwal, 2020-11-28:
It looks like the commands are getting mixed. can you try with reverting changing from this PR https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1276?

guruathwal, 2020-11-28:
The `initMachineSettings` should be called only once at boot time only.

Bastlwastl84, 2020-11-28:
I'm having also different Problems now with New FW from TFT (will create a new Ticket) Layer 1, Layer High 0,16 shows me 2,00mm and if i click on preheat bed only preheat PLA its starting tonpreheat and jumps lo load/unload extruder window. If i'm canceling print it unloads the filament completely.

kageryu42, 2020-11-28:
Hi, I don't know if you saw my last post in Issue #1294 that I reported, but for the issue where the SD card on the board is not available from the touch mode, if you disable "#define LONG_FILENAME_HOST_SUPPORT" in the Marlin firmware, the SD card becomes available again, at least for me.  I think this is a big clue into what is going on here, but more importantly, can give users a work around while a fix is being developed.  The only other thing I noticed with the SD card on the board, and maybe this is expected behavior, is that if you switch SD cards without restarting the machine, it freezes on the screen Loading... when you try to access the SD card from touch mode.

radek8, 2020-11-28:
New finding
I click on the SDBoard icon, the screen freezes on the Loadin ...
If, however, I first open the list of files on the TFT SD card, return, and then open the files on the SDBoard, the files will be displayed normally.

radek8, 2020-11-28:
The problem is loading long names
The screen freezes before loading long names.

[image]

If I open the SD TFT first, the long names will be loaded correctly and the display will not freeze

[image]

digant73, 2020-11-29:
@radek8 
try to disable long file support and verify if there is no freeze. If so, probably the problem is on mygcodef.c changed by PR 1244

radek8, 2020-11-29:
> @radek8
> try to disable long file support from TFT feature (or config.ini) and verify if there is no freeze. If so, probably the problem is on mygcodef.c changed by PR 1244

Even if I turn off long name support on TFT, the behavior is still the same.

oldman4U, 2020-11-29:
Hi Radek.

What is the status of the reported issues?

1. damaged M105 command and restart or reset
2. switching from Marlin to TFT, all information is lost (seems to be fixed with #1305, right)?

Please let me know

Thank you

radek8, 2020-11-29:
1. Probably interference on the cable, I'll try a shielded cable.
2. Yes PR1305

oldman4U, 2020-11-29:
OK, thank you.

bigtreetech, 2020-11-30:
Hello, Can you confirm that #1305 is effective?  In fact, I can't reproduce your problem. 
And in theory, initMachineSetting() will reset the parameters and the touch screen will automatically resend the gcode
M503, M92,M115,M211.
I will merge #1305 immediately If you can confirm.

oldman4U, 2020-11-30:
Hi BTT.

#1304 and #1315 would be great to be merged.

Thank you

radek8, 2020-11-30:
> Hello, Can you confirm that #1305 is effective? In fact, I can't reproduce your problem.
> And in theory, initMachineSetting() will reset the parameters and the touch screen will automatically resend the gcode
> M503, M92,M115,M211.
> I will merge #1305 immediately If you can confirm.

Yes, PR #1305 solves the problem of losing printer parameters when switching from Malin mode to touch mode.

radek8, 2020-11-30:
The problem with loading SD on board persists.
If I don't show SD TFT first, SD on board will freeze on Loaing ...
If I open the TFT CD, then the SD on board works.

oldman4U, 2020-11-30:
Thank you BTT for merging the two PR's.

Radek I have some time today, so I can help you with the SD Card issue. Just checked and I am using a firmware from nov. 15th at the moment and everything works fine. I can select a file from the on board SD card directly after power on, without selecting another storage device first.

I will check the fw which is available right now with the two PR from above and let you know.

oldman4U, 2020-11-30:
Somehow my printer is not good for testing.

I can not reproduce the issue with the on board SD card using SOR E3 DIP mainboard Marlin 2.0.7.2 and TFT 35 E3 v3 with FW from today. I was able to select the on board sd card without the need to select another storage before.

radek8, 2020-11-30:
Oldman if you can't reproduce the same error with an SD card, with the same FW in TFT, there will be a difference in Marlin
Can you send the marlin configuration? I'll compare with my configuration what's the difference.

oldman4U, 2020-11-30:
nono. I can reproduce it now.... SORRY!

Does not matter if the sd card is already inserted at startup or not, accessing it as the first possible "storage" device brings up the LOAD... forever. Could you please try to find out which PR caused this?

Thank you

radek8, 2020-11-30:
demonstration of SD card behavior on board
https://youtu.be/tR6j0eAkBAo

digant73, 2020-11-30:
@radek try to revert pr 1272. It changed m27 polling interval. Maybe it causes the freeze. You should see the polling from the terminal. However I don't have that problem

radek8, 2020-11-30:
I have another finding.
I compiled FW with PR1272 and SD onboard card is available.
after adding PR # 1276 the display freezes when trying to open the SD on board card. See the video above.
@kisslorand  do you have any idea why this happened?

kisslorand, 2020-11-30:
Please edit you messages so PR numbers link to the respective PR.
Like PR #1272 and PR #1276.
I do not use SD onboard so I cannot test anything with SD onboard. Might be a RAM issue, check with lowered terminal buffer.
In SendGcode.c line 6 replace 
` #define TERMINAL_MAX_CHAR ((LCD_WIDTH / BYTE_WIDTH) * (LCD_HEIGHT / BYTE_HEIGHT) *8 )`
with
` #define TERMINAL_MAX_CHAR ((LCD_WIDTH / BYTE_WIDTH) * (LCD_HEIGHT / BYTE_HEIGHT) * 6)`

test and report pls.

digant73, 2020-11-30:
Just for a correct information, It was a Heap/Stack issue not a RAM issue. However in a new PR, line 300 on parseAck.c must be removed. It sets m27 auto report even during no print.You can see the periodic reports on terminal menu.
Also L2 buffer was now reduced to 512 bytes but L1 is copied in L2 without any check on L2 buffer overflow until \n is found. You must verify if maybe this is what it is happining on some printers. Are all the file names on onboard sd provided in a single line exceeding 512 bytes? Did you try to put on the sd just few files and verify if freeze is still present?

kisslorand, 2020-11-30:
All filenames are presented in a new line, so there's an "\n" after each filename. There is no single message from Marlin that can exceed 512 bytes.
Anyhow L2 cannot overflow, it is a circular buffer.

radek8, 2020-11-30:
So I tested.
I found 3 combinations when the SD onboard card loaded well.

1combination
#define DMA_TRANS_LEN 2048
#define TERMINAL_MAX_CHAR ((LCD_WIDTH / BYTE_WIDTH) * (LCD_HEIGHT / BYTE_HEIGHT) * 6)
Unfortunately, in the terminal, the M503 command reads only 4 lines.

2nd combination
#define DMA_TRANS_LEN 3072
#define TERMINAL_MAX_CHAR ((LCD_WIDTH / BYTE_WIDTH) * (LCD_HEIGHT / BYTE_HEIGHT) * 4)
Unfortunately, in the terminal, the M503 command is loaded with only 1 screen,
M503 S0 loads 6 screens

3combinations
#define DMA_TRANS_LEN 1024
#define TERMINAL_MAX_CHAR ((LCD_WIDTH / BYTE_WIDTH) * (LCD_HEIGHT / BYTE_HEIGHT) * 8)
Unfortunately, in the terminal, the M503 command reads only 4 lines.

radek8, 2020-11-30:
Other findings.
The mentioned problem with SD onboard card and current FW
only occurs when there are more than 6 files to print on the card.
If there are fewer files, the files will be loaded for the first time.

It's crazy behavior.

oldman4U, 2020-11-30:
I finally saw the issue with 1 and 2 files  on the sd card.

kisslorand, 2020-11-30:
I still think it's a RAM issue. Filesystems need a lot of RAM.

oldman4U, 2020-11-30:
Why does it work when you select another storage device before?

kisslorand, 2020-11-30:
I really don't know. I do no t have SD Onboard so cannot test this issue. On my side everything works fine (no onboard SDCard).

digant73, 2020-11-30:
It is L1 handled as a circular queue in function SynchL2cacheFromL1 in parseAck.c

kisslorand, 2020-11-30:
???

blueeagle69, 2020-12-01:
Oh joy.
I updated my TFT firmware yesterday.
And now when selecting print from onboard SD I get the dreaded black screen with Loading message.
Which of course I got the gist of from reading this thread 😀
Oh well, back to Marlin mode I go until it's fixed!

digant73, 2020-12-01:
@kisslorand 
on BTT TFT35 you have at least 4 serial port used, so (with a 4K buffer size) a minimum of 16K will be allocated on the heap. Adding other dynamically allocated data here and there, the system goes out of heap space.
You (and me) don't have the problem because our TFT uses 3 serial ports. If from the total amount of RAM (48K) you also remove the stack space (a minimum of 4K on BTT TFT35) there is even less space reserved to the heap

oldman4U, 2020-12-01:
OK. I have been testing a lot with my TFT35 E3 v3 based printer and I would like to add @mehmetsutas to this ticket. Seems that his code brought a great new feature, but also caused some new issues. In general I can see 3 issues:

1. Layer Info while printing from on board sd, and TFT based storage
build from Nov. 18.: **No layer information while printing from on board SD**, no problem while printing from TFT
build incl #1272: Layer Info is shown while printing from on board SD, **RANDOM Layer update while printing from TFT**
build incl all PR from Nov. 30th: **NO Layer Info is shown while printing from on board SD, **RANDOM Layer update while printing from TFT which can be updated by leaving the main screen and going back****

2. Behaviour after a print has finished
build from Nov. 18.: ALL fine, does not matter which storage has been used
build incl #1272 and build incl all PR from Nov. 30th: Fine when printing from on board SD, printing from TFT based jumps back to main screen. For me this is the better method, but I understand that this is maybe different for other users, so it would be good to have this optional I believe.

3. Accessing the on board sd storage
build from Nov. 18 and build incl #1272: No problem at all, all storage option can be access at any time
build incl all PR from Nov. 30th: Accessing the on board SD card right after turning the printer on gives a Loading ... forever. Accessing another storage before accessing the on board SD card solves the problem. This is independent of the number of files and happened also with a single very very small file (less than 400kbytes).

Or in terms of issues introduced by the PR's:
#1272: 
RANDOM Layer update while printing from TFT
Printing from TFT based jumps back to main screen. It would be good to have this optional.

Latest firmware:
Accessing the on board SD card right after turning the printer on gives a Loading ... forever.

I have also seen issues where the filename on the storage was destroyed after a print and also that the print time was not reset after a print, but I am not able to reproduce them at the moment. 

@mehmetsutas 
I believe THE best thing after a print would be a finished print window which has all the features in a new window. The design is on my desk, would you be interested to develop this for the community? Please let me know, thank you.

oldman4U, 2020-12-01:
@kisslorand 
I would test it by myself, but the building process seems to be broken.
Does #1319 fix the layer update for printing from all 3 methods?

mehmetsutas, 2020-12-01:
@oldman4U about the layer issue I will create a new PR. I will keep the old logic for TFT SD and TFT USB prints and use the new logic only for BOARD SD prints. I think this will solve the issue.

About the new screen please send it to msutas@gmail.com. Let me have a look If I can manage and if there is enough space for new icons on TFT, I will do it.

kisslorand, 2020-12-01:
Get the master from my repo, it's not broken (yet). It's one merge behind the BTT master. It compiles just fine.
You have a point, it seems the layer is updated differently based on the gcode source. My fix refers only to the print made from TFT SD or TFT USB. In an hour or so I will go over PR #1319 and try to make it work in all situation. 
I cannot test SD Onboard but hopefully someone will try it and give feedback.

blueeagle69, 2020-12-01:
Jut an update.
I just tried #1319 and selecting the onboard SD is still broken.

kisslorand, 2020-12-01:
#1319 is there for updating the layer during print from onboard SD, not for the onboard SD browsing bug.

blueeagle69, 2020-12-01:
> #1319 is there for updating the layer during print from onboard SD, not for the onboard SD browsing bug.

Okay, I'll revert to a working version.

Thanks

radek8, 2020-12-01:
The integrated card works if you open the SD card to TFT first

blueeagle69, 2020-12-01:
Cheers @radek8 I'll do that for now.

kisslorand, 2020-12-01:
@digant73 
> on BTT TFT35 you have at least 4 serial port used, so (with a 4K buffer size) a minimum of 16K will be allocated on the heap. Adding other dynamically allocated data here and there, the system goes out of heap space.
> You (and me) don't have the problem because our TFT uses 3 serial ports. If from the total amount of RAM (48K) you also remove the stack space (a minimum of 4K on BTT TFT35) there is even less space reserved to the heap (a minimum of 20K on BTT TFT35)

Thanks for the 48K hint, I think I am on something.

@oldman4U 
> 3. Accessing the on board sd storage
>    build from Nov. 18 and build incl #1272: No problem at all, all storage option can be access at any time
>    build incl all PR from Nov. 30th: Accessing the on [...]

I think I found something.

oldman4U, 2020-12-02:
>    1. Accessing the on board sd storage

👍🏻

blueeagle69, 2020-12-02:
@kisslorand Thanks for your  hard work.

PR #1468 (2021-01-07): save/load infoSettings Struct instead of save/load every variable individually, Fix Screen freezes/Click to Resume message/damaged M105 command in Marlin mode
- The settings are stored in the last 2KB of the program memory. earlier a separate entry was used to save/ load each variable. With this the **infoSettings** Struct is copied at once to/from buffer instead of saving/loading every variable individually, removing the need to add save/load entries for each variable. This also eliminates bug related to wrong/invalid settings being loaded. The prebuild settings size check now works accuratel [... cut]
closed 2020-12-02 by radek8