bigtreetech/BIGTREETECH-TouchScreenFirmware #1298: [FR]  the button terminalACK

radek8, 2020-11-27:
@digant73 your modification of the menuTerminal is great.
Could it be further improved with a button that would turn "terminalACK" on and off directly from the terminal environment?
Thank you

kisslorand, 2020-11-29:
What would be the benefit of it?

radek8, 2020-11-29:
Directly in the terminal window I can turn on and off the communication of the display with the printer without having to go to the menu of printer parameters

oldman4U, 2020-12-02:
Should we add this to the FR list?

radek8, 2020-12-02:
If no one else wants it, you don't have to.
It's not that important.

oldman4U, 2020-12-02:
YOU want it and you do a lot for this project. So I add it and hopefully it will be available one day. 👍🏻

radek8, 2020-12-02:
:-)

closed 2020-12-02 by radek8

oldman4U, 2021-01-25:
Thank you @digant73

Please check PR #1563

PR #1563 (2021-01-30): Added new param in FeatureSettings menu + toggle ACK in Terminal menu…
**NEW FEATURES:**
1) Added in FeatureSettings menu the enumerated parameters defined in config.ini such as:
 * enabled, disabled
 * enabled, disabled, auto
 * etc...
 The purpose of this addition is to allow to set some settings without the need to use so frequently the config.ini.
2) Added in Terminal menu a button to toggles the terminal ACK. The changed value is not made permanent on flash (it's meant just as a temporary setting change) . This means that in order to make the change permanent on falsh the usual FeatureSet [... cut]