bigtreetech/BIGTREETECH-TouchScreenFirmware #1300: [BUG] open Extruder Menue after Clicking on Preheat Bed TFT35 E3 V3

Bastlwastl84, 2020-11-28:
clicking on preheat Bed opens Extruder Menue
[Configuration.h TFT35E3V3.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5610674/Configuration.h.TFT35E3V3.txt)
[platformio.ini TFT35E3V3.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5610676/platformio.ini.TFT35E3V3.txt)
Video:
https://drive.google.com/file/d/1AFLJ0R-qH5XDzpzdTQB7rrH-FDbx7M5g/view?usp=sharing

Also happens that sometimes the TFT does a long Beep (happens at 115200 Baud also) after clicking at a Button till i click again at any other Button

Time is incorrect  and Layer. Did a Printjob before and it showes 2mm at Layer 1 (0,16mm Layerhigh). Print starts at USB_Stick at TFT. Sometimes the Display jumpes to correct high 

[image]

kisslorand, 2020-11-29:
Check PR #1304. It might help you.

oldman4U, 2020-11-29:
Hello @Bastlwastl84 .

What is the status of the reported issues?

1: Extruder Menu when clicking Preheat button - should be fixed with #1304 - THX @kisslorand 
2. long beep - is this a new issue or have you seen this also with a previous version?
3. Time is incorrect? Is this a new issue or.... What exactly is wrong with it? It should show the elapsed time.
4. Layer hight. Are you sure it does not show the layer hight at a Z hop retract?

More information is needed, so please let us know.

Thank you

oldman4U, 2020-11-30:
Regarding 4. The layer  hight (when printing from TFT SD or uDisk) is only updating "by mistake" every few layers. Leaving the main screen and going back refresh the Layer hight to the correct value temporarily.

Bastlwastl84, 2020-11-30:
> Hello @Bastlwastl84 .
>
> What is the status of the reported issues?
>
> 1: Extruder Menu when clicking Preheat button - should be fixed with #1304 - THX @kisslorand
> 2. long beep - is this a new issue or [...]

Z-hop was off.
Time: it has shown actual 1:30h at beginning. Was printing a part before.
Long beeps without stipping till next Interactions i had at old version also. Waiting till new version an have done a downgrade yet, cause of all the bugs.

panlamb81, 2020-12-02:
good morning
I have the same problem.

kisslorand, 2020-12-02:
It's already fixed.

oldman4U, 2020-12-02:
Well. This ticket is about 3 issues. So which one do you mean?

It is important to avoid such additional comments by closing unneeded tickets asap.

panlamb81, 2020-12-02:
open Extruder Menu after Clicking on Preheat Bed

oldman4U, 2020-12-02:
This issue is fixed in #1304 which is available as part of the master firmware.

Maybe you want to wait a day or so, in case you are printing from on board SD card. There is a fix on the way for this.

Happy printing

oldman4U, 2020-12-03:
All PRs are in master. Please test the firmware and update this ticket accordingly. 

Thank you

panlamb81, 2020-12-03:
I went through the last upgrade and the problem was fixed.
Everything works fine
Thanks

oldman4U, 2020-12-04:
Not sure you know it, but there is more to come with this PR: https://github.com/mehmetsutas/BIGTREETECH-TouchScreenFirmware

Fixes a bug with the print time not always cleared, brings a new print finished window which includes the elapsed time and filament use for all storage devices and the landing page is now the storage device in case you cancel a print or the main page in case the print finished properly.

Hope you like it, all credits go to @mehmetsutas

oldman4U, 2020-12-06:
@Bastlwastl84 

What is the status of the reported issues?

Bastlwastl84, 2020-12-07:
Long Beep indidn't had since last update at friday, but didn't checked if layerhigh was correctly. I clicked at print and was checking with esp cam if all right.     Having new issues with freezes when booting the esp32-cam connected at UART-Port RX3 TX3 and Power from Display. 2k and 1k ressistors between like at luc-Githubs Wiki. The esp3d v3.0 branch. It does not freeze when booting in Marlin Mode than switching to Touch. Could that be a Problem with the to low RAM usage like at a PR discriptioned?

oldman4U, 2020-12-07:
This ticket is about: open Extruder Menu after Clicking on Preheat Bed, nonstopping beep and layerhigh TFT35 E3 V3

Start a new ticket in case of a new issue.

oldman4U, 2020-12-10:
All 3 issues should be fixed using master firmware from today. Please check and close this ticket in case all reported issues are solved.

Thank you

closed 2020-12-12 by Bastlwastl84