bigtreetech/BIGTREETECH-TouchScreenFirmware #1302: Menus not working properly

slwise, 2020-11-28:
Menus are not working properly.

Installed the latest firmware for the TFT35 E3 V3.0. The update went beautifully, no errors or warnings. I went to change filament and preheated the nozzle by going to the preheat menu. That worked fine. Once heated, I selected the extruder menu to unload the filament. It took me to the preheat screen. Ok, so I selected the correct preheat setting and it took me to an extruder screen with only the unload, load and nozzle buttons were displayed. When I pressed the unload button the extruder didn't move but the z-axis moved up about 10mm and stopped. I tried reinstalling the firmware with the same results. I also compiled a version myself using PlatformIO with the same results. I went back to a previous release and everything is working as it should so it has to be something in the current release.

kisslorand, 2020-11-28:
It's not a bug, it's an intended function. Your Z moved up and the printer was waiting to reach the selected preheat temperature. Once that temperature reached, filament would load or unload based on your previous selection.

oldman4U, 2020-11-28:
Thank you. This and some more bugs have been reported already. Please see the tickets from earlier today.

oldman4U, 2020-11-28:
@kisslorand 

For all direct extruders this is a bit of a strange behavior. Could you please make the extrude function optional?

kisslorand, 2020-11-28:
Yes, I can restore its (Extrusion icon) original functionality and make the the "Un/Load" function a separate one in the Move menu, where now it is the Extrusion moved. I could even add a "Preheat" and a "Heat" option there.
Ideas? Opinions?

The "Un/Load" automatic menu must go from the Preheat menu! It pops up right after you select some preheat option.

oldman4U, 2020-11-28:
I BELIEVE that a separate load/unload would be the right way to do it and this should always include the option to heat the nozzle.

The same time I can not really see the need to (un)load filament after the preheat but it would be great if this could be activated optionally.

My 2 cents

kisslorand, 2020-11-29:
PR #1304 is up, I hope it suits everyone.

oldman4U, 2020-11-29:
@slwise 
Do you know how to compile firmware using VSC and PIO? Have you tested kisslorands PR?

Please let us know

Thank you

oldman4U, 2020-11-30:
The fix is merged in the meantime.

Please download the new firmware with the fix and close this ticket afterwards.

Thank you

slwise, 2020-11-30:
Yes, I can try it this evening. I have a print running right now. I'll close the ticket then. Thanks to everyone.

oldman4U, 2020-11-30:
👍🏻

But there are some other bugs which should be fixed the next few days.

kisslorand, 2020-11-30:
PR #1317 is up, adds some issue fixing to PR #1304.

slwise, 2020-11-30:
I just compiled with fixes and everything looks and works great. This is definitely better. Thanks guys for the quick action and your help.

closed 2020-11-30 by slwise