bigtreetech/BIGTREETECH-TouchScreenFirmware #1303: [BUG] The new Unload'Load funtion

kisslorand, 2020-11-28:
### Description

I opened this bug report to be able to discuss how it should be handled

### Steps to reproduce

1.  Press Extrude in the main menu
2.  Preheat automatically appears
3.  After selecting a preheat value the Load/Unload menu appears

or

1. Select Menu -> Heat
2.  Select "Preheat"
3. Select a a preheat option

**Expected behavior**

Extrude should go to Extrude menu, not preheat and than Load/Unload.
After a regular Preheat selection, it should not jump right away to the Load/Unload menu

**Actual behavior**

Main->Extrude goes to Preheat menu than jumps to Load/Unload menu
Selecting an option in Preheat menu jumps right to Load/Unload menu. If you selected a wrong preheat option there's no turning back.

### Additional Information

I agree a preheat is needed before load or unload but Marlin takes care of that (M701, M702). It's true it preheats the nozzle to a fixed value (in my case to 180 degree Celsius). However it is not logical to preheat both the nozzle and the bed for a load or unload.
As for the regular extrude, it is the user's duty to preheat the nozzle, it should be the same for load/unload.
Load/unload feature is a welcome feature but I do not think it has a greater importance than the regular Extrude menu, hence I do not see justified its presence in the main menu in the detriment of the Extrude menu, it being  buried deeper in the menu structure.

kisslorand, 2020-11-29:
PR #1304 is there for this bug.

kisslorand, 2020-11-30:
I found some new issues since PR #1304 was merged, so there will be a new PR. Stand by...
If Load/Unload procedure was initiated, the user can click around on the TFT. Since the printer is in a waiting state (executing M702/M701), it will freeze the TFT eventually until M701.M702 is finished. This can confuse a regular user.

kisslorand, 2020-11-30:
PR #1317 is there for the latest issues.

mehmetsutas, 2020-12-01:
Actually these are all personal flavor. I don't mind how it is handled. This firmware is used by many people and I can't force my preferences to the majority. I can run my machine according to my preferences. But I would like to share my thougts.

In my opinion extruder is the 4th axis of the machine and keeping it under the move menu is more logical as in Marlin's menu structure. I just used the extrude icon for filament load/unload since I lack the necessary skills for creating a new icon and that icon seemed reasonable for representing the load/unload operation.

I personally only use the preheat option for loading and unloading filament and it is easier to just press a buton for the right filament and afterwards choosing whether to load, unload or exit. As the preheat menu have a wide range of choices, I believe it is more practical to select a preconfigured temperature value with one push rather than each time typing in the right temperature.

I almost never use the extrude option. I use it very rarely in the case of a filament jam.  That's why I found it more advantageous to move the rarely used operation under a submenu and to put a frequently used load/unload operation on the main menu.

Also putting the extrude operation under move menu, placing the filament load/underload in the main menu, having the filament load operation after a preheat selection makes the user interface logically parallel to the Marlin's built-in user interface.

kisslorand, 2020-12-01:
You have a point here, and your addition of Load/Unload menu is a great welcome. I also agree that you tailored it to your personal  workflow. 
However it broke the flow of the whole menu system users are used to.  What if someone just wanted a preheat? In your case it jumped right to the Load/Unload menu.  After pressing Load or Unload the menu jumped to start. After an Unload the majority of users would want a Load. And so on...
I made a [PR](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1317) about this, just wanted to filter out unusual behaviours and tailor it to the regular Menu workflow/usage/behaviour.
I also agree that Load/Unload menu and Extrusion menu should be swapped since Load/Unload is a procedure and Extrusion deals with the 4th axis.
I also have many mods to my menu system that I do not publish because they are just my personal preferences and might seem illogical  to the way the original menu system is built.

oldman4U, 2020-12-04:
@kisslorand 

Seems the PR is merged. Could you please be so kind and close this ticket, once you do not need it anymore.

Thank you

closed 2020-12-04 by kisslorand