bigtreetech/BIGTREETECH-TouchScreenFirmware #1306: TFT70 V3.0 Touch not working

steirerman, 2020-11-29:
TFT70 V3.0 the touch function does not work for me and the font is moved. Any Idea ? The software is current version.
[image]

traffic-light, 2020-11-29:
Did you try this:

Q: Touchscreen doesn’t respond/ stay’s in the “Choose mode” selection

    Answer: .......

    b) The screen doesn't respond to any touch
    The touchscreen settings could be corrupted. The screen doesn't know at this moment how to correctly respond to a touch.
    A way to solve this:
    Place a text file reset.txt in the root of a SD-card and insert this in the screen
    Reset the screen.
    When the screen is booted there will be a touchscreen callibration screen. Calibrate the screen and it should be solved.

See: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1047

traffic-light, 2020-11-29:
Did you update the fonts after updating the screen?

steirerman, 2020-11-29:
THANKS !! Working :)

traffic-light, 2020-11-29:
That's good to hear 😉 

Next time you could first check the readme for faster fixes

Please close the ticket when it's solved 😉

steirerman, 2020-11-29:
Many thanks for the support

closed 2020-11-29 by steirerman