bigtreetech/BIGTREETECH-TouchScreenFirmware #1307: [BUG] TFT35 stucks on loading from onboard SD card

Davidoff77test, 2020-11-29:
On TFT35 E3 V3 I want to print from onboard SD card in touch mode. But if SD card contains some .gcode files display stucks on loading... message. From Marlin Mode it works.

### Steps to reproduce

1. Select Print->On Board SD Card
2. Select On Board SD Card

**Expected behavior**
I can see list of gcode files.

**Actual behavior**
Display stucks with Loading... message

### Hardware Variant
TFT35 E3 V3
### TFT Firmware Version & Main Board Firmware details
Marlin: 2.0.7.2
TFT fw: commit 198f14e from master branch

### Additional Information
[config.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5612014/config.zip)

radek8, 2020-11-29:
This is a duplicate ticket.
please see # 1294 and # 1297

oldman4U, 2020-11-29:
@Davidoff77test 

Please help the community and close this ticket.

Thank you

closed 2020-11-30 by Davidoff77test

kisslorand, 2020-12-01:
Check PR #1326.