bigtreetech/BIGTREETECH-TouchScreenFirmware #1308: weird extrude menu open up when clicked on any preheat button and back button kicks you out to main screen

carryreynolds, 2020-11-29:
### Description
The extrude menu opens the preheat menu and the preheat button is opening a weird extrude menu which looks incomplete.
and then clicking on the back button takes you to the main screen. 
This firmware now feels like a badly patched sailboat. instead of making things easier to access it is making things more complicated.

### Steps to reproduce

open extrude menu in the main menu and click any preheat button.

**Expected behaviour**
open extrude menu

### Hardware Variant
TFT35-V2
### TFT Firmware Version & Main Board Firmware details
TFT35_V2.0.26.X Nov 29 2020

@bigtreetech Why are new features/codes being added without testing and any supervision? currently, there is no one supervising this repository. Someone like @guruathwal who is familiar with this firmware should be appointed to oversee the changes and development in this repository to make sure proper development in this repository and bugs like these are avoided.

traffic-light, 2020-11-29:
Did you also update the icons&fonts and the config.ini?

traffic-light, 2020-11-29:
And did you compile the firmware on your own?

carryreynolds, 2020-11-29:
yes all fonts, icons, config flashed and compiled the firmware myself.

kisslorand, 2020-11-29:
Check PR #1304.

oldman4U, 2020-11-30:
@carryreynolds 
This issue is solved.

Please check with latest firmware and close this ticket.

Thank you

oldman4U, 2020-12-03:
@carryreynolds 

You opened this ticket. Could you please test the latest master firmware and update the ticket accordingly.

Thank you

oldman4U, 2020-12-04:
9 day wonder.

It is very unfriendly not to answer.

Please close the ticket

carryreynolds, 2020-12-15:
I am sorry for not replying. Not everyone enjoys free internet.

closed 2020-12-15 by carryreynolds