bigtreetech/BIGTREETECH-TouchScreenFirmware #1309: Is it possible to delete duplicate menus?

AntoszHUN, 2020-11-29:
Hi!

Wondering if you can remove duplicate menus.?

Among other things, I am thinking here of disappearing the "Terminal" and "Custom" menus. These can also be found in the main menu as well as in the "Main menu - Options - Machine -" menu. I was thinking about this because maybe memory could be freed up if we removed one of the locations. Is it possible?

Hello, AntoszHUN

traffic-light, 2020-11-29:
You could do it on your own ;-)

I believe there was a issue explaining how to do this

AntoszHUN, 2020-11-29:
> 
> 
> You could do it on your own ;-)
> 
> I believe there was a issue explaining how to do this

I'm sorry. I didn't know it was about it. Maybe I can get information about it, how is it?

oldman4U, 2020-11-29:
Jonapot.

It is even in the FR list: #1053

Please close the ticket once you do not need it anymore.

Thank you

AntoszHUN, 2020-11-30:
Super! Thanks! I'm closing.

closed 2020-11-30 by AntoszHUN