bigtreetech/BIGTREETECH-TouchScreenFirmware #1310: Persistent "No Printer Attached!" Message, Unable to Control Printer (TFT35 E3 v3.0, SKR mini E3 v2.0

ThePootisMann, 2020-11-29:
Hello everyone. Yesterday my SKR mini e3 v2.0 and TFT35 E3 v3.0 arrived from Amazon today, and I was quite excited to get it set up and start printing. However, after I put everything together, I received the "No Printer Attached!" message. From thereon, I have tried exactly these things to try and fix the problem, but to no avail:

Fixing/Moving serial connector on the board and monitor

Updating the firmware on the TFT35 E3 v3.0

Updating the firmware on the SKR mini e3 v2.0 (modified in PlatformIO, changing serial port values)

Changing the Baudrate in the firmware and on the TFT35

Modifying the TFT35 firmware in PlatformIO

Attempting to use Marlin Mode (which resulted in the "LCD12864 Simulator" Message)\

I truly feel as if I am running out of options. I have searched for threads all over the internet relating to this issue and trying out their solutions, but none have worked so far. Could you please help me determine a solution to the problem? Thank you.

radek8, 2020-11-29:
Send a photo of how it's connected.
Attach configuration files

ThePootisMann, 2020-11-29:
[image]
[image]
How do I upload my configuration files? When I try to do it, it says the file format is unsupported.

ThePootisMann, 2020-11-29:
> Send a photo of how it's connected.
> Attach configuration files

ThePootisMann, 2020-11-29:
I closed it by mistake.

radek8, 2020-11-29:
use zip

radek8, 2020-11-29:
incorrectly connected EXP3 connector

[image]

radek8, 2020-11-29:
Send a better photo of this connector
[image]

ThePootisMann, 2020-11-29:
[image]
[Marlin-2.0.7.2-SKR-mini-E3-V2.0.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5612700/Marlin-2.0.7.2-SKR-mini-E3-V2.0.zip)

ThePootisMann, 2020-11-29:
[config.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5612704/config.zip)

radek8, 2020-11-29:
the photo is blurred

ThePootisMann, 2020-11-29:
[image]

radek8, 2020-11-29:
OK the serial cable is OK.
Did you reconnect the EXP3 cable?

You have a badly defined serial port in configuration.h.
-1 is a USB connector
2 is a serial port for TFT
 
I recommend a change
#define SERIAL_PORT 2
#define SERIAL_PORT_2 -1

ThePootisMann, 2020-11-29:
I did in fact reconnect the EXP3. I will change those values.

radek8, 2020-11-29:
you do not have some important parameters enabled in marlin
read the comment in config.ini

[image]

ThePootisMann, 2020-11-29:
I have just enabled them.

ThePootisMann, 2020-11-29:
Marlin Mode now appears to be working. However, I cannot say the same for the touchscreen mode.

radek8, 2020-11-29:
did you fix the serial port parameter?
#define SERIAL_PORT 2
#define SERIAL_PORT_2 -1

ThePootisMann, 2020-11-29:
I did.

radek8, 2020-11-29:
If you have the same speed set, touch mode should work.

ThePootisMann, 2020-11-29:
I have set the Baudrate in both the SKR and TFT to 115200. Touch mode still refuses to work.

radek8, 2020-11-29:
It responds to touch, but writes the printer is not connected?

radek8, 2020-11-29:
If the printer writes not connected, check the RS232 cable.
Can it be defective?

ThePootisMann, 2020-11-29:
It does respond to touch. The cable may or may not be the source of the issue.

oldman4U, 2020-11-29:
Hi. Have you read the ReadMe? There are 3 things to keep in mind regarding the serial connection.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

radek8, 2020-11-29:
Check the cable. If ok, try this:

Connect a PC to run the Pronterface program
Send command M503 and M115
Does the printer write connected?

ThePootisMann, 2020-11-29:
The printer can print normally when in Marlin Mode, but the touch screen mode does not communicate with the printer.

oldman4U, 2020-11-29:
And after reading the ReadMe you know that the two modes use completely different connectors and therefore it is possible that one mode works, but the other does not?

And you know what has to be done to make the serial connection work?

ThePootisMann, 2020-11-29:
I do not know what has to be done to make the serial connection work. Could you enlighten me?

oldman4U, 2020-11-30:
Reading this: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

oldman4U, 2020-11-30:
I just wanted to add, that once the TFT gets power, the touchscreen will "work". This means you can change the menu by clicking on buttons and so on, WITHOUT a connection to a mainboard. The warning you get means, that the TFT can not "talk" to the mainboard and the ability to do so relies on the 3 points mentioned in the documentation above. Not more and not less.

ThePootisMann, 2020-11-30:
> I just wanted to add, that once the TFT gets power, the touchscreen will "work". This means you can change the menu by clicking on buttons and so on, WITHOUT a connection to a mainboard. The warning [...]

I have tried changing the serial ports defined in Marlin, and I have made sure the Baud rate is matched. Could it be the cable?

oldman4U, 2020-11-30:
In case the serial port in Marlin is correct and also the Baud Rate matches, only the cable can be the problem. The serial connection is really easy compared to Marlin based connection.

3 things.

The pin in the middle of the connector which is attached to the mainboard looks a bit loose on the picture above. I would check it.

radek8, 2020-11-30:
The cable or one of the serial port pins may be defective. First check the cable (ohm meter).

oldman4U, 2020-12-04:
Would be nice if you could let us know, thank you.

ThePootisMann, 2020-12-04:
I am sorry for my lack of attention towards this thread. I have decided that the best solution may just be to order another serial cable from Bigtreetech, although I am still open to suggestion.

oldman4U, 2020-12-04:
You can easily take 5 cables and make your own. An old HDMI cable for example would work great.

ThePootisMann, 2020-12-04:
How can I do this? Could you send me a tutorial?

oldman4U, 2020-12-04:
Can you crimp a connector?

ThePootisMann, 2020-12-04:
I lack the tool to do that

radek8, 2020-12-04:
If you are going to put a new connector, make a shielded cable (for example, from a cable from a PC mouse).
You will avoid other possible problems in the future.
You don't have to crimp, just solder and a shrink tube. :-)

ThePootisMann, 2020-12-04:
I am going to close the issue for now. Thank you to all who assisted me. I may re-open if the need arises.

closed 2020-12-04 by ThePootisMann

oldman4U, 2020-12-05:
Ok.

Hope the new cable solves the problem

👍🏻