bigtreetech/BIGTREETECH-TouchScreenFirmware #1312: TFT 24 display will not work in Marlin Mode

chance182, 2020-11-29:
I have an mini board E3 V 2 with a TFT 24 display installed on an Ender 3. I can not get the Marlin Mode to function. I have reflashed the latest firmware successfully with the main and the Oldman4U branch with the same problem. My marlin config includes CR10 display mode, as well as the advance config modes requested in the config ini of the TFT. The TFT works fine in touch screen mode. When I switch to dual mode I see the Marlin mode.  I can not switch to the mode with the knob as I have seen done on vids. When touch selected I do see the LCD simulator text. Text brightens when I press the button, but no marlin display. The 10 pin is connected to the EXP1 and the serial connector is plugged in. Reversed the serial and get the printer not connected. so thats not the problem. Looked through the previous issues but no success. Hope someone can help. Thanks

chance182, 2020-11-29:
[config chanc182.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5612497/config.chanc182.zip)

radek8, 2020-11-29:
The TFT24 V1.1 display does not have an EXP3 connector, so it does not support connection with SKR nini E3V2 in marlin mode

radek8, 2020-11-29:
You bought the wrong combination of SKR board with TFT

chance182, 2020-11-29:
My marlin config files attached

chance182, 2020-11-29:
I was sold as a combination stating that it would work in the Marlin mode. https://www.banggood.com/BIGTREETECH-SKR-MINI-E3-V2_0-Control-Board-+TFT24-V1_1-Touch-Screen-Kit-for-3D-Printer-Parts-Creality-3D-Printer-p-1757905.html?rmmds=myorder&cur_warehouse=CN

chance182, 2020-11-29:
Would another LCD controller selection in Marlin config work?

radek8, 2020-11-29:
I recommend complaining to the seller

radek8, 2020-11-29:
It could be connected.
You would have to overwrite the file with a pin connection
pins_BTT_SKR_MINI_E3_common.h
you would also have to reconnect the 10pin connector so that there is 1 connector on one side and 2 connectors on the other.
Complicated for a man who has never done it.

oldman4U, 2020-11-29:
At AliExpress they offer the TFT24 for Ender3 and I am sure, you can get the cable also separate. It is the one I dragged the purple frame around.

Pleas help the community and close the ticket once you do not need it anymore.

Thank you

[image]

oldman4U, 2020-11-30:
I also updated the ReadMe to reflect this combination of hardware properly:

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

Please help the community and close this ticket.

Thank you

chance182, 2020-11-30:
Will an update to the Marlin software be necessary with this cable? What is the pin to pin connection of this cable? Thank you.

radek8, 2020-11-30:
Try this.
Use the display driver in Marlin
#define CR10_STOCKDISPLAY

[image]

chance182, 2020-12-02:
This fix was successful once a custom cable was constructed. Thank you for your support

oldman4U, 2020-12-03:
👍🏻

Please help the community and close the ticket once you do not need it anymore. Thank you

chance182, 2020-12-03:
Thank you for your assistance

closed 2020-12-03 by chance182

oldman4U, 2020-12-04:
Happy printing

;-)