bigtreetech/BIGTREETECH-TouchScreenFirmware #1313: [Q] (Your Question?)

Robenc, 2020-11-30:
No way to roll back to 26.1 for the tft70 that is not buggy as the new version is

oldman4U, 2020-11-30:
Hi.

It would be great if you could update the title of this ticket to something more useful, thank you.

You can "go back in time" doing the following:
Click on Code and then on Commits like shown here: 
[image]
Then scroll down to the build you would like to use and click on the sign which I marked with the red frame:
[image]
This allows you to download an older firmware.

Please let me know if this solves your problem

Thank you

Robenc, 2020-12-01:
well there is no rollbacks to prior versions , please fix my tft 70 is not working and need the lastest before the merge

oldman4U, 2020-12-01:
Doing what I described above you can go to any version you want. Not sure what else I can do.

Robenc, 2020-12-03:
Been busy will check later

Robenc, 2020-12-04:
found it thanks had to back before the 27th

closed 2020-12-04 by Robenc

oldman4U, 2020-12-04:
Great you found it!

What are the bugs you are referring to?

Robenc, 2020-12-04:
Load the screen with the btt skr mini v2