bigtreetech/BIGTREETECH-TouchScreenFirmware #1314: [BUG] (short description)

Chukozor, 2020-11-30:
I tried the firmware in the folder "copy to SD", unified menu, TFT35 3.0, used in the commit 198f14e1f2f0137d7497d42cdf7aab9b8817c7ea (on a BTT SKR 1.3).

I found two issues  : 

     - the "first icon" extrude in the menu shows preheat menu (the other extrude icon below "move" is working fine).
     - during print, layer height doesn't update at each layer change. Sometimes it updates, sometime it doesn't.

oldman4U, 2020-11-30:
Hi.

The first "issue" is a feature which was implemented to make sure that a preheat is made before extrusion. It caused confusion and will be changed with the next release of the firmware.

Regarding Layer hight. Does this happen while printing from on board SD or a storage which is attached to the TFT? When it does not update, do you mean it updates late or does it update to a wrong value or does it skip certain layer hight at all?

Please let us know, thank you

@kisslorand @radek8

kisslorand, 2020-11-30:
@Chukozor Do you use ARC in your gcode? (G2, G3)
With G2, G3 commands the TFT doesn't capture the X, Y, Z coordinates.

Chukozor, 2020-11-30:
no, there is only G0 and G1 in the gcode I used

Chukozor, 2020-11-30:
@oldman4U ok thanks.
For the layer issue, I printed with SD card on TFT in TFT mode it skips certain layer height on the screen but not in the print (hopefully).

oldman4U, 2020-11-30:
Out of 10 how many times does it skip? And have you seen this before?
Which layer hight are you using? Can you attach a gcode file?

Chukozor, 2020-11-30:
@oldman4U I don't know, I only updated the firmware yesterday and made one print.
The layer stopped updating at 0.85mm height.
[Here the gcode I used](https://cdn.discordapp.com/attachments/763032939923111937/782994002634866688/Issue-layer-height.gcode)

oldman4U, 2020-11-30:
.85 or .75?

oldman4U, 2020-11-30:
Forget it.

Initial layer hight is .25 and all following are .2. All good.

Testing it with a dry print at 999% speed;-)

oldman4U, 2020-11-30:
Could you please try it again and next time the Layer hight stops - click on More and then Back - to see if the Layer hight updates to the correct value.

Thank you

Chukozor, 2020-11-30:
ok I'll try that, but I won't be able to print for several days. I'll try this for sure next print. Thanks

oldman4U, 2020-11-30:
If you want, you can close this ticket.

I am able to reproduce the issue here, so next I will find out which PR caused this and contact the developer.

Thank you

kisslorand, 2020-12-01:
I found the bug. It was introduced by PR #1272.
During printing the actual coordinates are not pulled from printer, they are gathered from the sent gCode. The mentioned [PR](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1272) was getting info from printer, which were not given.
A PR might follow or you can just edit PrintingMenu.c:
instead of:
`    if(curLayer != coordinateGetAxisActual(Z_AXIS)){
      curLayer = coordinateGetAxisTActual(Z_AXIS);`
write:
`    if(curLayer != coordinateGetAxisTarget(Z_AXIS)){
      curLayer = coordinateGetAxisTarget(Z_AXIS);`

Chukozor, 2020-12-03:
Hi again, @kisslorand  I tried what you wrote and I had the issue "nozzle dive in the bed" at each layer on the same point (not even at the layer change). [Photo here](https://cdn.discordapp.com/attachments/763032939923111937/783889803212488744/20201203_034149_compress61.jpg)

Chukozor, 2020-12-03:
but I didn't try the update #1319 , I just changed the line you told and compiled it. So maybe this update has no more this issue .... ? @kisslorand

oldman4U, 2020-12-03:
Please try the current master firmware which includes all fixes. 

Please help the community and close the ticket once you do not need it anymore. Thank you

kisslorand, 2020-12-03:
@Chukozor The "layer height doesn't update at each layer change and extruder icon issue" should have been resolved already. This new thing you have should be treated in a separate ticket. If you open a new ticket about it, please check before if the behaviour is present on  any gcode you try to print and please provide a sample gcode also, a smaller one, if possible.

oldman4U, 2020-12-03:
And in case it has been working before please provide also the good
firmware

closed 2020-12-03 by Chukozor