bigtreetech/BIGTREETECH-TouchScreenFirmware #1316: [BUG] (Print information screen disappears after the end of a print)

fftkatarn, 2020-11-30:
### Description

After the last upgrade, the print information screen automatically exits after the end of a print instead of being persistent.
I'm not sure if it was intentional but there should be the possibility to choose the behavior.

### Steps to reproduce

1. Start a print
2. In the end there are two "beeps" and the information screen (where print time, temperature, flow etc can be seen) automatically exits.

**Expected behavior**

Like in the old firmware, there should be the possibility to manually exit the information screen with a "back" or "exit" button.

**Actual behavior**

The information screen exits as soon as the print ends.

### Hardware Variant

TFT35 E3 v3.0

### TFT Firmware Version & Main Board Firmware details

TFT35_E3_V3.0.26.x Nov 27 2020

Marlin 2.0.7.2

oldman4U, 2020-11-30:
Hi.

Would you please verify that the reported issue does NOT exist in case you are printing from on board sd card, but exists when printing from TFT uDisk and TFT SD card.

Thank you

fftkatarn, 2020-11-30:
Hello,
Unfortunately I can't verify since I just realized that when I try to print from the on-board sd card, the TFT gives me an empty error dialog..

oldman4U, 2020-11-30:
Select the TFT sd card or udisk first before you select the on board sd card.

fftkatarn, 2020-11-30:
Unfortunately that didn't work, I still have the attached error together with a noisy continuous beep.

[image]

oldman4U, 2020-11-30:
Ok. Thank you no problem. I have all information and let you know asap.

👍🏻

fftkatarn, 2020-11-30:
Thank you very much!

GregSKR, 2020-12-01:
Hi
This is caused by PR https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1272 wich introduce lot af bugs...
Here is the faulty code (in green):
[image]

oldman4U, 2020-12-02:
Finally it is more a feature than a bug, but a option would be great. Finally a better handling of a finished print would be great.

@fftkatarn Do you need anything for now?

fftkatarn, 2020-12-02:
Ok, for me it's important to have an option because after every print I take note of the print time on the display, so I need to see it somehow.
Thank you!

oldman4U, 2020-12-02:
Understand.

In this case I would recommend to stay with a firmware which has been generated right before the PR has been merged, until a new functionality has been added, like this one: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/68615eceaf531ad659826820bb8e9b19dc38f199

I assume you are printing from the on board SD card, right?

fftkatarn, 2020-12-02:
Nope, I can't print from the onboard SD since the TFT returns me the error in the screenshot above (is this a bug? Should I open another issue about that?).
Anyway I always printed from the on-screen SD card.
Thank you

oldman4U, 2020-12-02:
This ticket is about exiting after print. In case you have another issue, I would recommend to open another ticket.

Using the same TFT I can not see the error above, but issues accessing the on board SD card where there is already a PR available for this. Printing from TFT based storage has been working with all versions of firmware I have been testing the last couple of days. So you can open a new ticket or you can wait for the on board SD card fix to be merged and see if this solves your 2nd problem.

Happy printing

oldman4U, 2020-12-02:
You are very welcome!

IF YOU WANT, you can try this firmware, which has hopefully all storage related issues fixed: https://github.com/kisslorand/BIGTREETECH-TFT/tree/ONBOARD_SD-Hangs

mpcolman, 2020-12-02:
Im always print from the TFT SD card slot, the issue exist there too... no print info when finished printing-----

oldman4U, 2020-12-02:
Printing from TFT there is no information. Printing from on board sd there was such an information which included the elapsed time, but this seems to be gone.

oldman4U, 2020-12-03:
Please check #1333 which shows a Print finished screen independent of the used storage.

fftkatarn, 2020-12-03:
> Printing from TFT there is no information. Printing from on board sd there was such an information which included the elapsed time, but this seems to be gone.

Maybe there was a misunderstanding, I always printed from the TFT SD (never from the on-board SD) and the post-print information was always there.

> Please check #1333 which shows a Print finished screen independent of the used storage.

That's good, I personally prefer the good ol' screen if possible to have it back though!

oldman4U, 2020-12-03:
I do not know this good old screen but maybe you want to give it a try and report issues if you find one.

fftkatarn, 2020-12-03:
That's exactly the purpose of this issue..

> ### Description
> After the last upgrade, the print information screen automatically exits after the end of a print instead of being persistent.
> I'm not sure if it was intentional but there should be the possibility to choose the behavior.

I try to explain myself better.
Step 1: I start a print from the TFT SD card reader, the display shows me a screen with the temperature, fan, flow and time elapsed information;
Step 2: The print ends: in the old firmware the information screen in step 1 remains there, in the new one no, it exits automatically.

What I'm asking is if it is possible to restore the old behavior.

Thanks!

oldman4U, 2020-12-03:
You mean the landing screen. I would call it a bug that you had to stop the print after the print was done. What was the benefit of having this information persistent and that you had to stop the print at the end of the print?

fftkatarn, 2020-12-03:
The benefit was the fact that I could take note of the print time, for me it wasn't a bug but a useful thing.
It would be nice to have an option in the config.ini file to choose to make it persistent or not.

oldman4U, 2020-12-03:
Makes sense. The new method shows not only the elapsed time but also the amount of filament used in case this information is available. Check the PR and let me know.

fftkatarn, 2020-12-04:
> IF YOU WANT, you can try this firmware, which has hopefully all storage related issues fixed: https://github.com/kisslorand/BIGTREETECH-TFT/tree/ONBOARD_SD-Hangs

I tried that firmware but no popup showed at the end of the print printing from the TFT SD reader, still not working the print from onboard SD.

fftkatarn, 2020-12-04:
That's the screen I'd like to have back. It was not a bug, it was useful and always worked.

[image]

oldman4U, 2020-12-04:
Hi.

And this is what you get with the new firmware. Elapsed time and Filament used.

[image]

In addition the print does not stutter, the Layer hight is shown correctly, there is no useless Pause button and you do not have to stop a print which has already finished.

Hope you enjoy it

fftkatarn, 2020-12-04:
I never stopped an already finished print, what are you talking about?
The popup it's ok but in the firmware you told me to try it didn't work for me.

oldman4U, 2020-12-04:
Depending on the firmware version, the storage device which has been used for the print, and if the print has been canceled before the print started or once the print has been going on or if the print has finished - different landing screens have been used and sometimes there was a print finished dialog and sometimes not. It was a mess. In addition some landing screens had still a Stop button, even the print has finished already. We unified this so now there is always a print finished screen which includes the elapsed time and in case available also the amount of filament used.

This is available in a PR which will hopefully be available soon in the master firmware and which can be downloaded from this private repo: https://github.com/mehmetsutas/BIGTREETECH-TouchScreenFirmware

Please let me know

fftkatarn, 2020-12-04:
available in a PR which will hopefully be available soon in the master firmware and which can be downloaded from this private repo: https://github.com/mehmetsutas/BIGTREETECH-TouchScreenFirmware
> 
> Please let me know
Just tried but no popup..

oldman4U, 2020-12-04:
Have you self compiled the firmware?

fftkatarn, 2020-12-04:
Nope, I used the pre-compiled firmware

oldman4U, 2020-12-04:
Thats the problem.

Can you compile your own?

oldman4U, 2020-12-11:
Hi.

All changes are in the master branch. Without the Print finished popup screen activated, you should get your classic landing screen.

Please install, verify and let us know or close this ticket in case the print screen stays like expected.

Thank you

fftkatarn, 2020-12-11:
Hi, sorry for the late reply, I was quite busy and I didn't have time to compile and test the suggested version in the last days.
I just tried the pre-compiled version in the master branch and it works! Now when a print finishes, the print screen is persistent and I have to press "back" to exit.
Thank you very much, you made me happy :)

closed 2020-12-11 by fftkatarn

oldman4U, 2020-12-11:
👍🏻