bigtreetech/BIGTREETECH-TouchScreenFirmware #1318: Can't compile with new STSTM32 library

BsCmOD, 2020-11-30:
Hi to every one I've just update all my platformio libraries but now I've got a problema, when I try to compile the firmware this message appears: Missing MCU or Product Line field. I think something is missing in platformio.ini but I don't know what... May you can help me?

kisslorand, 2020-11-30:
Something's wrong with the master repository. New PRs also cannot compile.

BsCmOD, 2020-11-30:
What do you mean with "New PRs cannot compile"?

kisslorand, 2020-11-30:
I just made a Pull Request and it cannot compile. however I can compile it locally.

BsCmOD, 2020-11-30:
Thank'you, so it will take some time so solve I think

JulienThomas49, 2020-12-02:
Hi , i confirm that the old version of STSTM32 (9.0.0) works instead of the new one.
You can temporary fix the problem by forcing using the old library version

you can do this by manually add the version in your plateform.ini (in the correct env section for your hardware) 
platform      = ststm32@9.0.0

bye !

BsCmOD, 2020-12-03:
Did I see that compiling is still with the 9.0 library, isn't compatible with the new one? Just curiosity

oldman4U, 2020-12-04:
BsCmOD

As long as nobody from the community does the work, the new library will not be available.

Would you be able to do it?

kisslorand, 2020-12-04:
The new platform wants to compile with the included USB_HOST and USB_OTG instead of the one included in this repository. I couldn't figure it out yet how to change that without deleting the from the v10 platform.

BsCmOD, 2020-12-04:
Ok so we have still to use the 9.0 library otherwise it don't works

kisslorand, 2020-12-04:
I just compared the compilation with 9 vs 10. There's not a single byte difference in terms of RAM  usage and Flash usage.
So I guess v9 is still good to go.

oldman4U, 2021-01-03:
Seems v11 is available. Has someone of you tried the new version?

kisslorand, 2021-01-03:
Yes, I did it... "silently".

oldman4U, 2021-01-03:
Do good things and talk about it. This is great 

Thank you

oldman4U, 2021-01-03:
Please help the community and close the ticket once you do not need it anymore. Thank you

BsCmOD, 2021-01-03:
Ok I close it!

closed 2021-01-03 by BsCmOD

kisslorand, 2021-01-03:
> Do good things and talk about it. This is great
> 
> Thank you

There's nothing great to talk about. You'll  have to stay at version 9 until someone will do something about it.  My PERSONAL build works just great with v11.
ST took a turn and they are sticking to it.