bigtreetech/BIGTREETECH-TouchScreenFirmware #1322: BTT SKR 1.4 Turbo +BTT TFT35 V3.0 last code d'not attach printer. Release of April work.

Foxcor, 2020-12-01:
Hello.
The problem is that the last code does not connect to the printer, reducing the speed to 115200 does not help.
What needs to be fixed in config.ini for it to work ????
At the same time, the firmware in the April releases works fine on both 115200 and 250,000, i.e. wires are normal.
English is not my first language, sorry.

kisslorand, 2020-12-01:
Check "default_envs =" in PlatfromIO.ini, line 31. Set it to "BIGTREE_TFT35_V3_0".

Foxcor, 2020-12-01:
> Check "default_envs =" in PlatfromIO.ini, line 31. Set it to "BIGTREE_TFT35_V3_0".

I use "Copy to SD Card root directory to update/"

oldman4U, 2020-12-02:
So you are using pre compiled firmware and do not self compile using VSC and PIO?

Have you read the ReadMe from the main page: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware
 and understood what's needed to get the serial connection to work?

What is the baud rate used under April version?

Foxcor, 2020-12-02:
> So you are using pre compiled firmware and do not self compile using VSC and PIO?
Yes
> and understood what's needed to get the serial connection to work?
Yes )))
> What is the baud rate used under April version?
250000 it work. And also the esp-01c is connected to the TFT 250000. 
The only thing I found in the config.ini - baudrate: 5

oldman4U, 2020-12-02:
You can change the baud rate on the TFT.  Please check and select 250000
there.

oldman4U, 2020-12-03:
Baudrate 5 in config.ini means 115200, 6 is the correct value for 250000

Foxcor, 2020-12-03:
> Baudrate 5 in config.ini means 115200, 6 is the correct value for 250000

:( 
I check this.

Foxcor, 2020-12-03:
I compiled everything myself, it worked !!!
thanks

oldman4U, 2020-12-03:
Sounds great.

Compiling the own firmware is usually the best way to solve problems.

Thank you for letting us know. Please close the ticket once you do to need it anymore.

Happy printing

closed 2020-12-03 by Foxcor