bigtreetech/BIGTREETECH-TouchScreenFirmware #1324: BTT screen firmware corrupting gcode on Core XY printer.

BoothyBoothy, 2020-12-01:
I have a HEVO Core XY machine based on a SKR 1.4 Turbo with TMC2209 drivers. When printing through the screen SD card I get the following result where there are multiple gcode error warnings and the gcode appears to be corrupted on the fly causing the printer to run off the print and resume shortly after leaving a filament loop - see video link. I have tried numerous versions of Marlin all with the same result. I have also tried a BTT TFT70, a BTTFT35 & a BTT35 E3, all with the same result.

1. Using the onboard card reader it does not occur.

2. The above mentioned displays work fine on all my other six Cartesian printers.

3. I also note that Makers Muse had a similar experience when recently reviewing the BIUQ Kickstarter printer with the TFT70 screen.
4. I exclusively slice with Simplify 3D so do not know if this is relevant but I do not have any issues with my other machines.

I really want to use the advantages of the touch screen and now fed up of purchasing variants to see if I can make it work.

Anyone any ideas? Thanks in advance. Boothy

You Tube link showing issue: https://youtu.be/zMzgkPCPEDU

Maker`s Muse video: https://youtu.be/Hz2GGUbufG8?t=477

radek8, 2020-12-01:
Interference oscillates into the serial cable.
Use a shielded cable to connect the serial line

BoothyBoothy, 2020-12-01:
> Interference oscillates into the serial cable.
> Use a shielded cable to connect the serial line

Thanks, I will give that a try tomorrow. Best Boothy

radek8, 2020-12-01:
What speed of communication do you use?

BoothyBoothy, 2020-12-01:
115200

radek8, 2020-12-01:
ok

BoothyBoothy, 2020-12-02:
To undertake a quick test on this premise I wrapped a new serial cable (the existing seemed warn) in layers of aluminium foil and tapped each end. Unfortunately both test prints the issue still occurred (both faster than normal but with no Gcode errors. I was hoping this was a cure. I do have a quite large mossfet near the SKR board unlike my other printers which may be producing interference and your suggestion seems very logical. I am not sure where or if I can purchase a compatible insulated serial cable to make a proper test. Thanks again Boothy

BoothyBoothy, 2020-12-02:
I could try printing without the heated bed to see if that as any effect?

radek8, 2020-12-02:
Interference can be caused by PWM control of the bed, nozzle, or motor controller.
Recently, I also began to experience interference.
I use a speed of 250,000. In order not to have to reduce the speed, I will also make a new shielded serial cable.
A cable from an old computer mouse or keyboard can be used well.
They have 4 wires + shielding :-) Just crimp new connectors.

BoothyBoothy, 2020-12-02:
Yes! I have I am just printing a Benchy without the heated bed, 1 hour 10 minutes through and no issues so far. Could be the heated bed mossfet. I will set about shielding the main board from this and report back on my findings. Thank you so so much for your input and inspiration regards this issue. I will also knock out a new cable as  per your instructions. Very best Boothy

oldman4U, 2020-12-03:
Sounds great 

Please help the community and close the ticket once you do not need it anymore. Thank you

BoothyBoothy, 2020-12-03:
Hi Radek8 & oldman4U
Unfortunately this is still ongoing but getting warmer.
Radek8. I created a steel shield for the Mossfit and further shielded the serial cable but still with no success.
I then thought that all my printers that are working with the BTT kit such as the Ender 3 and 5 the display is remote from the mainboard. I have therefore since placed a BTT screen 300mm from the board with the heated bed on and the print has now been running for an hour with no issues.  I am therefore coming to the conclusion that it is the display itself that requires shielding not necessarily the cable. Tomorrow I will shield the back of a display and see if this makes a difference when still near the mainboard. My HEVO design is based on having all the parts together to reduce cable runs and I do not really want a $20 display dictating the design and do not want to completely rewire things to ,make this work. I am currently also designing a new concept printer and I really want to sort this if I am going to use any BTT kit. Best wishes Boothy

radek8, 2020-12-03:
Interesting knowledge,
So if the shielded cable does not help, it will be necessary to shield the Display or the Marlin board ...

BoothyBoothy, 2020-12-03:
Probably easier to shield the display.

oldman4U, 2020-12-04:
Hi Boothy.

Please help the community and close this ticket once you do not need it anymore.

Thank you

BoothyBoothy, 2020-12-04:
Ok, just to conclude, I have insulated a BTT TFT70 display with a couple of layers of tin foil inside a Ikea sandwich bag and and I printed a Benchy with absolutely no issues. I will report back if things vary, I did intend to design some super awesome insulated base enclosure but life is too short so I might opt for a the tin foil and a smaller bag! Radek8 thanks again. Boothy
[image]

closed 2020-12-04 by BoothyBoothy

radek8, 2020-12-04:
Print the box on the display and stick it on the inside of the foil. It will be more elegant :-)