bigtreetech/BIGTREETECH-TouchScreenFirmware #1325: [BUG] (sd card tft 24 v.1 not work in marlin mod)

romucifer, 2020-12-01:
### Description

sd card in tft 24 v1.1 work in touch mode but in marlin mode not work (motherboard skr e3 dip v1.1)

radek8, 2020-12-01:
Bad combination of SKR board and TFT display
Look at  #1312

radek8, 2020-12-01:
the same problem

romucifer, 2020-12-01:
> Mauvaise combinaison de la carte SKR et de l'écran TFT
> Regardez [# 1312](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1312)
my mod marlin work i have make cable exp3 (skr e3 dip v1.1)to exp1 and exp2 (tft24 v1.1)
but only sd card tft not work in marlin mod

radek8, 2020-12-01:
The EXP3 connector does not connect to an external SD card.
In Marlin mode you can only use the SD card on skr e3 dip v1.1

However, the condition is a correctly configured Marlin
you have to allow
#define SDCARD_CONNECTION ONBOARD

romucifer, 2020-12-01:
> Le connecteur EXP3 ne se connecte pas à une carte SD externe.
> En mode Marlin, vous ne pouvez utiliser la carte SD que sur skr e3 dip v1.1
> 
> Cependant, la condition est un Marlin correctement configuré,
> vous devez autoriser
> #define SDCARD_CONNECTION ONBOARD

for sd on board skr e3 work perfect in marlin mode it is posible mod pin for active sd card tft in marlin mod

radek8, 2020-12-01:
Marlin can do a lot, but your board has an EXP3 connector and it doesn't have the pins needed to connect an SD card in the TFT.
The SD card in TFT communicates via pins which are connected in the EXP2 connector.
It is understandable?

romucifer, 2020-12-01:
I understood that the skr e3 dip v1.1 only has exp3 but I wonder if by remapping the pins not to use as for example the swd pins to the entries of the sd support on the skr e3 dip v1.1 card

oldman4U, 2020-12-02:
romucifer

This repository is about software and radek answered your questions already. Please be so kind and start a ticket in the corresponding hardware related repositories for additional information and close this ticket here.

Thank you

radek8, 2020-12-02:
Everything is possible if you know what you are doing :-)
It will be more suitable for you to buy new HW that will better suit your needs.

oldman4U, 2020-12-02:
The SKR E3 DIP is a direct replacement for the Ender 3 which comes with an onboard SD card and a 12864 display which has no SD Card. So you need another mainboard or you use the TFT based storage options with the touch mode. Those two options you have without the possibility to brake your hardware.

romucifer, 2020-12-02:
ok thank you very much i have look pin on marlin for remap on ISP 
i have close

closed 2020-12-02 by romucifer

oldman4U, 2020-12-02:
👍🏻

radek8, 2020-12-02:
You will have to find where the pins are routed under the motor drivers, but you will miss 2 more pins

[image]

radek8, 2020-12-02:
Isn't this better?

[image]