bigtreetech/BIGTREETECH-TouchScreenFirmware #1327: Problem with the board JSON files?

Andy-ABTec, 2020-12-02:
Just picked up an update to Platform IO while compiling for my BTT Mini E3 V3 earlier today, now The TouchScreenFirmware wont compile...

I'm seeing: "AssertionError: Missing MCU or Product Line field:"

I think it requires a tweak to the JSON board files - adding "product_line: STM......?

Anyone else?

Cheers,
Andy B.

oldman4U, 2020-12-02:
Please see #1318 and close this ticket in case it is the same issue

Thank you

Andy-ABTec, 2020-12-02:
It looks like its the same problem but the folks over on platform io appear to be recommending a tweak to the board JSON files as the appropriate solution...

https://community.platformio.org/t/compilation-failure-using-platform-ststm32-10-0-0/17491

oldman4U, 2020-12-02:
Please see @kisslorand reply 4h ago: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1326

Andy-ABTec, 2020-12-02:
Nuff said...

closed 2020-12-02 by Andy-ABTec

oldman4U, 2020-12-02:
;-)
Hope there is a permanent solution for this soon...

Happy printing