bigtreetech/BIGTREETECH-TouchScreenFirmware #1330: [BUG] M75 command issue

BsCmOD, 2020-12-02:
Hi to every one I have to report a newly discoverd issue, with the TFT35V3 in touch mode the M75 command in start gcode lock the print, the hotend and bed won't reach the desired temperature and job won't to start. Hope to be usefull. Bye

oldman4U, 2020-12-02:
Hi.

So the work around is to avoid the timer command in start gcode, right? But why would I define M75 in my start gcode, where the elapsed time is always shown automatically? By mistake?

BsCmOD, 2020-12-02:
Right, it's the timer start command and yes, the timer start automatically but I noticed that if I print in touch mode the print time Marlin counter does not increase so I tried to "solve" this problem by using M75 command although it shouldn't be necessary to use it since the timer autostart is enabled in configuration.h in Marlin. That's it

oldman4U, 2020-12-02:
The print time on screen or that there is no accumulated print time?

BsCmOD, 2020-12-02:
I'm referring to accumulated print time, number of jobs, total extruded filament etc... those counters works only if I print in Marlin (native) mode. I'm not totally sure but I think some time ago this funtion has worked. It's sin to loose/not use these counters, it's not a foundamental functions anyway it could be helpfull to know the works hours of the machine.

oldman4U, 2020-12-03:
@radek8 do you know if M75 is somehow implemented using the touchscreen?

BsCmOD, 2020-12-03:
I don't know

oldman4U, 2020-12-03:
radek8 is the one who knows everything;-)

radek8, 2020-12-03:
The M75 command is not used on TFT

oldman4U, 2020-12-03:
Thank you Radek, thought so.

@BsCmOD If you want I can add this to the FR list. To do so, please change the label to FR and if you want, you can close this ticket afterwards. I will take care that it is added to the FR list.

Happy printing

oldman4U, 2020-12-03:
Maybe you also want to edit the headline and add some more information how you would see this feature being configured and used in touchscreen mode.

BsCmOD, 2020-12-03:
Closed for FR

closed 2020-12-03 by BsCmOD

oldman4U, 2020-12-03:
Please describe how this should  work.
 Which options are shown where and how and when is the collected data shown
or what does it „execute“?

BsCmOD, 2020-12-05:
M75 it's a command which activate Marlin timer during printing. I'd tried to use because I saw that the total amount counters in Marlin didn't increase if I use TFT (touch) mode, no problem in Marlin mode. When I tried to use it the results was a freeze of the machine after preheating and M75 injection command I solved obviously by deleting M75....

kisslorand, 2021-01-02:
I actually use M75 in the start gCode of the TFT without any issues. No freezing whatsoever.

digant73, 2021-07-27:
@oldman4u this was probably a bug that is no more present so this FR could be close