bigtreetech/BIGTREETECH-TouchScreenFirmware #1332: MKS DLC v.2 +TFT 24

Kakauko, 2020-12-03:
on the laser engraving on the display after inserting the sd card I select the file to be engraved and I want to press the border, but the display beeps and reboots.  In pc and GRBL the border works, but offline from the sd card does not, just the engraving itself.  Thank you for your answers and guidance on where the problem might be.

oldman4U, 2020-12-03:
Has this been working before?

A movie showing the issue could be helpful.

Kakauko, 2020-12-04:
I have already solved it, downloaded the file 
MKS_TFT24_CNC_V1.0.2 Release file.zip
, unzipped, opened the Examples folder and copied all 4 folders to the SD card, inserted it into the switched off engraver, turned it on and it is automatically updated.  the laser is finally showing me the border of the image.  Perhaps it will help others with a similar problem. 
Sorry for english, Google translate.

oldman4U, 2020-12-04:
Hallo Kakauko.

Thank you for letting us know.

Please help the community and close this ticket once you do not need is anymore.

Thank you

oldman4U, 2020-12-06:
Please close this ticket.

Thank you

closed 2020-12-07 by Kakauko