bigtreetech/BIGTREETECH-TouchScreenFirmware #1334: [BUG] Illegal Flash App!

Doyle4, 2020-12-03:
Is it possible to have this disabled at all? All I wanted to do was update my config, nothing else on SD card, I get this error and have to re-flash the whole thing, or at least when it says it a reboot of the printer/screen rolls back so the printer can be used and not locked out of the whole printer. thanks :)

oldman4U, 2020-12-03:
Hi.

So the config.ini and the installed firmware are from the same version and even so, you had this problem?

oldman4U, 2020-12-03:
Welcome back, btw.

Have you got a TFT35 in the meantime?

Doyle4, 2020-12-03:
Hey @oldman4U sadly im back haha.

yeah both from same version, I did everything the same as before when worked, and today this happened, if its something that can happen now n again and cant be patched, then knowing this will help, but Im hoping it could be something that can be avoided maybe? I like the idea of rollback at least so we can still use the printer.

Yeah, the TFT35 E3 V3 running a firmware you passed last time

oldman4U, 2020-12-03:
It’s not sad that you are back. I still hope that you will join the caravan
one day.

I have a standard SD card which causes this error almost every time I use
it. And there are 9 others which are working fine. Usually I copy firmware,
config and TFT folder onto the card and remove the card for a restart after
the update. This works flawlessly- except for this particular  SD card😁

Not sure what can be done to avoid it and most likely the time needed to
improve this is better used for new features or fixing bugs. Resources are
VERY limited.

Doyle4, 2020-12-03:
Im by the campfire wondering whats going on :rofl:

Ah, now that is something I didn't think of, I couldn't find my usual SD card and used at least a 5year old SD card, tomorrow I'll test with my usual now I have found it.

Didn't think of it to be used as a bug hunt, maybe an option to disable? and if the screen detects Illegal Flash App, instead of semi bricking, update fails and goes to main screen,  the user would then know the firmware failed, just a thought :P 
Sorry if this isn't possible, I'll put my hand up and say I have no idea how coding works, Im surprised I can compile my own Marlin firmware, would love to know how to compile my own TFT firmware.

oldman4U, 2020-12-03:
I am no developer too but try to bring the stuff together 🤩 Check the card and I guess it will solve your problem.

Doyle4, 2020-12-04:
@oldman4U Tried my main card, seemed fine so far :)

Ah I'm glad its not just me who sits back and watches everyone else do all the work and then learn what they did :P haha, all good fun though!

Do you know if its possible to create BTT TFT menus? Id love to have my icon shapes, icon layout, remove some features I never use, maybe even add my own? Thanks :)

oldman4U, 2020-12-04:
I do not know what the difference is, but the SD card makes a difference;-)

Well I learn by helping others here, the problem is that sometimes I only BELIEVE that I can help... Together we then find out and I learned something new again.

You can change the look and feel and since last week, there are 3 different UIs available. To be able to remove and add functions you need good dev. skills which I do not have. There is even a FR in the pinned FR ticket regarding this, but it seems to be very complex.

Doyle4, 2020-12-04:
I see myself like you describe your self, I'll help and give information on what I THINK is correct, but will always say "Wait for an experts advice" unless I know my help is the fix for a certain problem.

Ah cool! I'll haver to check that out, Marlin looked so alien to me at first, had no idea what was what, but over the months become more comfortable. But sounds like for TFT its more complex, but i'll defiantly have a read.

Could you link me to the new firmware please? I can only see an April release and one that was updated 16days ago.

Doyle4, 2020-12-04:
found it :)

oldman4U, 2020-12-04:
Are u using Unified, Round or Red?

Doyle4, 2020-12-04:
Unified as thats the cleanest out of them all, but I have already tweaked them icons, would love my own layout, main features on homesceeen rather than submenus

oldman4U, 2020-12-04:
I have a full concept for a modified interface with better navigation (max3 clicks to EVERY function) including a new flat UI on my desk. Unfortunately there are no developers which would do such a massive rewrite.

Doyle4, 2020-12-04:
I like the idea of a smart phone or tablet, able to swipe etc. I can see why people have jumped to Octoprint etc with these new screens that accept the Rasp Pi

oldman4U, 2020-12-04:
Swipe is not possible. The TFTs are not only displays but also computers
and for the price they work great.

oldman4U, 2020-12-04:
> Aww man! If I knew how I would of taken that task

... we would develop it together😁

Doyle4, 2020-12-04:
Im up for a challenge if you are ha, I think it deserves a makeover.

Yeah for the price they are good, wish it was a bit user friendly to customise, i guess it is for some but they have the knowledge 😂

oldman4U, 2020-12-04:
My skills are 30 years old and not really up to date.... Guess you would have to make this alone;-)

The software improves but it takes time. Open Source...

Doyle4, 2020-12-04:
Everything I know so far is 4months old :P im a proper newbie, but love learning.

oldman4U, 2020-12-05:
Do you have any programming skills?

Doyle4, 2020-12-05:
Yes an no, haha

Yes, as in I now know how to make my own Marlin firmware and code LED's with Arduino.

No comes into play as there has been times on here and Marlin section where I have no idea what they are saying to me

oldman4U, 2020-12-06:
Happens to me all the time. Especially when developers start talking to each other 🥸

oldman4U, 2020-12-06:
Hi.

Seems that the initial reason for the ticket has been solved. Maybe you can close the ticket now.

Happy printing and see you soon

Doyle4, 2020-12-07:
Thanks again for the help!

closed 2020-12-07 by Doyle4

oldman4U, 2020-12-07:
👍🏻You are welcome