bigtreetech/BIGTREETECH-TouchScreenFirmware #1335: [BUG] Error working with BTT SD Cloud after finishing printing or emergency stop.

Foxcor, 2020-12-03:
I flash the latest version. TFT35 V3.0. All good. I am using BTT SD Cloud. But there was a problem, after the end of printing or an emergency stop, there was no access to the Flash card, while SD Cloud was connected. Reloading the screen helps regain access. Can you fix this?

oldman4U, 2020-12-03:
This problem like many others are most likely related to the WebDAV server running on the cloud device.

Which computer os are you using?

What kind of files can you copy onto the cloud device?

Is the copying process 100% reliable or do you have to copy files a second time from time to time?

Foxcor, 2020-12-04:
@oldman4U 
> Which computer os are you using?
OS Windows 10, SMB.

> What kind of files can you copy onto the cloud device?
Anyone, I was flashing the screen through SD Cloud.

> Is the copying process 100% reliable or do you have to copy files a second time from time to time?
For example, a model was printed. She's defective. Remake, make adjustments. Do not turn off the printer. Re-copy required, but SD Cloud does not see TF card.
First I copy the file to SD Cloud
[image]
TF card is locked by Marlin during printing
[image]
After the end of printing, the TF card is also blocked, but there are no more messages.
[image]
In this case, the TFT screen itself sees the TF card and its contents.

oldman4U, 2020-12-04:
Would be good to have someone with the cloud connected to the mainboard to
see if Marlin is able to release it after a print.

Foxcor, 2020-12-04:
@oldman4U 
I still have BTT ESP-01S connected to TFT. )))
When I press the TFT SD button, I get this picture.
[image]
After 20 seconds everything is empty again.
[image]
In the April firmware, everything worked without problems in the releases. Thus, there is clearly something in the screen code, what exactly ???

Foxcor, 2020-12-04:
It is necessary to somehow compare the versions from April and the current one.

oldman4U, 2020-12-04:
It would be best to check firmware version between April and now to see where it got broken. I also have the ESP, but I took out the cloud because it was useless. SO you mount the cloud using smb://yourIPaddress? I can test this also, because the https: is a nightmare.

Foxcor, 2020-12-04:
> smb://yourIPaddress
It for MacOS, help for you ))) SMB native protocol for Windows and my path \\yourIPaddress.
I found differences in the code. The new version adds support for config.ini and for some reason the data type is different.
[image]
Iam not programmer, but in old code is not in BIGTREETECH-TouchScreenFirmware-master\TFT\src\User\API\interfaceCmd.c
`  # ifdef SERIAL_PORT_2
        case 20: //M20
          if (!fromTFT)
          {
            if (startsWith("M20 SD:", infoCmd.queue[infoCmd.index_r].gcode) ||startsWith("M20 U:", infoCmd.queue[infoCmd.index_r].gcode))   {
            if(startsWith("M20 SD:", infoCmd.queue[infoCmd.index_r].gcode)) infoFile.source = TFT_SD;
            else infoFile.source = TFT_UDISK;
            strncpy(infoFile.title, &infoCmd.queue[infoCmd.index_r].gcode[cmd_index + 4], MAX_PATH_LEN);
            // strip out any checksum that might be in the string
            for (int i = 0; i < MAX_PATH_LEN && infoFile.title[i] !=0 ; i++)
              {
                if ((infoFile.title[i] == '*') || (infoFile.title[i] == '\n') ||(infoFile.title[i] == '\r'))
                {
                  infoFile.title[i] = 0;
                  break;
                }
              }
            Serial_Puts(SERIAL_PORT_2, "Begin file list\n");
            if (mountFS() == true && scanPrintFiles() == true){
              for (uint16_t i = 0; i < infoFile.f_num; i++) {
                Serial_Puts(SERIAL_PORT_2,infoFile.file[i]);
                Serial_Puts(SERIAL_PORT_2,"\n");
              }
              for (uint16_t i = 0; i < infoFile.F_num; i++) {
                Serial_Puts(SERIAL_PORT_2,"/");
                Serial_Puts(SERIAL_PORT_2,infoFile.folder[i]);
                Serial_Puts(SERIAL_PORT_2,"/\n");
              }
            }
            Serial_Puts(SERIAL_PORT_2, "End file list\nok\n");
            purgeLastCmd();
            return;
            }
          }
          break;

        case 23: //M23
          if (!fromTFT)
          {
            if (startsWith("M23 SD:", infoCmd.queue[infoCmd.index_r].gcode) || startsWith("M23 U:", infoCmd.queue[infoCmd.index_r].gcode))
            {
              if(startsWith("M23 SD:", infoCmd.queue[infoCmd.index_r].gcode))
                infoFile.source = TFT_SD;
              else
                infoFile.source = TFT_UDISK;
              resetInfoFile();
              strncpy(infoFile.title, &infoCmd.queue[infoCmd.index_r].gcode[cmd_index + 4], MAX_PATH_LEN);
              // strip out any checksum that might be in the string
              for (int i = 0; i < MAX_PATH_LEN && infoFile.title[i] !=0 ; i++)
                {
                  if ((infoFile.title[i] == '*') || (infoFile.title[i] == '\n') ||(infoFile.title[i] == '\r'))
                  {
                    infoFile.title[i] = 0;
                    break;
                  }
                }
              Serial_Puts(SERIAL_PORT_2, "echo:Now fresh file: ");
              Serial_Puts(SERIAL_PORT_2, infoFile.title);
              Serial_Puts(SERIAL_PORT_2, "\n");
              FIL tmp;
              if (mountFS() && (f_open(&tmp, infoFile.title, FA_OPEN_EXISTING | FA_READ) == FR_OK) )
              {
                char buf[10];
                sprintf(buf, "%d", f_size(&tmp));
                Serial_Puts(SERIAL_PORT_2, "File opened: ");
                Serial_Puts(SERIAL_PORT_2, infoFile.title);
                Serial_Puts(SERIAL_PORT_2, " Size: ");
                Serial_Puts(SERIAL_PORT_2, buf);
                Serial_Puts(SERIAL_PORT_2, "\nFile selected\n");
                f_close(&tmp);
              }
              else
              {
                Serial_Puts(SERIAL_PORT_2, "open failed, File: ");
                Serial_Puts(SERIAL_PORT_2, infoFile.title);
                Serial_Puts(SERIAL_PORT_2, "\n");
              }
              Serial_Puts(SERIAL_PORT_2, "ok\n");
              purgeLastCmd();
              return;
            }
          }
          break;

        case 24: //M24
          if (!fromTFT)
          {
            if ((infoFile.source == TFT_UDISK) || (infoFile.source == TFT_SD))
            {
              if (isPause())
              {
                setPrintPause(false, false);
              }
              else {
                Serial_Puts(SERIAL_PORT_2, "ok\n");
                purgeLastCmd();
                infoMenu.cur = 1;
                menuBeforePrinting();
              }
              return;
            }
          }
          break;

        case 25: //M25
          if (!fromTFT)
          {
            if (isPrinting() && !infoHost.printing)
            {
              setPrintPause(true, false);
              Serial_Puts(SERIAL_PORT_2, "ok\n");
              purgeLastCmd();
              return;
            }
          }
`
In my Marlin Configuration.h #define SERIAL_PORT_2 0
My hardware config SKR 1.4 Turbo+TFT 35 V3.0+SD Cloud+ESP-01S
Please help me)))

Foxcor, 2020-12-04:
My suggestion. TFT display blocks all contents of TF SD card during printing. After finishing printing, for some reason, it does not unlock SD for external access. Thus, when the code is executed (mountFS () && (f_open (& tmp, infoFile.title, FA_OPEN_EXISTING | FA_READ) == FR_OK)), exclusive access occurs. I think after printing it is necessary to somehow remove this exclusive access.
The logic assumes that to replace the file, the SD card will be pulled out and after it is reinserted, everything will be mounted again. It is required to programmatically make a remove-insert SD card.

Foxcor, 2020-12-04:
Now I checked it again. When you press the button in the menu item
Print-> SD Card TFT-> file list, after that all, files for SD Cloud are not available. Those. after mounting the SD card with the display, it becomes inaccessible to anyone.
It is required to add the unmount script after exiting the menu, and after finishing printing or stopping.

closed 2020-12-04 by Foxcor

oldman4U, 2020-12-04:
Why did you close the ticket?

Foxcor, 2020-12-04:
> Why did you close the ticket?

I create [FR] #1343