bigtreetech/BIGTREETECH-TouchScreenFirmware #1338: BTT-TFT35-E3-V3.0 filament runout don't beep any alarm and X & Y mottors are lock disabled after a while...

mpcolman, 2020-12-03:
Well... before no solution of my question about this issue I publish now like an issue,  need to be solved it's a big problem....

When filament out, TFT activate "filament runout" and appears the alrt message but without any beep sound like alarm, can I set up a beep alarm with a specified time beeps?? in Marlin mode it's work fine but not with the TFT, have connected the sensor directly to the TFT screen.

And when a certain time passes (don't have certainly time lapse) the motors of the X&Y get unlocked, free moves and lost positions, bed and filament heaters still get on, but the motors don't in lock mode......, in Marlin mode its works fine......
thank for the help

I have instaled the latest firmware.
in Marlin have loaded this parameters...
ADVANCED_PAUSE_FEATURE
#define ADVANCED_PAUSE_FEATURE
#if ENABLED(ADVANCED_PAUSE_FEATURE)
#define PAUSE_PARK_RETRACT_FEEDRATE 80 // (mm/s) Initial retract feedrate.
#define PAUSE_PARK_RETRACT_LENGTH 5 // (mm) Initial retract.
// This short retract is done immediately, before parking the nozzle.
#define FILAMENT_CHANGE_UNLOAD_FEEDRATE 20 // (mm/s) Unload filament feedrate. This can be pretty fast.
#define FILAMENT_CHANGE_UNLOAD_ACCEL 25 // (mm/s^2) Lower acceleration may allow a faster feedrate.
#define FILAMENT_CHANGE_UNLOAD_LENGTH 150 // (mm) The length of filament for a complete unload.
// For Bowden, the full length of the tube and nozzle.
// For direct drive, the full length of the nozzle.
// Set to 0 for manual unloading.
#define FILAMENT_CHANGE_SLOW_LOAD_FEEDRATE 6 // (mm/s) Slow move when starting load.
#define FILAMENT_CHANGE_SLOW_LOAD_LENGTH 0 // (mm) Slow length, to allow time to insert material.
// 0 to disable start loading and skip to fast load only
#define FILAMENT_CHANGE_FAST_LOAD_FEEDRATE 20 // (mm/s) Load filament feedrate. This can be pretty fast.
#define FILAMENT_CHANGE_FAST_LOAD_ACCEL 35 // (mm/s^2) Lower acceleration may allow a faster feedrate.
#define FILAMENT_CHANGE_FAST_LOAD_LENGTH 60 // (mm) Load length of filament, from extruder gear to nozzle.
// For Bowden, the full length of the tube and nozzle.
// For direct drive, the full length of the nozzle.
//#define ADVANCED_PAUSE_CONTINUOUS_PURGE // Purge continuously up to the purge length until interrupted.
#define ADVANCED_PAUSE_PURGE_FEEDRATE 6 // (mm/s) Extrude feedrate (after loading). Should be slower than load feedrate.
#define ADVANCED_PAUSE_PURGE_LENGTH 100 // (mm) Length to extrude after loading.
// Set to 0 for manual extrusion.
// Filament can be extruded repeatedly from the Filament Change menu
// until extrusion is consistent, and to purge old filament.
#define ADVANCED_PAUSE_RESUME_PRIME 0 // (mm) Extra distance to prime nozzle after returning from park.
//#define ADVANCED_PAUSE_FANS_PAUSE // Turn off print-cooling fans while the machine is paused.

                                              // Filament Unload does a Retract, Delay, and Purge first:
#define FILAMENT_UNLOAD_PURGE_RETRACT 13 // (mm) Unload initial retract length.
#define FILAMENT_UNLOAD_PURGE_DELAY 5000 // (ms) Delay for the filament to cool after retract.
#define FILAMENT_UNLOAD_PURGE_LENGTH 1 // (mm) An unretract is done, then this length is purged.
#define FILAMENT_UNLOAD_PURGE_FEEDRATE 25 // (mm/s) feedrate to purge before unload

#define PAUSE_PARK_NOZZLE_TIMEOUT 300 // (seconds) Time limit before the nozzle is turned off for safety.
#define FILAMENT_CHANGE_ALERT_BEEPS 10 // Number of alert beeps to play when a response is needed.
#define PAUSE_PARK_NO_STEPPER_TIMEOUT // Enable for XYZ steppers to stay powered on during filament change.

#define PARK_HEAD_ON_PAUSE // Park the nozzle during pause and filament change.
//#define HOME_BEFORE_FILAMENT_CHANGE // If needed, home before parking for filament change

#define FILAMENT_LOAD_UNLOAD_GCODES // Add M701/M702 Load/Unload G-codes, plus Load/Unload in the LCD Prepare menu.
//#define FILAMENT_UNLOAD_ALL_EXTRUDERS // Allow M702 to unload all extruders above a minimum target temp (as set by M302)
#endif

poumsoon, 2020-12-05:
I suffered from the same problem for a while.
The tft35 version 3.0 can use the screen in two modes,
These two are touch mode and marine mode.
When using the filament sensor in touch mode, connect the filament sensor to the Fil-det terminal on the back of the LCD, and turn on the filament sensor in the setting menu of the touch screen screen. In this case, the firmware is set and the operation is irrelevant.
And when using in marine mode, you have to set the filament sensor in the firmware and connect the filament sensor to the board.

oldman4U, 2020-12-06:
mpcolman

What is the status of the reported issue? Some users tried to help but no answer from you. Maybe you could also close at least one of the tickets.

Thank you

oldman4U, 2020-12-11:
Please try to reset the TFT memory under Menu - Settings - Feature - Reset default settings... to see if this solves your problem with the locked motors.

Let us know

oldman4U, 2020-12-14:
Not to answer and to keep the ticket opened is very unfriendly.

closed 2020-12-15 by mpcolman