bigtreetech/BIGTREETECH-TouchScreenFirmware #1339: Bug [BUG] (short description)

CamWynd, 2020-12-03:
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

oldman4U, 2020-12-04:
Self or precompiled firmware? Send a picture of Menu Settings Info.

oldman4U, 2020-12-04:
Maybe you want to change the title of the ticket.

CamWynd, 2020-12-04:
It's the precompiled BIGTREE_TFT35_V3.0_E3.26.x.bin file that I'm using. I could send pics, but they would only show the Classic menu option (on the new firmware) or the Unified menu on the old, both of which are on the site; or are you looking for something else?

I agree on the ticket title rename; how do I do that (first time posting a ticket)?

oldman4U, 2020-12-04:
Can you self compile?

What is the version under Menu Settings Info?

You can click on the title and change it

CamWynd, 2020-12-04:
Never mind; I bought a new sd card and it works now?! 
Really weird that the 2 older cards both gave me the Classic menu, but the new (larger) card works. 

I'm really starting to hate technology.

closed 2020-12-04 by CamWynd

oldman4U, 2020-12-04:
we hate it and we love it....

Happy printing