bigtreetech/BIGTREETECH-TouchScreenFirmware #1343: It is required to add the unmount script after exiting the print menu, and after finishing printing or stopping.

Foxcor, 2020-12-04:
Now I checked it again. When you press the button in the menu item
Print-> SD Card TFT-> file list, after that all, files for SD Cloud are not available. Those. after mounting the SD card with the display, it becomes inaccessible to anyone.
It is required to add the unmount script after exiting the menu, and after finishing printing or stopping.

_Originally posted by @Foxcor in https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1335#issuecomment-738962627_

oldman4U, 2020-12-04:
I am not sure this is really something which can be fixed in the TFT firmware, but maybe someone can check it.

@digant73
@kisslorand
@mehmetsutas 

Is there a way to check what happens file system/storage wise once a print has been canceled or finished. Is there some kind of release needed. The problem is, that once the sd card on the cloud device has been accessed, a restart of the printer is needed to be able to access the device again. Not sure why this does not work, but works fine with all other storage options.

Maybe you can help

Thank you

kisslorand, 2020-12-04:
It can be done. I have a BTT SD Cloud but I was never able to make it work with my MKS TFT 28., so even if I statr working on this I cannot test it. :(
Another problem is that it is such a rare situation that it's not very probable anyone would invest time and effort in this.  A regular SD card inserted does not need Init and ReInit after a print or cancel. 
Maybe a faster (but inconvenient) solution would be to remove and reinsert the BTT SD . It would trigger a FatFS DeInit on th e TFT.

oldman4U, 2020-12-04:
Thank you for your answer.

So the problem is really in the TFTs firmware and how it handles file system events!? Every manual interaction drives the whole concept ad absurdum.;-)

Anyway, I added it to the FR list and maybe one day this piece of hardware will be useful.

Thanks again

kisslorand, 2020-12-04:
I would call it the TFT's firmware problem, rather an unfortunate hardware combination. It's like 2 people wanting to drink from the same cup. One has to put it down so the other can pick it up.

mehmetsutas, 2020-12-05:
The same issue exist when the SD slot on the TFT is used as the board SD. However this is a very rare case, one either prefers a TFT SD print or BOARD SD print. I don't think both board sd and tft sd prints would be used frequently by the same user. That's why I did not go over it.

For the SD_CLOUD , using a TF_CLOUD and mounting the TF_CLOUD on the BTT SKR board would be the easiest solution. Enabling the on board TF slot on Marlin would be needed.

Since this is a very rare case and special hardware is needed to test it, I think we should wait for a interested programmer who have the necessary hardware (BTT SD CLOUD) for solution.

Foxcor, 2020-12-05:
@mehmetsutas
@oldman4U 
Those. it turns out if a person buys such a [https://aliexpress.ru/item/1005001369454754.html](kit) he will not be able to use it normally ?! We'll have to use "churches". )))
It's just that SD Cloud is a great option for me, the case is completely closed and nothing extra sticks out. Perhaps, if someone sees my version, and I communicate on the forum of printer owners, seeing the capabilities of SD Cloud, they will buy it. And so it turns out a good thing, but with nuances. No need to use SD card extenders or print a separate screen case.
[image]

kisslorand, 2020-12-05:
Well... What can I say? I bought the BTT Cloud as well and it's useless for me. I cannot use it with nothing! Not only in the printer, in PC, tablet, raspberry, etc. It doesn't connect  to the WiFi when it is inserted into the SD slot, only if it is powered by USB, which of course is useless for me, it's not its intended purpose. 
So yeah, it's a gamble when you buy that product.
I would like to emphasize again that it's nothing wrong with the TFT it wasn't meant to be used with BTT SD Cloud. I agree, it would be nice, it's just not the case.

oldman4U, 2020-12-05:
Why you haven’t contacted your dealer?

kisslorand, 2020-12-05:
Because it's Aliexpress. :)
It's a known bug, it was never solved, no one was able to fix if they had this issue.

oldman4U, 2020-12-05:
Even AliExpress dealers have to take care. Refund or new item.

kisslorand, 2020-12-05:
@oldman4U I just saw you were aware of this "no WIFi when used in printer".
https://github.com/bigtreetech/BTT-SD-TF-Cloud-V1.0/issues/17

oldman4U, 2020-12-05:
> bigtreetech/BTT-SD-TF-Cloud-V1.0#17

I wrote a whole conclusion after testing it for almost one week. See #18.
But never tried smb and now I can not find it anymore 🥴

Foxcor, 2020-12-05:
@kisslorand 
> Not only in the printer
Why?
At pin 4, 3.3V must be applied and should work.

kisslorand, 2020-12-05:
Yeah it should but it doesn't.
When connected to SD Card (except the printer) of PC I can access the SD card inserted to it, so it has power.
Anyway, it's not the proper place to truobleshoot my device.

kisslorand, 2020-12-05:
@oldman4U 
> I wrote a whole conclusion after testing it for almost one week. See #18. But never tried smb and now I can not find it anymore 

Sorry but I cannot see how #18 is related to this issue.

oldman4U, 2020-12-05:
Cloud repo.

Can not copy it on the phone.

kisslorand, 2020-12-05:
The PR  #18 has the title: "User-Definable 12864 Mode Colors & Color Cleanup"

oldman4U, 2020-12-06:
I found my SD Cloud device and tried to mount the device using smb instead of http and I was not able to access it. This hardware is a piece of s... and it is my opinion that every minute invested to adjust the TFT firmware is waste of time and money. Besides the TFT35 v2, this is the second failed product from BTT. Useless.

kisslorand, 2020-12-06:
Just as  I said, I threw it away and moved on.

closed 2021-02-11 by stale[bot]