bigtreetech/BIGTREETECH-TouchScreenFirmware #1350: [Q] (Your Question?)

Chukozor, 2020-12-06:
Is the time displayed on the TFT supposed to be the actual printing time since pressed "print" ? If so there is a big issue on commit 9f59d037 : it displays like 5 times the actual printing time since pressing "print". What's the situation ? Thanks

oldman4U, 2020-12-06:
Please update the title of this ticket.

oldman4U, 2020-12-06:
Are you printing from on board SD, TFT SD or uDisk?

Short movie could also be helpful, like Marlin version and gcode file.

Thank you

Chukozor, 2020-12-06:
I print from TFT SD, with marlin 2.0.7.2 on a SKR 1.3.
[Here a gcode I printed with the issue occured](https://cdn.discordapp.com/attachments/763032939923111937/785214893917601821/sample-file.gcode) (but every gcode I use has this same issue)

oldman4U, 2020-12-06:
Which TFT do you have?
Does it count 5sec every second or 1sec every 1/5th of a second?

Using a TFT35 E3 v3 with the firmware from Dec. 3rd, this works wonderful and has not the issue you reported.

oldman4U, 2020-12-06:
But I have a problem with your gcode file, which heats the T and the Bed, does a homing and stops then for about 1min before it starts. Attached a gcode file, please check if the time is correct using it and let us know.

Thank you

[3minwithOutEndgCode.gcode.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5649434/3minwithOutEndgCode.gcode.zip)

Chukozor, 2020-12-07:
I have a TFT35 3.0. I can't see the diference between "count 5sec every second or 1sec every 1/5th of a second".
Yes that's normal : I wait for the nozzle to ooze outside the print area when heating up.
I'll print this tomorrow and come back here.

oldman4U, 2020-12-07:
How does it count so fast by counting 5x per second or counting 0, 5, 10, 15,... every second?

Agent-616, 2020-12-07:
I have this same issue. I came in about 20 to 30 mins after starting a print and it showed having been printing for over 2 hours. SKR E3 mini ver 1.2 and the TFT35 E3 on an Ender 3 Pro with BL Touch

oldman4U, 2020-12-07:
Can you tell me how it counts? Any info how this can be reproduced?

Agent-616, 2020-12-07:
> Can you tell me how it counts? Any info how this can be reproduced?

tbh I am unsure what is happening to make it do this. I am 4% into a print that is meant to be taking about 28hrs and it is already showing 4 hrs 19 mins. But it has been printing no more than about 2 hours.

oldman4U, 2020-12-07:
Have you made a 2hplus print before?

Is the elapsed time counting faster than real-time?

Agent-616, 2020-12-07:
I printed a small cable controlling bracket earlier today that Cura said would take approx 58 mins. on it completing the screen showed the print time was almost 3 hours.

I do not have the start or end gcode settings turned on in the firmware of my tft35 screen.

oldman4U, 2020-12-07:
So the print starts with elapsed time 00:00:00 but after 58min the print time shows 3h, so there are 4 possible ways to get this:

A: The print does not start with 00:00:00 but with 02:02:00 and counts from there correctly
B: The print starts with 00:00:00 but increases 3sec instead of 1sec for every second
C: The print starts with 00:00:00 but increases for 1second three times per second
D. Something else which is beyond my imagination

A, B, C or D?

Agent-616, 2020-12-07:
Looking at my current print it is counting the seconds fine. It is almost as if it adds 2 hrs at the start of a print while the bed and nozzle are heating. So I am thinking A from your list. I started the shorter print earlier and then had to go out of the room to my daughter. So I missed how it happened. But my current print is counting time correctly. It just has about 2 hrs extra on the clock.

oldman4U, 2020-12-07:
There is a PR coming related clearing the print time at the start of each
print.

Agent-616, 2020-12-07:
Is this gonna be a new version of the firmware? Any idea when it is going live?

oldman4U, 2020-12-07:
Please see #1333

Chukozor, 2020-12-08:
indeed that seems to be what happens : no time issue on first print after boot but issue of print time since the second print.
Do I close the ticket ?

oldman4U, 2020-12-08:
Sounds great.

You can close it now or once the fix is available. Whatever works better
for you

Thank you

oldman4U, 2020-12-11:
@Chukozor

The fix is now available in master firmware. Please check and close this ticket afterwards.

Thank you

closed 2020-12-11 by Chukozor