bigtreetech/BIGTREETECH-TouchScreenFirmware #1352: [BUG] TFT35 E3 V3.0 Freezes after changing from Marlin mode back to Touch mode on SKR Mini E3 V2.0

cirorebelo, 2020-12-06:
### Steps to reproduce

1. Start the Printer
2. Change  from Touch mode into Marlin emulation mode by holding the knob and selecting Marlin.
3. After checking the screen, no changes made...
4. Change back into touch mode by holding the knob and selecting Touch mode.
5. TFT freezes after a second.

**Expected behavior**
Touch mode activates normally and reconnects to printer

**Actual behavior**
After a second or so, and at the same time: the TFT Beeps; a yellow dot notification mark appears on the top right corner; and the TFT freezes while continuously beeping and stays frozen/beeping till powered down.

### Hardware Variant
TFT35 E3 V3.0

### TFT Firmware Version & Main Board Firmware details
TFT:
**All pre-compiled TFT35_E3_V3.0.26.x.bin I have download after Nov-30-2020.**

The last one I have that does not freeze was downloaded on Dec-1-2020, the file is dated Nov-30-2020 on the zip, and on the info screen it shows "TFT35_E3_V3.0.26.x Sep 12 2020"
The one downloaded on 7-11-2020, dated Set-4-2020 on the zip, also does not freeze.

Board/Marlin
SKR Mini E3 2.0 running Marlin 2.0.7.2  ( **with the options requested on the tft config.ini** ) 
 Also Tried Marlin 2.0.6.1 same freeze.

### Additional Information
Please request as needed and I will provide the files.

radek8, 2020-12-07:
If you start the printer in touch mode, does the display work normally?

Upload the current FW to the TFT display and Marlin bugfix 2.0.x to the SKR
Enable in config.ini
serial_always_on: 1
Do not use the rotary knob but the touch screen to switch between modes.

cirorebelo, 2020-12-07:
> 
> If you start the printer in touch mode, does the display work normally?
> 

It does. It only froze when switching to Marlin and then back to Touch mode.

> Upload the current FW to the TFT display and Marlin bugfix 2.0.x to the SKR
> Enable in config.ini
> serial_always_on: 1
> Do not use the rotary knob but the touch screen to switch between modes.

That did the trick. In fact, after I have tried your suggestion to set serial_always_on", I managed to also test the following:
- Using the rotary knob - **it works fine**
- On my regular (non bugfix) 2.0.7.2 and  **it also worked**

### Additional information
In case it might my useful: pre-compiled  firmware I downloaded before Dec-1-20 did not require "**serial_always_on: 1**" and worked just fine with "**serial_always_on: 0**" without the freezing. 
Newer firmware required that change to work.

closed 2020-12-07 by cirorebelo

radek8, 2020-12-08:
Yes Marlin without error correction works well, but it may contain known errors.
I do not recommend using the rotary knob with SKR mini E3 V2
if you don't make another adjustment, it will have a side effect on Marlin.
I definitely do not recommend clicking on it in touch mode!