bigtreetech/BIGTREETECH-TouchScreenFirmware #1353: Question about "fil_runout:" in the config.ini file

slwise, 2020-12-06:
I have a TFT35 E3 V3.0 updated with the latest firmware which includes the filament sensor (#1336)bug fix. I also have the BTT SFS connected to the TFT. I changed the "fil_runout:0" in the config.ini file to "fil_runout:2" for the smart sensor. After the firmware update is done installing, I go into the settings menu and it only shows the filament sensor as being "on" and not set as "smart". I have to manually select "smart". Is this the correct way it is supposed to work? Would this have any effect on how the sensor responds?

I am still getting false positives when printing. I think the main reason may be I have a Micro Swiss direct drive/hotend and the SFS is mounted where the extruder use to be on my Ender 3 Pro. I feed the filament from the side due to height limitations where the printer is located. I have the SFS connected to the direct drive through a bowden tube as a guide. I have tried different values in the "fil_noise_threshold:" and "fil_runout_distance:" with limited success. If anyone has any suggested settings for a direct drive I would really appreciate it.

oldman4U, 2020-12-11:
Hi.

Could you please reset your settings under Menu - Settings - Feature - Reset default settings to see if this solves your problem.

Thank you

slwise, 2020-12-12:
I reset the TFT as you described and also tried reseting the TFT with the reset.txt file. Both set the filament runout sensor to off. When I load the config.ini with "fil_runout:2" then feature menu still shows the filament runout sensor set to "on" not "smart". This is a minor issue. I'm just trying to get the SFS to work with a direct drive extruder and just wanted to make sure this wasn't affecting it's operation. BTW, I did just load the latest precompiled version of the firmware and it still sets the runout to "on" and not "smart". Thanks for your help.

oldman4U, 2020-12-12:
OK. After installing a new firmware also to the Reset default settings ...

slwise, 2020-12-12:
I'm sorry. I meant to say I tried it before the firmware update and after with same results.

oldman4U, 2020-12-12:
OK. I can confirm this behaviour. When you define smart in config.ini, on TFT it shows ON, but not smart.

Any other issue?

oldman4U, 2020-12-12:
@guruathwal Is this your area?

slwise, 2020-12-13:
No other issues. Thanks.

guruathwal, 2020-12-13:
Yes,
Fixed

Regards,
GuruAthwal

oldman4U, 2020-12-13:
👍🏻

PR #1384 (2020-12-15): Add cold extrusion check, bug fixes, Cleanup and remove duplicate codes
- Cleanup and remove duplicate codes.
- Fix `fil_runout:` read incorrectly (also mentioned in #1379).
- Change font colour logic for live icons.
- Add cold extrusion check.
- Fix load-unload popup (#1378).
- Remove duplicate global variables.
- Fix wrong settings after (#1333).

Fixes #1353, #1378
PR #1379 (2020-12-15): resolves #1377 + cleanup
**BUG FIXES:**
* resolves #1377 (Mesh Editor freeze after one of the last merged PR)
* resolves #1364: Z Offset will start from probing Z0 point instead of absolute Z0 point. It means original Z offset is not applied when Z offset is enabled on Z offset menu
* resolves #1363 (probably solved now)
* resolves #1353 (value "SMART" for "fil_runout" is now correctly read from config.ini)
* MBL safe starting Z point: initial Z points are now set to 1mm instead of Z0. This should make more safe the MBL process in case of warped bed or not good manual leveling
* some cleanup

 [... cut]
closed 2020-12-15 by bigtreetech