bigtreetech/BIGTREETECH-TouchScreenFirmware #1354: [Q] Fan speed, feedrate, flowrate updates frequency?

mehmetsutas, 2020-12-06:
With PR #1283 the problem with these updates have been somehow solved. Now, the update frequency have been reduced by not sending another gcode to the printer until a certain time pass. 

This was a good solution but I think sending the incremental changes (not full, half, stop actions) to the printer when the screen is exited would even be a better solution for decreasing the amount of communication between the printer and the TFT.

If you like this logic, please inform me. I am already using this logic on my printer and can create a pull request. Actually, #1283 were introduced just before my PR. :)

@digant73
@kisslorand
@oldman4U

kisslorand, 2020-12-06:
I myself thought about this very thing. I came to the conclusion that people will freak out if they do not see immediate reaction of the printer. This is why I choose 500ms delay because it is really a long time for the MCU but is "immediate" for a human.
I am all for accepting/sending  only the final value, but 90% of the people will see it as a lag/bug/flaw.
My daily job is to find errors/problems so I am used to stress devices with the dumbest inputs possible and to make stuff idiot-proof.
I know, it's hard to detach yourself from the one who made the thing and know its quirks and test it as an average/regular user.

closed 2020-12-07 by mehmetsutas

oldman4U, 2020-12-07:
Why are you closing it?