bigtreetech/BIGTREETECH-TouchScreenFirmware #1356: overextrusion caused by tft35

Stormwing01, 2020-12-07:
### Description
I have confirmed that the tft35 v3 e3 on my skr mini v2 causes over extrusion at the start of extrusion on infill and walls, but when I use my old ender 3 display, all overextrusion disappears.

The overextrusion only goes as high as my z fade height and then suddenly disappears.

### Steps to reproduce

I honestly have no idea.

**Expected behavior**
Smooth, even prints, no over extrusion, following the settings that I have set.

**Actual behavior**
The starts of extrusions are basically blobs 2 to 6 times larger than the nozzle width. The amount almost seems to be at random for the blobs, but the z fade overextrusion stays constant at about 30 percent (seemingly) more extrusion than the rest of the print.

### Hardware Variant
TFT35 E3 V3

### TFT Firmware Version & Main Board Firmware details
Marlin 2.0.7.2

tft35 V3.0 E3.26.x

### Additional Information

I literally just flashed the same version of firmware as is on the github page, as is. I did not modify it at all, except to change the fan controller count in config.

I will try to get video, but my camera is awful and the last time I tried, the video was a mess.

Stormwing01, 2020-12-07:
here is a video I finally managed to get... well enough.

it is an unlisted video: https://youtu.be/8BD3jPL5tHY

as well as some images: 
[image]
[image]

even if it is not a bug, i'd love to know what I can do to stop this issue.

my wiping, coasting and priming are disabled in cura and I have linear advance on. (K=0.31)

radek8, 2020-12-07:
What speed do you use on the serial line

Stormwing01, 2020-12-08:
> 
> 
> What speed do you use on the serial line?

I use 115200, same as is set up in my marlin firmware.

radek8, 2020-12-08:
I would try to reduce the print speed or increase the serial communication speed to 250000

Stormwing01, 2020-12-08:
I'll give that a shot (setting the baudrate higher) but im already printing at 35mm/s (max for printing and 260 max for travel moves) and the filament is as smooth and easy to press through the nozzle as if it was butter. if it was the speed, the blobbing wouldn't be there for just 4 or 5 lines, after printing 2 perfect first layers and then printing perfectly after those 4 or 5 faulty layers. as I also said above, it only appears when I've activated both linear advance and set the abl value in the tft's menu to 1 (on)

I doubt that the print speed is an issue, because it's as if the nozzle is lagging at the start of the lines.

radek8, 2020-12-08:
Try to set the buffer according to the attachment

[image]

Stormwing01, 2020-12-08:
I set marlin to 250000 as well as the tft in it's settings, but now it absolutely blasts my ears out with constant screaming as soon as it makes a connection with the printer until I press one of the icons on the menu.

I just reflashed the tft as well, but there are still some menus missing, and there are still menus present that I have not activated in marlin. (i specifically mean the touch mode by the way.)

I will try to set the buffer now, but im setting my baud back to 115200 and then I will check to see if the overextrusion is gone as well.

by the way, is there some setting I need to change in marlin specific to btt's tft35 v3 E3? maybe that's the issue or something? asking just in case.

radek8, 2020-12-08:
Read the comment in config.ini

You have interference on the serial line at 250000.
Use a shielded cable. The original cable is bad.

Stormwing01, 2020-12-08:
> 
> 
> Read the comment in config.ini
> 
> You have interference on the serial line at 250000.
> Use a shielded cable. The original cable is bad.

I doubt its interference, because it immediately stops and completely disappears the moment I press... literally anything, and its a scream, like when you get an error, but constant. a deliberate beep, like a programmed beep, but one that doesn't stop. I sometimes get them on 115200 as well, as well as the tft randomly asking me to go to marlin mode whenever I go between menus a lot, but it's rare and manageable.

The blobs are still there even with the buffer changes. if anything, they seem worse. now its like the printer is hanging up and lagging, until the first 5 layers have passed. The menu is still its own thing as well.

radek8, 2020-12-08:
How many steps / mm are you set in the Z axis?

Stormwing01, 2020-12-08:
> 
> 
> How many steps / mm are you set in the Z axis?

it is about 401 per mm... Im gonna print a test to see if the z  steps per mm is correct, but that might take a little while. I usually print that test quite tall.

radek8, 2020-12-08:
401 is fine

Stormwing01, 2020-12-08:
> 
> 
> 401 is fine

yeah, the 150mm rod I printed came out at 150.4mm which isn't too bad at all. ima lower the steps per mm a little, but not much

radek8, 2020-12-08:
I was wondering if you didn't have a high step value.
Too high a value could exploit the processor in the first layers when compensation is applied.
Then there would be a short stop and more material leaking out.

Stormwing01, 2020-12-09:
> 
> 
> I was wondering if you didn't have a high step value.
> Too high a value could exploit the processor in the first layers when compensation is applied.
> Then there would be a short stop and more material leaking out.

That's alright. but yeah, onto the next theory, lol

Stormwing01, 2020-12-15:
> 
> 
> Any news?

Nope. 

The z-height overextrusion disappeared when I did mesh bed levelling and had atleast one point set to 0 and the rest either over or under 0mm instead of every single measurement being above or below 0mm. my highest point now is 0.06mm and my lowest mesh point is -0.04 and that seemed to help a bunch... somehow.

the blobbing? it's still there, but it seems to be less with a careful bed mesh.... which is just ridiculously odd.

I've otherwise concluded that people don't seem to care about this bug, or really have no clue what's up, same as me.

oldman4U, 2020-12-16:
Maybe others can not see it because there is a little setting which makes the difference. Hope you will fix it one day.

kisslorand, 2020-12-16:
It's hard or impossible to fix something that cannot be reproduced.

oldman4U, 2020-12-16:
Usually it is combination of things and not a single setting, even changing a singe setting sometimes fixes it...;-)

We can not reproduce it and because of this I did not add it to the Bug list. Hopefully we can reproduce it in the future.

Stormwing01, 2020-12-17:
that's all fair. I do think it has something to do with the eeprom because it disappears when I comment out "eeprom_settings" in marlin and then reappears when I uncomment it.....  and right now it disappeared when I clicked "initialize eeprom" in marlin mode.... so far I've been nearly bug free, but I have no clue how to reproduce it. if it pops up again, I'll dinkle around a bit and do my best to see what causes it.

thanks people.

radek8, 2020-12-17:
Can't have eeprom damaged in processor?
I would try to move eeprom to SD card
Simulate eeprom to SD and see if it solves your problem.

Stormwing01, 2020-12-17:
as I said above, my issue is now completely solved with just pressing the "initialize eeprom" in marlin mode in the tft. my settings are saved just fine too.

buut I'll definately give your idea a shot when it pops up again, thanks!

by the way, how do I save eeprom settings to sd? I cannot find any obvious setting like that in marlin, and googling it is basically useless cos all my search results give me are bugs and no guides.

radek8, 2020-12-17:
> 
> by the way, how do I save eeprom settings to sd? I cannot find any obvious setting like that in marlin, and googling it is basically useless cos all my search results give me are bugs and no guides.

To a file that defines the pins of your marlin board
fill in the line:
#define SDCARD_EEPROM_EMULATION
I have defined it as follows:

[image]

The onboard sd card is used.
It should be defined:
#define SDCARD_CONNECTION ONBOARD
in the configuration_adv.h file
The SD card must be inserted into the slot when turning on, saving and loading.

Marlin 32b boards usually do not have a separate eeprom memory and simulate an eeprom in Falsh memory.
The number of fash overwrites is limited, and if you exceed the number of writes by frequently writing values in the simulated eeprom, irreversible memory corruption occurs, which is part of the processor.
Therefore, it is good to use another medium such as eeprom. This will extend the life of the Marlin board.
If the SD card is damaged, I will simply replace it.
A better option is to use the eeprom on the I2C connector if you have it on the board.
[image]

Stormwing01, 2020-12-17:
That file is for the skr v1.4 though. I have the skr mini e3 v2.

radek8, 2020-12-17:
pins_BTT_SKR_MINI_E3_common.h

radek8, 2020-12-17:
Does your board have an integrated I2C eeprom?

Stormwing01, 2020-12-17:
pins_BTT_SKR_MINI_E3_V2_0.h does not have "sdcard_connection".

it has integrated eeprom, but that's besides the point. I want to know how I can make it save the eeprom settings to sd card to see if that fixes my problem from earlier incase it arises again because the eeprom itself seems to be the cause of my earlier issues.

this is all that "pins_BTT_SKR_MINI_E3_V2_0.h" contains, and you'll see there are no sdcard related settings in it:

#pragma once

#define SKR_MINI_E3_V2

// Onboard I2C EEPROM
#if NO_EEPROM_SELECTED
  #define I2C_EEPROM
  #define MARLIN_EEPROM_SIZE 0x1000                 // 4KB
  #undef NO_EEPROM_SELECTED
#endif

#include "pins_BTT_SKR_MINI_E3_common.h"

#define BOARD_INFO_NAME "BTT SKR Mini E3 V2.0"

// Release PA13/PA14 (led, usb control) from SWD pins
#define DISABLE_DEBUG

#define NEOPIXEL_PIN                       PA8   // LED driving pin

#define PS_ON_PIN                          PC13  // Power Supply Control

#define FAN1_PIN                           PC7

#ifndef CONTROLLER_FAN_PIN
  #define CONTROLLER_FAN_PIN               FAN1_PIN
#endif

/**
 * TMC220x stepper drivers
 * Hardware serial communication ports.
 */
#if HAS_TMC_UART
  #define X_HARDWARE_SERIAL  MSerial4
  #define Y_HARDWARE_SERIAL  MSerial4
  #define Z_HARDWARE_SERIAL  MSerial4
  #define E0_HARDWARE_SERIAL MSerial4
#endif

radek8, 2020-12-17:
Try this

[image]

Stormwing01, 2020-12-17:
if I add that definition I get a list of errors and compiling fails.

In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\HAL.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_swspi.cpp:21:
Marlin\src\HAL\STM32F1\dogm\../../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."       
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\MarlinSerial.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\SPI.cpp:43:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\Servo.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_if_iic.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\HAL_SPI.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_flash.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_sdcard.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_bl24cxx.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_swspi.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_wired.cpp:27:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\MarlinSerial.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_if_iic.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\Servo.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\SPI.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_flash.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_bl24cxx.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_sdcard.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_wired.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\onboard_sd.cpp:16:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
========================================================== [FAILED] Took 10.53 seconds ==========================================================
Environment           Status    Duration
--------------------  --------  ------------
STM32F103RC_btt_512K  FAILED    00:00:10.528

Stormwing01, 2020-12-17:
if I uncomment and add that, I get a list of errors and the compilation fails.

In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\MarlinSerial.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\Servo.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\SPI.cpp:43:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\HAL.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_bl24cxx.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_swspi.cpp:21:
Marlin\src\HAL\STM32F1\dogm\../../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."       
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_if_iic.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_sdcard.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\HAL_SPI.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_flash.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\MarlinSerial.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\Servo.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_wired.cpp:27:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_bl24cxx.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\SPI.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_swspi.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_if_iic.cpp.o] Error 1
Marlin\src\HAL\STM32F1\eeprom_sdcard.cpp: In static member function 'static bool PersistentStore::access_start()':
Marlin\src\HAL\STM32F1\eeprom_sdcard.cpp:56:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (; bytes_read < MARLIN_EEPROM_SIZE; bytes_read++)
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_sdcard.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_flash.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_wired.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\onboard_sd.cpp:16:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2218:6: error: #error "Please select only one method of EEPROM Persistent Storage."
     #error "Please select only one method of EEPROM Persistent Storage."
      ^~~~~
========================================================== [FAILED] Took 10.23 seconds ==========================================================
Environment           Status    Duration
--------------------  --------  ------------
STM32F103RC_btt_512K  FAILED    00:00:10.225

radek8, 2020-12-17:
Try to comment on this as well

[image]

Stormwing01, 2020-12-17:
that seems to have done the job perfectly, thank you very much!

I also had to comment out #define SDCARD_READONLY in configuration_adv.h otherwise I get a lot of errors, but everything works perfectly.

oldman4U, 2020-12-18:
Looks good.

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2020-12-20 by Stormwing01