bigtreetech/BIGTREETECH-TouchScreenFirmware #1358: [BUG] Onboard TF and TFT SD are changed

Bastlwastl84, 2020-12-07:
Skr Mine E3 V2
TFT 35 E3 V3
ESP01 with ESP3D V3.0

The ESP3D Software can see it correctly, at TFT Firmware it's turned.

Config is here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1357

oldman4U, 2020-12-07:
Have you reported the same on luc's repository?  v3 is very new and would expect some bugs are still present.

Bastlwastl84, 2020-12-08:
This error with not printing happened before some weeks with esp3d v2.1 and not used the webinterface, too. There i it happened, like yesterdayas i canceled it with the tft. Only a power off makes it useable. I opened esp3d to check and tested the new interface.
-------- Ursprüngliche Nachricht --------Von: oldman4U <notifications@github.com> Datum: 08.12.20  00:46  (GMT+01:00) An: bigtreetech/BIGTREETECH-TouchScreenFirmware <BIGTREETECH-TouchScreenFirmware@noreply.github.com> Cc: Bastlwastl84 <thomas.aumeier@gmail.com>, Author <author@noreply.github.com> Betreff: Re: [bigtreetech/BIGTREETECH-TouchScreenFirmware] [BUG] not printing after aborting a printjob, Onboard TF and TFT SD are changed (#1358) 
Have you reported the same on luc's repository?  v3 is very new and would expect some bugs are still present.

—You are receiving this because you authored the thread.

oldman4U, 2020-12-08:
Ok. Which firmware have you been using and how is this reproducible?

Bastlwastl84, 2020-12-08:
Marlin 2 Bugfix and tft FW from 5.12.Von meinem Sprechknochen gesendet.

oldman4U, 2020-12-08:
Half of the questions answered.

Bastlwastl84, 2020-12-08:
How this happened: i dont know it happened. I stopped and tried starting again and filmedVon meinem Sprechknochen gesendet.

oldman4U, 2020-12-08:
But you are somehow able to reproduce it on your printer!?

Bastlwastl84, 2020-12-09:
> But you are somehow able to reproduce it on your printer!?

I don't know what did it. 
Actually a Printjob now is running with Marlin Mode for more than 1 day Printtime. After that i can retry.

oldman4U, 2020-12-09:
Please do so.

Bastlwastl84, 2020-12-11:
> Please do so.

downloaded now new firmware and test it in some minutes

oldman4U, 2020-12-11:
Could you please reset your settings under Menu - Settings - Feature - Reset default settings.

Bastlwastl84, 2020-12-11:
> Could you please reset your settings under Menu - Settings - Feature - Reset default settings.

now that the TF onboard and TFT are changed are gone now after resetting the tft. after updating the FW i coudn't connet to Marlin Mode (now that error is gone, too. i'll try some more things now

Bastlwastl84, 2020-12-11:
Everything reseted (board and tft) and i had the same error again with time is running, but not printing.I did before testing some M150 light settings before and than tft sd->printVon meinem Sprechknochen gesendet.

oldman4U, 2020-12-11:
How is this problem reproducible?

Bastlwastl84, 2020-12-11:
I didn't had found out what does start the issueVon meinem Sprechknochen gesendet.

oldman4U, 2020-12-11:
I would recommend to close this ticket and start a new one with more
information regarding the print not start issue.

Bastlwastl84, 2020-12-17:
Bug is gone

closed 2020-12-17 by Bastlwastl84

oldman4U, 2020-12-17:
Because of...?