bigtreetech/BIGTREETECH-TouchScreenFirmware #1363: [BUG] (short description)

holtt, 2020-12-09:
### Description

When activating Z Offset via TFT firmware with existing negative BLTouch offsets, the TFT sends GCODE commands to the printer which seem to drive it beyond the stop movement limits.   

### Steps to reproduce

1. Via GCode, configure X, Y and Z offsets for BLTouch location to all be negative
1. Via touch screen UI, go to Menu > Movement > Bed Leveling > Z Offset
6. By default, Z Offset will be in "OFF" state, and correct negative Z Offset displayed at top of screen
7. Press Z Offset OFF button to turn it on

**Expected behavior**

- Printer homes X, Y, and Z to zero based on limit switches (controller board firmware to use BLTouch for zeroing Z is NOT being used)
- Z Offset modification is now enabled
- Z offset value can be changed via the UI and fine tuned to optimize bed clearance as needed

**Actual behavior**

- Printer homes X, Y, and Z axes, stopping at limit switches
- Printer then proceeds to try and move Z axis below level of bed and past stops
- Neither X or Y homing put the extruder tip on the bed, so there is no impact with the bed, however extruder tip clearly moves to a position several MM below the bed
- Horrible noises ensue due to it trying to exceed mechanical limits
- Printer powered down at this point to avoid mechanical issues
- As a result, one is never able to actually modify the Z offset value via the UI, and must use GCode messages to change values.

Note that printer and BLTouch work just fine and do exactly what they should do when printing.  Printer is able to automatically probe print bed, compensate accurately, and otherwise print without issue.

### Hardware Variant

- Ender 3 Pro
- SKR mini E3 V2.0
- TFT 35 V2
- BLTouch 3.1 with mount-specific X, Y and Z offsets set via direct GCode and all values negative (-42, -10, -1.18)

### TFT Firmware Version & Main Board Firmware details
SKR mini E3 V2 firmware with BLTouch NOT used for z homing (https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/firmware-bltouch.bin)

TFT 35 V2 firmware (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/BIGTREE_TFT35_V3.0_E3.26.x.bin)

### Additional Information

Reviewing the TFT firmware source code, when you select the Z Offset ON option via the TFT, it calls probeOffsetEnable() in src/User/API/ProbeOffsetControl.c 

When you do this, here is the gcode that is sent to the printer.  I've added what function the gcode is generated in via # type comments.

    ; In probeHeightEnable()
    M211 S0; disable software end stops

    ; In probeOffsetEnable()
    G28; home, which takes all axes to the point where they hit their stops

    ; In probeHeightStop()
    G91; set relative position mode
    G1 Z<z_probe offset;  move nozzle to relative z offset coordinate
    G90;  return to absolute position mode

    ; In probeOffsetEnable()
    G91; set relative position mode
    G1 X<probe offset> Y<probe offset>; move nozzle to x and y probe coordinates

    ; In probeHeightStart()
    G90; absolute position mode
    G1 Z0;  move nozzle to absolute Z0 coordinate
    G91;  back to relative position mode

I believe what is happening is this...
- It's disabling software end stops
- It is taking X, Y, and Z all the way to zero based on the mechanical stops (micro-switches) via G28 command
- It then tries to move relatively by the X, Y and Z offsets
- Since the offsets are negative, it is trying to move PAST the X/Y/Z mechanical stop limits, which causes the noise

Here is a Dropbox link to a video showing the issue -> https://www.dropbox.com/s/6mv4gh2ja8723hk/Video%20Dec%2008%2C%208%2049%2021%20PM.mov?dl=0

I am not 100% certain which axis is being exceeded to cause the noises.  It may actually be the X axis limit, since it is larger.

oldman4U, 2020-12-09:
Please update the title.

digant73, 2020-12-09:
@holtt 
the logic of z-offset menu is correct. When you enable z-offset from z-offset menu, the nozzle is first homed by G28. The current z-offset is also applied by G28. After homing, the nozzle is moved to Z 0 position. So, if the nozzle touch the bed when you enable z-offset it means that you are starting with a wrong z-offset. Try to increase it from the "Probe Offset" setting in the "Parameter" menu. and then you can use z-offset menu.
The end-stops are disabled just to allow to lower the nozzle (so to lower z-offset). They are eventually re-enbled when you disable z-offset.
Disabling z-offset simply apply the new changes (G28 is executed and it applies the new z-offset)

Foxcor, 2020-12-09:
Check your settings in Marlin
#define ENDSTOPPULLUP_ZMIN_PROBE
#define Z_SAFE_HOMING
//#define MIN_SOFTWARE_ENDSTOP_Z

closed 2020-12-15 by bigtreetech

Octet-nl, 2020-12-18:
@holtt 
(see also conversation at #1379)
In `ProbeOffsetControl.c`:
As `probeOffsetEnable` and `probeOffsetDisable` are only called (afaik) from the menu, it seems safe to change these functions to only contain one statement, `probe_offset_enabled = true` and `probe_offset_disabled = false`. The whole homing and moving can be removed. This way the menu choice does what it advertises, switching the value on and off in the menu without any unexpected side effects. Until a better solution is provided, I will try this one.

digant73, 2020-12-20:
try PR #1415. It should solve your problem (set the new parameter on config.ini to 0)

Octet-nl, 2020-12-21:
@digant73
Works, Thanks!

PR #1379 (2020-12-15): resolves #1377 + cleanup
**BUG FIXES:**
* resolves #1377 (Mesh Editor freeze after one of the last merged PR)
* resolves #1364: Z Offset will start from probing Z0 point instead of absolute Z0 point. It means original Z offset is not applied when Z offset is enabled on Z offset menu
* resolves #1363 (probably solved now)
* resolves #1353 (value "SMART" for "fil_runout" is now correctly read from config.ini)
* MBL safe starting Z point: initial Z points are now set to 1mm instead of Z0. This should make more safe the MBL process in case of warped bed or not good manual leveling
* some cleanup

 [... cut]
PR #1415 (2020-12-24): Configurable XZ probing point on Z Offset menu (fixes #1363)
Based on further clarifications on bug #1363, The problem was not on Z axis but on XY axis.
On some HW configuration, the XY probing point set for homing is not reachable by the nozzle applying the XY offset configured on Marlin FW.
This PR allows to configure if nozzle must be moved or not on the XY probing point when the Z offset tuning process is started on Z Offset menu

fixes #1363 (definitively)
fixes #1416 (absolute coordinates and soft endstops were not restored after MBL process was finished)
fixes #1420
fixes #1425
commit 7c43a71e (2020-12-24): Configurable XZ probing point on Z Offset menu (fixes #1363) (#1415)