bigtreetech/BIGTREETECH-TouchScreenFirmware #1364: [FR] (feature request title)

Chukozor, 2020-12-09:
I have TFT35 3.0, I like the Zoffset menu which enable to live test it, but I think it would be great to add some safety option. What I mean is that if the Zffset is at the moment of -3mm and if I make some changes on the carriage and need something like a -2.8 Zoffset the nozzle will dive into the bed of 0.2mm (-2.8 -(-3)) before I can add the 0.2mm to the Zoffset on this menu. So it's a serious problem here. Actually I send some Gcode to change the Zoffset because of that issue.

I propose to add an option to change directly before launching the {homing+G1 Z0}, and then tune it like it's actually made.

oldman4U, 2020-12-10:
Hi.

So this is a feature request, right?

@digant73 Is this your "area" and something where you could add your experience as a user and developer, please?

Thank you

digant73, 2020-12-10:
once some PRs will be merged and reviewed I will try to have some time for an improvement

oldman4U, 2020-12-10:
This would be great digant73, thank you.

So I add this to the FR list.

CBDesignS, 2020-12-12:
Please do not shoot the messenger here. but I think this kind of question or a one very close to the topic was asked in the marlin git or maybe a "trusted website" a good long time ago and their short and blunt " as they usually are" answer was if you adjust anything related to hotend, nozel or Z you MUST start from scratch with any z offset and recalibrate from Z0. I like the idea of a safe offset as I use a glass bed but I could imagine a random newbie to printing changing a nozel and buring it into the glass and smashing it whereas if they had to start from z0 (the long way) there is less chance of this happening and they learn how to do it "correctly" instead of guestimating how much difference there might be. it takes seconds to zero and adjust the offset with paper using the tft as is

Chukozor, 2020-12-12:
To be honest following your reasoning it's then stupid to name it a "Zoffset tuning menu", just rename it "decrease Zoffset menu" then that would make more sense because you can only decrease it if you "do the things as they are meant to be". @CBDesignS

Chukozor, 2020-12-12:
to be accurate : a nozzle which dives in the bed by even 0.01mm is still a nozzle diving in the bed !

closed 2020-12-15 by bigtreetech

digant73, 2021-07-27:
@oldman4u this was a bug that I fixed in PR 1379. So this bug/FR must be removed from the list

PR #1379 (2020-12-15): resolves #1377 + cleanup
**BUG FIXES:**
* resolves #1377 (Mesh Editor freeze after one of the last merged PR)
* resolves #1364: Z Offset will start from probing Z0 point instead of absolute Z0 point. It means original Z offset is not applied when Z offset is enabled on Z offset menu
* resolves #1363 (probably solved now)
* resolves #1353 (value "SMART" for "fil_runout" is now correctly read from config.ini)
* MBL safe starting Z point: initial Z points are now set to 1mm instead of Z0. This should make more safe the MBL process in case of warped bed or not good manual leveling
* some cleanup

 [... cut]