bigtreetech/BIGTREETECH-TouchScreenFirmware #1367: [BUG] Steppers Locking Up only on Touch Screen Mode

Graidian, 2020-12-10:
I have a Creality Ender 5+ which I just recently installed a SKR Mini E3 v2 board into and coupled that with the TFT35 v3. When I first installed the two upgrades, everything worked great. I spent some time dialing in the firmware on the SKR until I got that set up. I spent some a few days printing in that configuration (updated firmware on the SKR, stock firmware on the TFT35) and thing were going great. Out of curiosity I thought I would check what firmware the TFT was running, and sure enough, it was a little behind so I went to upgrade _that_ firmware as well. That's where everything took a turn.

Now if I boot up the printer in LCD mode, everything runs fine. If I switch it from LCD to touch mode, I can home the print head, but any other movements I attempt (using the 'move' menu) lock up the motor and they will not budge at all. I can't even reach up and move it by hand (yes, I know you shouldn't move them by hand, and normally I wouldn't, this was only a step I took in attempting to diagnose what was going on). If I boot up the printer in touch screen mode it is the same issue, and in fact if it is booted up in touch screen mode and I switch to LCD mode...I have the same lock-up issue. The _only_ time I do not have the issue is if I boot in LCD and _stay_ there. I would like to go back to using both modes as that was one of the biggest reasons I got this board/screen combination, and honestly even if I have to revert the firmware if that solution actually works) I am fine with that.

As far as the config.ini file I am using, well, when I first updated the firmware for the screen I _did_ go in and make a couple adjustments (LCD color, print bed size) however when I started having the issue I went back to stock ini file. In fact I have not even changed the bed size back yet because I want to get the stork firmware working again before I adjust anything which I will do one setting at a time to see what breaks it.

Is this something anyone else has experienced? Or can reproduce? Latest firmware of Marlin 2.0.7.2 is on the board (reverted back to 'almost' stock firmware, I did have to set the proper motherboard etc) and latest stock firmware on the screen.

I can upload my configuration files if you would really like, but honestly, they are all sitting at stock right now. Thank you for the help!

oldman4U, 2020-12-10:
Hi Graidian.

Have you compiled your own firmware for the TFT also, or have you been using the pre compiled firmware? We saw some issues with pre compiled and maybe a self compiled is worth to give it a try.

Graidian, 2020-12-10:
I have not tried compiling my own software yet, I can give that a shot.

Graidian, 2020-12-10:
I tried a couple iterations of personally compiled firmware. From stock to customized. Same issue.

Msq001, 2020-12-11:
Hello, Can you test after send "M502 / M500" gcode to reset EEPROM of Marlin? I have encountered a similar situation before. Home is completely normal, but move will not move because the wrong auto bed leveling data is loaded.

bigtreetech, 2020-12-11:
Hello, Can you test after send "M502 / M500" gcode to reset EEPROM of Marlin? I have encountered a similar situation before. Home is completely normal, but move will not move because the wrong auto bed leveling data is loaded.

Graidian, 2020-12-11:
I need to give you a cookie because that did it! My goodness that was driving me insane, can't believe I didn't think of that. Nice work, thank you so much!

closed 2020-12-11 by Graidian