bigtreetech/BIGTREETECH-TouchScreenFirmware #1371: "What's new" - in case you would like to release a new version one day

oldman4U, 2020-12-11:
Hi @bigtreetech and @Msq001.

So far the following new features have been added since the last official release:

- Add support for updating settings through a config file without need to recompile firmware.
- Add better support for Printer SD / TFT SD /TFT USB from ESP3D
- Add detection of Mainboard firmware type( Marlin or Smoothieware)
- Add parsing of fan speed, Add large font, Add FW Retract and linear advance
- Add support for TFT43/50/70
- Add model preview for Cura plug-ins
- Add power loss recovery enable option
- Add store script function
- Add LCD2004 simulator
- Improved menus, new menus, new popup type, custom ACK message handling
- Add Disable Steppers option to menu
- Added idle to the rotary encoder knob led
- Display Marlin version & machine type in info menu
- Add SPI flash usage detail in info screen
- Compatible Repetier
- Add auto FW retract parameter
- Added save EEPROM settings button with configuration
- Add non-blocking toast notifications
- Add a separate sound control menu
- Added support M601 gcode pause add M600 same as M601
- Preheat presets extended to 6 materials
- Add notification window when pressed on titlebar showing last 3 notifications
- Added M913 - Hybrid Threshold Speed
- Added Mesh Bed Leveling
- Option for Mixing_extruder
- Improved Terminal menu based on TFT size and resolution
- Add support for Case light
- Add support for Home offset
- Added manual input to screen showing a value 
- Add a preheat menu to the ABL menu
- Layer update during printing from on board SD card
- Optional Print summary popup at the end of a print

BTT, Geminiserver is waiting for an answer from you. Maybe you can check #1037

HTTYD9, 2020-12-11:
Add power loss recovery enable option >>> can it work on TFT mode ? (Bgt ups 24v v1.0 ,btt relay v1.2 both connected to skr pro v1.1 ) and current bug fix 2.0.8.

oldman4U, 2020-12-11:
Hi @HTTYD9.

This lists all features for a whats new document in case BTT releases a new firmware version soon. Please start a new Issue for feature requests and a bug you discovered. 

Please check the pinned issues for already reported tickets related to your request. 

Thank you

oldman4U, 2021-02-10:
Maybe we want to keep this ticket opened?

adelyser, 2021-02-20:
Why isn't BTT creating releases? New features and bug fixes shouldn't have to be tracked in an open ticket, that's why you can do releases in github. It's also difficult to tell what version is current without having to sort through the code or just loading the latest "3.0.x.x" binary to find out.

bttguy, 2021-03-20:
@bigtreetech #1400 collects the current features that are supported in RRF for the release :)

oldman4U, 2021-03-24:
@bttguy 

Could you please provide the RRF features which are new in the same format I used above, so I can add them to the list.

Thank you

closed 2021-04-03 by oldman4U

bttguy, 2021-04-03:
@oldman4U I saw that you or somebody already added the main implemented functions to the list, that's why I didn't answered, sorry. But with the new release notes RRF features are covered nicely as well, so thank you! I already started working on the RRF documentation, let's see if the Wiki page will be editable soon, or I must add it to the readme.

oldman4U, 2021-04-03:
Hi

No reason to be sorry. It is all good!👍🏻