bigtreetech/BIGTREETECH-TouchScreenFirmware #1375: [BUG] onboard SD card

radek8, 2020-12-11:
After merging the last PRs, the onboard SD card is not available in touch mode.

kisslorand, 2020-12-11:
Was it available with PR #1344?

radek8, 2020-12-11:
Tested. Caused after merging PR #1333

radek8, 2020-12-11:
@mehmetsutas  can you check your PR #1333?
After merging it, the onboard SD card is not available.

oldman4U, 2020-12-11:
This has to be related to the last PR's added right before the merge. Because I tested #1333 successfully except for the last minor changes.

Have you tried to reset to defaults radek8?

radek8, 2020-12-11:
Pardon. My fault. After updating config.ini again, everything is fine.

closed 2020-12-11 by radek8

oldman4U, 2020-12-11:
Can I now ask you to close this ticket, Please😁

mehmetsutas, 2020-12-11:
@radek8 Since I don't have access to printer right now, I will be testing it tomorrow. But my machine is runnig with these changes and don't have an issue right now. Even I had prints today.

In the last couple of days there were several merges, I did not update my machine with the latest merges. #1333 did not make any changes on lower level which may effect SD card system or serial communication to that level. I suspect the lighter memory allocation or combination of #1333 with another merge may have a side effect. I will test it in detail tomorrow.

Meanwhile, can you give some detail about the problem?

* Is the Board_SD icon available on print screen? or is it totally lost? If not lost what happens after you select it? Do you get the files list? or just black screen of loading? If you get the files list what happens when you select a file?

As #1333 introduces a new config option, configuration storage has to be updated so that options do not get mixed. As @oldman4U pointed out not to mix settings, reset to defaults has to be run after the firmware update. (Reset to defaults is at the end of the features screen after the last configuration option, it looks like config item but actually It resets the configuration storage) Did you try running it?

oldman4U, 2020-12-11:
Too late.....

🤚🏻

oldman4U, 2020-12-11:
Maybe I will add the reset information tomorrow.

mehmetsutas, 2020-12-11:
I think I should be less chatty. Before finishing my messages problems are solved. :)

oldman4U, 2020-12-11:
We enjoyed every single word you wrote 😆

radek8, 2020-12-11:
The config.ini file has become corrupted
When I wanted to reload the configuration from the config ini file, I found that it was corrupt and contained meaningless characters.
I uploaded a new file to the card and everything is fine after upgrading the printer.

kisslorand, 2020-12-11:
> I suspect the lighter memory allocation

Memory allocation is enlarged for the serial port that goes to the mainboard.