bigtreetech/BIGTREETECH-TouchScreenFirmware #1377: [BUG] (MBL - No Mesh Saved - No Mesh In Editor)

ETE-Design, 2020-12-12:
When using Manual Mesh Bed, before you have to choose what slot to save to, and you could even edit the mesh from the TFT, with newest version - Compiled Today 12/12-2020 you don't have the possibility to choose what slot to save to, and the Mesh Editor just show a screen with no dots and then freeze, so when start printing no mesh is used.

digant73, 2020-12-12:
different bugs were provided by one of the last PR. Restoring my previous code, the map is available again and there is no freeze. However due to other changes on other API, the screen is not refreshed in case of popup. Also the navigation cursor does not work properly.
I was working on a FR for Z offset menu, I will restore part of my code (working properly), but it will be needed work on other API to fix the bugs introduced on that API.

ETE-Design, 2020-12-12:
@digant73 With your code will it also be possible to again choose what slot to save the mesh to? Or is it just saved in Slot 0 cause you don't need others with MBL?
Yeah it is fustrating that someone tell you to do one thing, and then others dosen't... Is the working code on your Github? Cause right now MBL is broken :-(

digant73, 2020-12-12:
MBL didn't have a slot to select. It is UBL only.
I'm just creating a PR with the fix to Mesh Editor.
I will add then a FR on taht PR

ETE-Design, 2020-12-12:
@digant73 Still same problem, Seem's like it dosen't save the Mesh No matter What I try to do... As you can see at the picture it says Off even egen I enabled BL: On, and still freeze at the Mesh...
[image]
[image]

radek8, 2020-12-12:
I use bilinear leveling and I have the same problem.

ETE-Design, 2020-12-12:
@radek8 Did not solve the problem for MBL :-(

PR #1379 (2020-12-15): resolves #1377 + cleanup
**BUG FIXES:**
* resolves #1377 (Mesh Editor freeze after one of the last merged PR)
* resolves #1364: Z Offset will start from probing Z0 point instead of absolute Z0 point. It means original Z offset is not applied when Z offset is enabled on Z offset menu
* resolves #1363 (probably solved now)
* resolves #1353 (value "SMART" for "fil_runout" is now correctly read from config.ini)
* MBL safe starting Z point: initial Z points are now set to 1mm instead of Z0. This should make more safe the MBL process in case of warped bed or not good manual leveling
* some cleanup

 [... cut]
commit d0593f5b (2020-12-15): resolves #1377 + cleanup (#1379)
closed 2020-12-15 by bigtreetech