bigtreetech/BIGTREETECH-TouchScreenFirmware #1378: [BUG] at filament uload printer freezes

Bastlwastl84, 2020-12-12:
Filament unload menue makes printer freeze
### Steps to reproduce
Power on at Touch Mode
Preheat
Preheat PETG
Move
(Un)Load
Temperature stays and not updating
Moves Head, but not unloading, cause of not updating Temperature
If Temperature allready was reached: Temperature not updating, unloading and loading, but goes not back to mainmenu, Always popping up Paused for user continue.
Temperature is only corrected after power off and again on

Preheat and unload fail: https://youtu.be/QSXnkP-bYEY
switching back after preheat https://youtu.be/akdnzdOg2pM

had this issue, too:
[image]

[Bug 2020-12-12.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5683336/Bug.2020-12-12.zip)

### Hardware Variant
TFT35 E3 V3, Firmware 11.12.2020
SKR Mini E3 V2 Marlin Bugfix 2.0.8.x 11.12.2020
### Additional Information

Bastlwastl84, 2020-12-12:
after Power off and on it shows me this 173 deg. :

[image]

Bastlwastl84, 2020-12-12:
It must be a preheat-bug and a second load/unload bug. Tested a bit more and after preheat there was no update of Temperatur, while at manual setting tje Temperatur was working correctly

oldman4U, 2020-12-12:
And the filename is correct in first picture?

Bastlwastl84, 2020-12-12:
> And the filename is correct in first picture?

No.
[image]

oldman4U, 2020-12-12:
[Filament Unload 12.12.2020.mov.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5683606/Filament.Unload.12.12.2020.mov.zip)

Have you tried to wait, when it asks to do so?

Bastlwastl84, 2020-12-12:
2 minutes i had waited, not longer, that not shouldn't happens a overheat, but no changesVon meinem Sprechknochen gesendet.

oldman4U, 2020-12-12:
Have you checked the movie I have attached to my last answer?

Marlin v2.0.7.2 and TFT firmware from yesterday, self compiled.

Bastlwastl84, 2020-12-12:
> Have you checked the movie I have attached to my last answer?
> 
> Marlin v2.0.7.2 and TFT firmware from yesterday, self compiled.

it was working before i changed to this firmware like at your video

Bastlwastl84, 2020-12-12:
[image]
And than the 2nd time today that the printer doesn't like to heat up after starting a file from sd at tft. After stop and start again it workes.

kisslorand, 2020-12-13:
@Bastlwastl84 
When you pressed "Preheat" you had the option "Both" selected. So you stated to heat the nozzle and the bed also.
After you click "Load" or "Unload" and that popup appears, if you click OK, you'll see the temperature of the nozzle in the middle. 
Since you wanted to heat both the nozzle and the bed, you cannot exit that menu until both are not at the preset temperature.
It's exactly the scenario I was worried about when you would have to use preheat before load/unload.  People would not take care selecting just the nozzle, since that is the only one needed for Load/Unload.

Bastlwastl84, 2020-12-14:
> @Bastlwastl84
> When you pressed "Preheat" you had the option "Both" selected. So you stated to heat the nozzle and the bed also.
> After you click "Load" or "Unload" and that popup appears, if you click OK, you'll see the temperature of the nozzle in the middle. It will eventually rise, just have patience.
> Since you wanted to heat both the nozzle and the bed and clicked "Unload", you cannot exit that menu (Load/Unload) until both are not at the preset temperature ans unload procedure is done. It's a safety measure I introduced so people would not click around while the printer is in "wait" state. Without this safety measure the TFT would lockup after a few clicks because of the "wait" state of Marlin motherboard.
> It's exactly the scenario I was worried about when you would have to use preheat before Load/unload. People would not take care selecting just the nozzle, since that is the only one needed for Load/Unload.
> 
> Side note: in the whole menu system whilst a popup is active, the background menu is not active, it is not refreshed

Bouth, cause of change it before print with different colour. 

I was stick at the menu after it HAD done the change and didn't come out of this menu. 

The preheat didn't update, so the filament change did not starting.

If im doing a preheat manually tje Temperatur raises fast.

Bastlwastl84, 2020-12-14:
https://youtu.be/n5IC3YjZJUc
Preheat than extruder makes freeze the TFT. Nothing changed since uploading the config files

kisslorand, 2020-12-14:
What if you wait until the temperatures are reached?

Bastlwastl84, 2020-12-14:
Than didn't came out of Menu

oldman4U, 2020-12-14:
Most likely something went wrong during the update. I would reset the TFT and install the current firmware again.

Bastlwastl84, 2020-12-14:
Will try at afternoon Von meinem Sprechknochen gesendet.

oldman4U, 2020-12-15:
Hi.

New master is available. Please check

Thank you

Bastlwastl84, 2020-12-15:
> Hi.
> 
> New master is available. Please check
> 
> Thank you

New firmware, same Config:
Starting preheat PETG, but not raising Temperatur
[image]
After a bit waiting. Temperatur didn't raised, error
[image]
After power off and on:
[image]

Bastlwastl84, 2020-12-15:
After 4 Minutes waiting to cool down: no Update
[image]

kisslorand, 2020-12-15:
What if you heat separately, not using preheat function?

oldman4U, 2020-12-15:
Firmware from today?

Attach your config.ini.

Have you tried to reset the TFT?

Bastlwastl84, 2020-12-15:
Manually was working, only at preheat and after reboot it was not updating the changes.

.

After i did it manually preheat Nozzle and Bed seperately workes.

But if i'm trying at heated state to unload (workes) than load (workes), than i stick at this state..every click on ok the same message appears:
[image]

kisslorand, 2020-12-15:
As far as I see these messages (Kill, Pause) are coming from the printer, it has nothing to do with the TFT. I also see a massive temperature overshoot for the heated bed.
When TFT freezes it is because the printer is in a wait state.

Bastlwastl84, 2020-12-15:
Allready written: same configuration.h and resetted. Yes from this evening.Von meinem Sprechknochen gesendet.

Bastlwastl84, 2020-12-15:
> As far as I see these messages (Kill, Pause) are coming from the printer, it has nothing to do with the TFT. I also see a massive temperature overshoot for the heated bed.
> When TFT freezes it is because the printer is in a wait state.

I cooled down a bit. Was 85 deg before.
Serial always on perhaps the problem?

oldman4U, 2020-12-15:
I also expect the TFT not to be the problem. So I will load you config and
try Preheat after that.

kisslorand, 2020-12-15:
I also played today with preheat and couldn't make my TFT or printer to misbehave.

oldman4U, 2020-12-16:
Can’t find your config.ini. Please attach

Bastlwastl84, 2020-12-16:
Didn't used a precompiled FWVon meinem Sprechknochen gesendet.

oldman4U, 2020-12-16:
Precompiled or not. You would like to get more information about the
preheat issue so someone has to be able to reproduce the issue. Or should
we leave it as it is?

Bastlwastl84, 2020-12-16:
> Precompiled or not. You would like to get more information about the preheat issue so someone has to be able to reproduce the issue. Or should we leave it as it is? Bastlwastl84 <notifications@github.com> schrieb am Mi. 16. Dez. 2020 um 13:58:
> […](#)

> Didn't used a precompiled FW
i did everything new now, but also with Marlin bugfix 2.0 from today its the same. reloaded factory reset at board and resettet TFT. FW from TFT from 9.11. is working with that. now that installed
[208 board.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5703089/208.board.zip)
[tft.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5703090/tft.zip)

oldman4U, 2020-12-16:
OK. SKR mini I do not have, but SKR E3 DIP.

All I miss is the config.ini of the TFT.

Bastlwastl84, 2020-12-16:
^1

> OK. SKR mini I do not have, but SKR E3 DIP.
> 
> All I miss is the config.ini of the TFT.

i dont have a config.ini, i compiled it with configuration.h at BIGTREETECH-TouchScreenFirmware\TFT\src\User and used vscode

oldman4U, 2020-12-16:
Have you read the ReadMe how to update the firmware?

Bastlwastl84, 2020-12-16:
> […](#)

A pre-compiled Config need the language-file and config.ini, not a self compiled, cause you are setting these things before at configuration.h

oldman4U, 2020-12-16:
Aha. And where do you get this information from?

oldman4U, 2020-12-16:
I would really like to know this, because this would be a big misunderstanding in the documentation. And in case you have been able to update the firmware without settings a warning that the config.ini has to be updated too, this would be an issue in the software.

But for this I would have to know.

And maybe updating the config.ini also solves all your issues!!;-)

Bastlwastl84, 2020-12-16:
> I would really like to know this, because this would be a big misunderstanding in the documentation. And in case you have been able to update the firmware without settings a warning that the [...]

i never needed the config.ini since have the skr mini e3 v2 and tft35 e3 v3.

oldman4U, 2020-12-16:
You will never need the config.ini for the SKR, but for the TFT. Because of
this we mention it, that you have to upload it together with the firmware.
Having a mismatch of keywords in config.ini and the firmware can cause any
kind of issues which we do not know and this could explain some of the
strange issues only you have.

Please let me know what makes you believe you do not need a config.ini (in
case you use self compiled firmware)?

Bastlwastl84, 2020-12-17:
> You will never need the config.ini for the SKR, but for the TFT. Because of this we mention it, that you have to upload it together with the firmware. Having a mismatch of keywords in config.ini and [...]

All Things i had entered at configuration.h was at Firmware. If using a precompiled FW its askinf for languagefile and config, but not with self compiled FW. It Updates the fonts and icons if copying the TFT35 Folder on SD. Now using an older FW from my Backup without this Problems i had reported here.

oldman4U, 2020-12-17:
So all issues are gone using an old version from a backup?

Bastlwastl84, 2020-12-17:
These yes with version before the Vase Mode bug was fixed. But now running marlin mode and found new Bugs at Marlin Bugfix (reported at marlin github) ubl with more than 12 pints not saving and Marlin reboots and partcooling fan problems (Full or off, but nothing between)

oldman4U, 2020-12-17:
But now you are trapped. Because old firmware will not be fixe.

Bastlwastl84, 2020-12-17:
Waiting for next update, than i will checking it again

oldman4U, 2020-12-17:
Ok. And maybe try the config.ini in the meantime. I changed the ReadMe in
the meantime.

radek8, 2020-12-21:
> ^1
> 
> > OK. SKR mini I do not have, but SKR E3 DIP.
> > All I miss is the config.ini of the TFT.
> 
> i dont have a config.ini, i compiled it with configuration.h at BIGTREETECH-TouchScreenFirmware\TFT\src\User and used vscode

You must always load the Config.ini file, even if you are compiling your own bin file.

jedisintraining3, 2021-01-24:
I am having the same issue. Screen freezes randomly and when printing over usb, so does print! I am having a hard time brand new screen!

closed 2021-04-01 by stale[bot]