bigtreetech/BIGTREETECH-TouchScreenFirmware #1382: [BUG] Gen-L - "No printer attached" at start

FlailAway, 2020-12-12:
### Description
MKS Gen-L v1.0 board
Start up with the Touch selected.
Without a printer attached I get "No printer attached" 
Holding the button down I can see the Marlin option but tapping does nothing.
I plug in the printer and tap Marlin again and the screen goes black for a few seconds then I see Marlin booting up.
Once Marlin is running, I can unplug the printer and go back to touch which then works fine.

As above
### Steps to reproduce

1: As above

**Expected behavior**

**Actual behavior**

### Hardware Variant

TFT35, v3.0
### TFT Firmware Version & Main Board Firmware details

Both fresh downloads today
Marlin bugfix 2.0.8
BIGTREETECH-TouchScreenFirmware-Vx.x.26

### Additional Information

FlailAway, 2020-12-12:
Just found that if I start the printer in Marlin mode on the TFT35, it then works fine in Touch mode without ever connecting a printer.

radek8, 2020-12-12:
Check the cable connections, it looks like a bad connection.
Send photos with connection, save configuration files.

Foxcor, 2020-12-13:
@FlailAway 
In Marlin
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
#define BAUDRATE 250000
In config.ini
baudrate:6

FlailAway, 2020-12-13:
> Check the cable connections, it looks like a bad connection.
> Send photos with connection, save configuration files.

Connections OK, not new to this. The problem is that Marlin is not started and it declares the "No printer" Marlin needs to be booted by the TFT.

FlailAway, 2020-12-13:
> @FlailAway
> In Marlin
> #define SERIAL_PORT -1

Latest Marlin Bugfix rejects the "-" cannot remember the exact error but will NOT compile with a Minus there.

I left BOTH Serial ports as they are defined and the BAUD rate has nothing to do with it.

The TFT needs to start Marlin. That's why starting in the Marlin mode makes it work fine, but just a PITA to remember to turn it off in Marlin mode. Or do the merry dance of starting in TFT mode then going to Marlin then coming back to TFT. Not new to this and that's why I posted the follow up above.

radek8, 2020-12-13:
In Marlin mode, does the TFT display communicate well with the printer?

radek8, 2020-12-13:
For information
#define SERIAL_PORT -1 is for USB communication
Option 0 is for communication on the AUX-1 connector

kisslorand, 2020-12-13:
Use
SERIAL_PORT 0
SERIAL_PORT_2 -1

oldman4U, 2020-12-14:
Have you tried to start in touch mode with ONLY the 5pin cable connected. Does is connect then?

FlailAway, 2020-12-15:
> Use
> SERIAL_PORT 0
> SERIAL_PORT_2 -1

Thanks, but same thing, still no printer, but using the v25.bin works fine. See my next reply.

FlailAway, 2020-12-15:
> Have you tried to start in touch mode with ONLY the 5pin cable connected. Does is connect then?

Just did that and still "No Printer." 

I tried using the v25.bin .zip and it all works fine. I don't need to first go to Marlin and boot it, then back to touch. It is working perfectly and I have not touched Marlin since the initial TFT35 install. And yes, I made all the suggested changes to the Configuration_adv.h file suggested in the "config.ini" back at the first install.

But, I prefer v26 as it is showing the Co-ords on the "printing" screen that v25 does not. I was trying to debug some GCode and need those Co-ords.

Something is clearly missing or changed in v26, it just needs to boot Marlin. That is not happening as when I start v26, I have to first go to Marlin and it takes a while for Marlin to show the boot screen and then the "Info" screen. Going back then v26 will work as it should until the next power off-on and it is the Marlin dance again.

oldman4U, 2020-12-15:
What else do you have connected to the TFT?

Have you updated firmware, config.ini and TFT folder?

Have you tried to reset the TFT and/or EEPROM of the mainboard?

radek8, 2020-12-15:
Try the config.ini option
serial_always_on: 0

Foxcor, 2020-12-16:
@FlailAway 
> I tried using the v25.bin .zip and it all works fine.

I think I figured out the problem. Are you using pre-built assemblies? If yes, then I recommend compiling the firmware yourself. I had such a problem with ready-made assemblies, but when I made the build myself, everything worked

oldman4U, 2020-12-16:
@Foxcor 

Hi,

I would absolutely agree with you, there have been many problems with pre compiled versions in the past, but those issues SHOULD be gone. At least for the TFT's I use, the 24,28 and 35 (and soon 43), the precompiled versions are working well now. BUT it could be related and because it gives you more options, I would always recommend to compile the own firmware.

;-)

traffic-light, 2020-12-16:
> @FlailAway
> 
> > I tried using the v25.bin .zip and it all works fine.
> 
> I think I figured out the problem. Are you using pre-built assemblies? If yes, then I recommend compiling the firmware yourself. I had such a problem with ready-made assemblies, but when I made the build myself, everything worked

And that's why they should be updated after a merge or be completely removed...

oldman4U, 2020-12-16:
Using the firmware from Dec. 12th, the precompiled version was the same. So someone updated them or they have been updated automatically. Will check again with the next round of PRs pushed.

FlailAway, 2020-12-16:
Let me state **again** v25 works, v26 does not.
If it was loose cables or wrong Marlin settings why would v25 work?
A user should NOT have to compile their own .bin files.

FlailAway, 2020-12-16:
At least for the TFT's I use, the 24,28 and 35 (and soon 43), the precompiled versions are working well now. 

Where would I get those? I see only v26  as the latest.

FlailAway, 2020-12-16:
> Try the config.ini option
> serial_always_on: 0

Thanks, when it has finished this print I will try installing v26 again with that setting.

oldman4U, 2020-12-16:
You can also stay with v25. There is no need to update just because there is a newer version available.

Precompiled can never work 100% because of the many possible combinations of hardware and functionality. Fortunately compiling is very easy in general and unbelievable easy for the TFT's. I would give it a try. ;-)

This is the latest version: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/archive/master.zip

Let me know

radek8, 2020-12-16:
Why don't you send the configuration files?
TFT: config.ini
Marlin: configuration.h, configuration_adv.h
It works for everyone and not for you, so the problem will be in the configuration or connection.

FlailAway, 2020-12-17:
> You can also stay with v25. There is no need to update just because there is a newer version available.

Thanks, that is what I am doing. I was bug-fixing some GCode files for a friend and v26 shows the Co-ords on the printing screen, v25 does not. But they are all fixed now so largely moot.

FlailAway, 2020-12-17:
> It works for everyone and not for you, so the problem will be in the configuration or connection.

Rubbish!
I have TWO SD CARDS. I copy a fresh TFT35 Folder and .v25.bin on to one and the TFT35 Folder and .26.bin on the other. I have the SAME config.ini on both AND the same untouched Marlin on the Gen-L board. The v25 works, v26 does not UNTIL I hold down the button, tap Marlin then hold the button and go back to Touch. THEN the "Printer" message has gone.

closed 2020-12-17 by FlailAway

radek8, 2020-12-17:
Attention!
You cannot use the same config.ini file for V25 and V26
If you don't want to help, keep using the V25 if it works for you. But you will lose new features.